cryptoadvance/specter-diy #27: simulator freezes

gorazdko, 2019-12-17:
## Abstract

> gorazd@gorazd-MS-7C37:~/Projects/specter-diy/src$ ../f469-disco/micropython_unix -c "import main; main.main()"

My simulator freezes after entering password:

generate new key -> continue - > Enter your password -> freezes

stepansnigirev, 2019-12-18:
I will try to reproduce tomorrow. I got a linux machine today, hopefully it will freeze on my laptop as well =)

stepansnigirev, 2020-01-06:
Ok, I verified that my linux build it also freezes at the same time as yours.

It might be that upython garbage collector frees some objects of littlevgl and crashes the gui.

GC is triggered when the key derivation happens (`init_keys` function) probably because I didn't optimize bitcoin module yet and it uses memory not very efficiently.
I will try to optimize the bitcoin library using `memoryview` and also get lvgl devs aware of the problem as soon as I make a reproducible example without any bitcoin-related stuff.

stepansnigirev, 2020-01-07:
After some time I figured out that it was caused by a bug in secp256k1 binding module - we use preallocated context there and on 64bit machines it should be larger. 
There should be a more elegant way to solve the problem but https://github.com/cryptoadvance/specter-diy/commit/0346ad7e24c9c96b2347577a9263347b1cfe3174 should fix it. 
Can you try?

gorazdko, 2020-01-07:
Yes, i tested. This works!

closed 2020-01-07 by stepansnigirev