cryptoadvance/specter-diy #37: Bug: None value issue when entered mnemonic

k9ert, 2020-01-28:
Sometimes, but not always, i get the below issue after i've entered my 12-word mnemonic (e.g. (11xghost machine") in the simulator and i'm clicking on "done".

You see this issue in the logs:

```
➜  specter-diy git:(testing) ✗ ./specter.sh sim    
Traceback (most recent call last):
  File "gui/screens.py", line 290, in cb
  File "main.py", line 346, in mnemonic_entered
AttributeError: 'NoneType' object has no attribute 'strip'
```

This is the code-snippet:

```
def mnemonic_entered(mnemonic):
    global entropy
    entropy = bip39.mnemonic_to_bytes(mnemonic.strip())
    ask_for_password()
```

stepansnigirev, 2020-01-28:
Just saw the same problem today. It happens when you click on the keyboard but not on the button itself.
Gonna fix it now...

stepansnigirev, 2020-01-28:
Should be fixed with https://github.com/cryptoadvance/specter-diy/commit/a165b564f7cfd562de51a77e49acc7d61c1839fd

closed 2020-01-29 by stepansnigirev