cryptoadvance/specter-diy #44: SD card support

stepansnigirev, 2020-02-22:
We have SD card slot, we can use it.

What we would need:
- check if SD card is there
- mount SD card, read file from it
- save file to SD card
- unmount
- repeat

At the moment if you turn on the device with SD card, Micropython will try to load `main.py` file from SD card and execute it. It can be solved by placing a special file to the flash - don't remember the name exactly, something like NOSD.

If this file is in flash SD card will be ignored on boot and firmware will be loaded from flash. 
After that I **can** do the steps above **once**, but as soon as I unmount the SD card I can'd mount it again anymore.

I think this requires changing micropython itself, in particular, our fork: https://github.com/diybitcoinhardware/micropython

stepansnigirev, 2020-10-19:
Partially solved with #83 , for now only for key storage.
But looks like all the problems are solved so we could load PSBTs from SD card as well.

SaroC, 2022-12-11:
Issue:
- Initial FW initial_firmware_v1.8.3.bin flashed with `st-flash write ~/development/specter-bins/initial_firmware_v1.8.3.bin 0x8000000`
- If I boot the device simply (without a SD card inserted into the socket) all works properly I insert the PIN and the OS show me the menu.
- If I boot the device with a SD card into the socket (FAT formatted)...After that I insert my PIN I receive the follow output:
  `Error! Somenthing unexpected happened...
  Traceback (most recent call last):
  File "specter.py", line 76, in handle_exception
  File "specter.py", line 156, in main
  File "specter.py", line 186, in initmenu
  File "keystore/sdcard.py", line 110, inis_key_saved
  File "platform.py", line 95, in mount_sdcard
  OSError: [Errno 19] ENODEV`

The 32GB SD card was formatted with Win (FAT32 with 16kB allocating size) first and also with Linux's mkfs program (with `-t vfat argument`). The SD card was used with and without file inside in.
If I remove the SD card, also while the device is in on state, the FW works properly again showing me the menu again.
Seems that the presence of the SD card is enough to cause the error.

 Any suggestion? maybe the size of SD card could be the root cause?  a smaller one could fix the issue?
 Many Thanks for the support.

schnuartz-ai, 2026-08-29:
## Audit of current HEAD (3f3c831)

I investigated this tracking issue against the current code. Summary of the original TODO list vs. current implementation:

| TODO item | Status | Evidence |
| --- | --- | --- |
| SD card for key storage (#83) | ✅ Done | `SDKeyStore` in `src/keystore/sdcard.py` — stores an AEAD-encrypted mnemonic on the SD card, keyed by a device-derived secret (`enc_secret`). |
| Load / save PSBT transactions (#130) | ✅ Done | `SDHost` in `src/hosts/sd.py` loads `.psbt`/`.txt`/`.json` from SD (`get_data`) and writes the signed result back (`send_data`). Menu entry "Open SD card file". |
| Import / export wallet in our format (#130) | ✅ Done | Wallet export to SD as plaintext JSON via `Wallet.export_menu` (`src/apps/wallets/wallet.py:67`); import via the `addwallet` host command / `.txt` descriptor files (`src/apps/wallets/manager.py:241`). |
| Import / export wallet in CC (Coldcard) format | ❌ Not done | Only **xpub** export in Coldcard JSON exists (`src/apps/xpubs/xpubs.py:230` `export_coldcard`). No import/export of *wallet descriptors* in Coldcard format. |
| Export / import seed in encrypted form | ✅ Done | `save_mnemonic`/`load_mnemonic` write/read the mnemonic with `save_aead`/`load_aead` (`src/keystore/ram.py:90` → `aead_encrypt`), so the seed on the SD card is authenticated-encrypted and only readable by this device. |

### Side report in comments (ENODEV with 32GB SD card)
@SaroC's 2022 report (`OSError: [Errno 19] ENODEV` at mount time with a 32GB FAT32 card) is **not** specifically handled in current code. `SDCard.mount()` (`src/platform.py:74`) calls `os.mount(self._sd, "/sd")` with no `try/except`, so a mount failure still propagates as an unhandled traceback. This overlaps conceptually with #233. The trace references a now-removed standalone `mount_sdcard`, but the same uncaught `os.mount` path exists today.

### Reproduction / validation
- Code inspection only (no hardware). The SD code paths are hardware-dependent (`pyb.SDCard`); the simulator uses a stub `SDCard(None, None)` (`src/platform.py:128`) so mount failures can't be reproduced there.
- No automated tests cover the SD host/keystore (no `test_*sd*` cases found).

### Security review
- Seed-at-rest on SD is AEAD-encrypted with a device-derived key — appropriate.
- Wallet descriptor export to SD (`export_menu`) is written as **plaintext** JSON. This is by design for sharing with coordinators, but it does expose xpubs to anyone with the SD card. Worth documenting for users; not a fund-loss risk.
- SD files are host-controlled input parsed by embit's PSBT parser; trust boundary is unchanged from other host inputs.

### Recommendation
This is a meta-tracking issue. With 4 of 5 TODO items implemented, the only genuinely outstanding work is **Coldcard wallet-descriptor import/export**. I'd suggest either:
- **Recommendation:** close this as mostly-resolved and open a focused issue for "Import/export wallet in Coldcard format", referencing this issue; **or**
- **Recommendation:** keep open but update the checklist to mark the 4 completed items and retarget the issue at the single remaining CC-format task.

Either way, the ENODEV mount-failure robustness gap is better tracked under #233 / a dedicated issue rather than in this feature checklist.