cryptoadvance/specter-diy #45: Improve recovery screen (bip39 words)

stepansnigirev, 2020-02-22:
Entering the recovery phrase is still pretty painful.

I see two solutions:
- change the orientation to landscape mode just for this screen - keyboard will be larger
- add suggestions on top of keyboard, similar to modern phones
- try something like swipe keyboard

dimaatmelodromru, 2020-06-16:
* allow entering only first 4 characters of each word, proceed to **Next word** automatically (turn on/off on the input screen)
* optimize swipe for BIP39 words for extra robustness and speed of input
* sound feedback via Waveshare scanner buzzer when auto-proceeding to the next word (turn on/off on the input screen, default - on for 4-char input). Screen could flash in silent mode.
* display cursor or highlight active input field for clarity

dimaatmelodromru, 2020-06-16:
BTW, there is a glitch: if you start inputting words, then press **Back**, then press **Enter Recovery Phrase** again, you are presented with an empty word input screen that is being filled with your *previous input* **after** you start new input

stepansnigirev, 2020-07-04:
In #66 we added autosuggestions, still not perfect but already better.
Swipe would be great but looks pretty complicated to implement.
I am also thinking about adding a "rotate screen" button for screens with the keyboard, then buttons would be larger.

schnuartz-ai, 2026-08-29:
## Audit of current HEAD (3f3c831)

This is a UX-enhancement request for the recovery-phrase entry screen. Checking each proposed idea against the current implementation in `src/gui/screens/mnemonic.py` (`RecoverMnemonicScreen`):

| Idea | Status | Notes |
| --- | --- | --- |
| Landscape mode for larger keyboard | ❌ Not implemented | No rotate/orientation control exists anywhere in `src/`. The keyboard is a fixed-portrait `HintKeyboard` (`mnemonic.py:186`) at height 260. |
| Suggestions above keyboard | ✅ Implemented | PR #66 added an autocomplete button-matrix (`self.autocomplete`, `mnemonic.py:184`) driven by a `lookup` function. The "Next word" button auto-accepts the single remaining candidate (`mnemonic.py:329-334`). |
| Swipe keyboard | ❌ Not implemented | Stepan noted in the comments this is "pretty complicated to implement"; still absent. |

Additional community suggestions (from @dimaatmelodromru) — mostly **not** implemented:
- 4-char auto-proceed: only partial — `get_mnemonic` enables "Next word" once the typed prefix yields a single candidate (`mnemonic.py:274-278`); there's no automatic advance after exactly 4 chars.
- Sound feedback on the scanner buzzer: not present.
- Cursor / active-field highlight: no explicit highlight beyond the `MnemonicTable`.

### Reported glitch (Back → re-enter shows stale input)
The comment from @dimaatmelodromru describes stale state when re-entering the recovery flow after pressing Back. I couldn't reproduce this on hardware; the screen is recreated per entry in the current code (`RecoverMnemonicScreen` builds a fresh `MnemonicTable` in `__init__`), so it may already be resolved, but I have no hardware/simulator confirmation.

### Reproduction / validation
- Code inspection only. The recovery screen uses LVGL and physical-keyboard layout that the simulator can render, but I did not run the full flow on hardware.
- Related: #223 asks for the same UX improvements (adds a "Previous word" button request) and is effectively a duplicate / continuation of this issue.

### Security review
No security impact — this is pure input UX. Mnemonic words are handled in memory and validated via the `checker`/`fixer` callbacks; no change to trust boundaries.

### Recommendation
**Recommendation: keep open**, but consider consolidating with #223 (both target the same recovery-entry screen). The autocomplete half is done; landscape rotation, swipe, and the finer UX tweaks (Previous-word button, sound, cursor) remain valuable, lower-priority enhancements. A single consolidated "recovery phrase UX" issue would track this more cleanly.