cryptoadvance/specter-diy #50: Keep track of receiving and change addresses

stepansnigirev, 2020-02-22:
Software wallets often have a gap limit of 20 addresses, so if my last change address was `m/.../1/123` and in the next transaction I use a change `m/.../1/9999` it can cause lock of funds.
We should **warn** the user if the change index is weird. To do this we should store the largest change index and warn if the next one is more than 20 larger.

gorazdko, 2020-05-14:
Is that meant when scrolling over wallet addresses (with Next/Previous button), we should warn at the address that exceeds the gap limit?

We could determine the address index from the scanned PSBT's inputs and store it into 
_wallet.json if bigger?

To determine the address index we need to loop wallet.address(idx, change)
with incrementing idx for both change values and match with PSBT inputs?

stepansnigirev, 2020-05-14:
My thinking is:
- when the wallet is created the max index for each wallet is set to `0` (both receive and change) and stored in `wallet.json` file
- when we are asked to sign a transaction we check indexes of every input and output, and if we see that it is higher than `index + gap_limit` - we show a warning.
- if the user confirms the transaction signing, we go through all inputs and outputs, find maximum indexes of receiving and change addresses, and if they are larger than stored indexes - update them

I think the address screen should start by default with a stored index, not zero. And it makes sense to add a warning if the displayed address is higher than `index+gap_limit`. Both in browse address screen and when we verify the address.

The warning can be just another label, with red text, or a "prompt", I don't know what's better.

What do you think, does it make sense?

stepansnigirev, 2020-05-14:
> To determine the address index we need to loop wallet.address(idx, change)
with incrementing idx for both change values and match with PSBT inputs?

PSBT inputs and outputs contain information about derivation indexes, so we don't need to loop through them, we can get them from PSBT directly

gorazdko, 2020-05-14:
Ok makes sense. I will grab this issue then.

PR #59 (2020-05-20): check for address gap limit violation
## Abstract

Close #50 

## Status

~WIP~ Ready for review:
- [x] final testing
- [x] extracting indexes from compressed and full PSBTs
- [x] noticed also: on verify address screen cancel button does not work if running telnet
   * opened a new issue #60
- [x] when creating a new wallet initialize indexes with -1, so that we correctly browse wallet addresses. If we initialize with 0  we cant know if the address was used already or not
- [x] when browsing wallet addresses we currently allow to go back to address Nr. 1 regardless of max indexes . Should we d [... cut]
closed 2020-05-20 by stepansnigirev