cryptoadvance/specter-diy #56: qr code animations

gorazdko, 2020-04-08:
## Abstract

I did a proof of concept for scanning animated qrs automatically (without buttons): https://github.com/gorazdko/specter-diy/tree/animiated_qr_video
  *  scanner is scanning in continuous mode with the following parameters:
```
# 100ms
INTERVAL_OF_SCANNING = 0x01
# 0x32 - enable 5 second delay
DELAY_OF_SAME_BARCODES = 128 + 0x32
```
Animations can be played with https://github.com/gorazdko/specter-desktop/tree/qr_animation
  *  `def qr_animate(qrtext, max_len=100):` max_len determines the maximum chunk size of the
frame
  *  interval of animating can be set in `window.setInterval(function(){full_psbt.qra_next()}, 250)`
  * qrs are set for high recovery level (30%): `QRCode.CorrectLevel.H`

## Results

Its feasible, though results arent nearly as good as in https://divan.dev/posts/animatedqr/ where 

> The best result was 1.4 secs, which is almost 9KB/s! This result has been recorded at a rate of 11 frames per second and chunk size of 850 bytes with Medium recovery level

Here, one setting that works ok is 250 ms interval with 100 byte frames. For optimal settings one should play with parameters

Should we prefer this method over the one with buttons?

stepansnigirev, 2020-04-08:
Sounds amazing! I think it's the right way to go. So awesome what you are doing!

I tried it out, specter-desktop part works great, but I wasn't able to scan it with specter-diy. Not sure why yet, I need to look into your code and understand the logic. Looks like it doesn't even try to scan multiple codes, not sure why.

I will get back to it later tonight and look carefully.

stepansnigirev, 2020-04-08:
Ok I was stupid... Frozen firmware ignored python files )))

Works like a charm!!! Beeps a few times and imports the wallet - very nice feeling! I like it!

gorazdko, 2020-04-08:
> I tried it out, specter-desktop part works great

the qrcode library is not ok. it has a bug:
https://github.com/davidshimjs/qrcodejs/issues/220

You will notice it when changing `max_len` when having QRCode.CorrectLevel.H

> I need to look into your code and understand the logic

i will write a couple of sentences later about how this works, the limitations and other things i noticed...

gorazdko, 2020-04-09:
## How it works

* The barcode scanner works in continuous mode with INTERVAL_OF_SCANNING 100ms. This means after it obtains a scan it waits 100ms and starts a new scan. The scan itself usually lasts much longer - the more data it encodes, the longer it lasts. No external trigger is necessary to initiate a scan. We stop scanning by changing the scan mode to command mode.

* DELAY_OF_SAME_BARCODES of 5 seconds means scanning the same barcode again (and sending it over uart) can happen only when the interval elapses. The interval either times out or resets if we scan a different qr code.

## Example: 
interval of animating 300ms, 200byte frames , qrCorrectionLevel=L
  * works fluently if scanner is correctly and firmly pointed towards the qrcode.
      * 1/2 multisig Full PSBT  is 6 frames long and is scanned in 6*300ms
           * for me it usually takes 7 or 8 beeps
  * if we make the code larger (either by payload or by correction level) the scanner cant keep up and has to try scanning additional rounds. To keep up with the tempo the animating speed needs to slow down

## Downsides:
 If our hand is not still and PSBT transaction is very large we can wait several rounds (and one round can get big, perhaps even 20 frames) until all the frames are scanned. Bad UX.
Also, if the scanner is not focused accurately on the qr code, the scanning speed can slow down dramatically.

## Deployment
I can make this into a PR if we decide we want this. It still has some bugs that i need to fix (e.g. buffer overflow when scanning multiple rounds etc.) and do some code cleanup. But it would also break the support for the other scanner (which i dont have) which specter diy uses.

Regarding specter desktop, i can find a better frontend qrcode library and push the changes without buttons. The current backend qr library is not suitable as the qr code animation is a frontend process.

stepansnigirev, 2020-04-12:
I think we should use animated QR codes only when it is super necessary - rare transactions with many inputs and outputs, very large multisig wallets. 

With current PSBT optimizations even a large multisig fits in one QR codes if there are not too many inputs and outputs. And if people want to make a huge transaction, they should be ready to hold their wallet still for quite some time )))

Let's make it into a pull request.
For specter-desktop I suggest to add a checkbox or button `animate QR code`. So normally QR code is large and shows full data, but if you check the checkbox it becomes smaller and animated.

PR #58 (2020-04-19): qrscanner: scan qr animations automatically.
## Abstract

Close #56 

After correcting some bugs, the scanning is now faster and more fluent.

## Status

* ~ready for review~
* does not support the *expensive* scanner
* ~wip: add pin trigger mode~
* ready for review
closed 2020-04-19 by stepansnigirev