cryptoadvance/specter-diy #60: Buttons not working in simulator when running telnet

gorazdko, 2020-05-18:


stepansnigirev, 2020-07-04:
Should be resolved now. Can you check?

gorazdko, 2020-07-08:
No. It doesnt work at all. I tried on port 22849 with a random string, then had to press enter several times to get an error. Simulator gets laggy and freezes.

Maybe show in terminal which port is for what.

Also I have noticed the the gap limit warning in a multisig wallet gets covered by buttons.

I know you put the buttons slightly into a qr padding field to make them more noticeable, but to me it doesnt look good.

stepansnigirev, 2020-07-09:
Ok, I added some print statements for simulator regarding telnet ports here: https://github.com/cryptoadvance/specter-diy/pull/67

So 22849 should be for QR code communication and it should throw away anything you send there unless you are scanning a QR code.
Can you check what telnet sends to the micropython when you run it?

When I run `micropython_unix`, create a uart there and read something that I've sent over telnet I get the following:
```py
>>> from pyb import 
>>> import pyb
>>> uart = pyb.UART("YA")
Running TCP-UART on 127.0.0.1 port 22849 - connect with telnet
# sending "qwe" over telnet in a different tab...
>>> uart.read()
b'qwe\r\n'
```

So in my case EOL is `b'\r\n'`. I wonder what is it in ubuntu... Unfortunately, I will be able to test it only in the evening, so would be nice if you could try and tell me how it behaves on your machine.

Note: gonna fix gap limits warning, QR buttons and padding now.

stepansnigirev, 2020-07-09:
Fixed QR padding and gap limit here: https://github.com/cryptoadvance/specter-diy/pull/68

gorazdko, 2020-07-09:
> So in my case EOL is b'\r\n'. I wonder what is it in ubuntu

It's the same.

As soon as I make a telnet connection the simulator becomes unresponsive. No buttons are clickable.

stepansnigirev, 2020-07-09:
I was able to reproduce it on Ubuntu in VM, gonna try to figure out what's the problem...

stepansnigirev, 2020-07-09:
I updated f469 submodule, added `self.client.setblocking(False)` for every client, should work now. Can you check?
It's a bit weird because I am setting non-blocking mode in the constructor. Looks like on Ubuntu this is ignored...

https://github.com/diybitcoinhardware/f469-disco/commit/e642ea86dced58983075fe17db7b1d23bb10aa47

gorazdko, 2020-07-09:
It's good now.

closed 2020-07-09 by stepansnigirev