cryptoadvance/specter-diy #72: Enhancement: make a bip85 app for deterministic entropy

stepansnigirev, 2020-07-14:
We have apps structure now, so the app could implement deterministic entropy derivation according to [BIP-85](https://github.com/bitcoin/bips/blob/master/bip-0085.mediawiki).
App has access to the keystore, so it could derive new seeds or WIFs or whatever from the root key of the keystore and show it on the screen. Completely stateless, no data storage, no communication with the host - only a GUI menu.

xavierfiechter, 2021-03-12:
https://github.com/ethankosakovsky/bip85

stepansnigirev, 2022-11-20:
Added implementation to embit, except RSA. Gonna implement in DIY soon-ish:
https://github.com/diybitcoinhardware/embit/pull/35

jdlcdl, 2023-01-10:
I've left a comment at the following url. https://github.com/diybitcoinhardware/embit/commit/41f0a38f83f9df77bee7e0ddece4b32e3d935879

stepansnigirev, 2023-12-10:
implemented in #265

closed 2023-12-10 by stepansnigirev