cryptoadvance/specter-diy #75: Backorder of dev board - Alternatives?

bitcoinheiro, 2020-07-23:
Hi, the recommended dev board is out of stock now.
I've been researching a bit, and wanted to check out if these other boards might be compatible:
STM32F746G-DISCO
STM32F429I-DISC1
STM32H745I-DISCO
STM32H750B-DK

Or better to wait for the other one (should be back in stock around August)

Thanks

stepansnigirev, 2020-07-23:
I wouldn't recommend another board yet. Even though it's not very complicated to port Specter to another board we didn't really work on that yet. We may add support for [32f769idiscovery](https://www.st.com/en/evaluation-tools/32f769idiscovery.html) at some point - it's like an upgraded version of f469, but it is pretty bulky.
Better wait until it's in stock again. 
- Looks like [farnell](https://de.farnell.com/stmicroelectronics/stm32f469i-disco/entwicklungsboard-discovery-arduino/dp/2500227?gclid=Cj0KCQjw6uT4BRD5ARIsADwJQ1-q-GdXnyVbhgmz7R4sh66aUW6UM36_LQNNNOO00TY2Pff1xInBR78aAu2NEALw_wcB&mckv=sQZUsVyHm_dc|pcrid|269443712295|kword|stm32f469i-disco|match|p|plid||slid||product||pgrid|57694363670|ptaid|aud-516979316352:kwd-323765684298|&CMP=KNC-GDE-GEN-SKU-MDC-Test111A-Category-Bidding) still has a few boards.
- On mouser it's out of stock
- On [digikey](https://www.digikey.com/products/en?keywords=f469i-disco) I didn't find information about availability :(
- I also found these boards on aliexpress: [one](https://www.aliexpress.com/item/33012502773.html?spm=a2g0o.productlist.0.0.6b46252cdq96ak&algo_pvid=4cd17e99-a91e-4edf-b8c5-f5f46cc67bd6&algo_expid=4cd17e99-a91e-4edf-b8c5-f5f46cc67bd6-0&btsid=0b0a0ac215955457466537491ed2ab&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) [two](https://www.aliexpress.com/item/4000711361555.html?spm=a2g0o.productlist.0.0.6b46252cdq96ak&algo_pvid=4cd17e99-a91e-4edf-b8c5-f5f46cc67bd6&algo_expid=4cd17e99-a91e-4edf-b8c5-f5f46cc67bd6-1&btsid=0b0a0ac215955457466537491ed2ab&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_). They are a bit more expensive and delivery time is also pretty large.
- [chip1stop](https://www.chip1stop.com/RUS/en/view/dispDetail/DispDetail?partId=STMI-0102233&utm_medium=cpc&utm_source=google&utm_campaign=Google_EN_SKU_2019%E8%87%AA%E7%A4%BE%E5%9C%A8%E5%BA%AB_%E6%B5%B7%E5%A4%96&utm_term=STMI-01022331722596612&gclid=Cj0KCQjw6uT4BRD5ARIsADwJQ18OmU410RRyH9DoRD76PkLfBYnRTjUh5Dee8r9VA9qjQXyPFTJKpNIaAo6hEALw_wcB) has a few.

We also plan manufacturing these boards ourselves, but it kinda breaks no-supply-chain risk assumption.

bitcoinheiro, 2020-07-24:
Got it!

And the barcode scanner? CM65 QR & BARCODE SCANNER MODULE would work?
Seems like it would work: https://raw.githubusercontent.com/DFRobot/DFResources/master/Others/DFR0660%20Datesheet.pdf

stepansnigirev, 2020-07-26:
I think so. Waveshare also uses QR scanning module CM65-S from Grow. Otherwise there is always a trigger mode as well.

bitcoinheiro, 2020-09-08:
Went for the wavescanner at the end. Took some time, but got the board and the scanner. Looking good! 🥇 

[image]

closed 2020-09-08 by bitcoinheiro

stepansnigirev, 2020-09-08:
If you don't mind I will add this pic to the [gallery](https://github.com/cryptoadvance/specter-diy/tree/master/docs/pictures/gallery) =)

bitcoinheiro, 2020-09-08:
Sure, no problem! :)

bitcoinheiro, 2020-09-10:
Just saw a very cool 1/2d scanner, had to post it here. Price is good. I believe it has all necessary connections.
https://m5stack.com/products/atom-2d-1d-barcode-scanner-kit

stepansnigirev, 2020-09-18:
Looks like a very interesting alternative to the waveshare scanner.
Looking at the docs I don't see a way to control the scanner with commands over serial, so probably it can be used only in trigger mode, but if the trigger is connected to D5 pin of the board it should just work.