cryptoadvance/specter-diy #79: Add user (active) input to seed creation process

bitcoinheiro, 2020-09-08:
I understand that the seed is generated automagically using user input/interaction with the board before the seed generation button is pressed. Assuming every user has a different PIN and different handling, this should be enough from the technical perspective.

Anyway, it would be nice to be able to be more active on the seed creation process by adding entropy (and even seeing feedback of how that entropy affects the seed?)

I find the Specter DIY quite nice for this process, as a secure seed generator, a dedicated Ian Coleman BIP-39 tool: https://iancoleman.io/bip39/
[image]

Openoms also referred to another project that is using the raspi for this: https://www.rudefox.io/burrow/

He puts a great point here for adding this feature, it really makes the difference for people that don't trust hardware: https://www.rudefox.io/blog/2020-07-16-show-my-work.html

He even adds a sheet so people can verify if their dice entropy is actually being used for the seed creation:
https://www.rudefox.io/custody/walkthrough/create-seed/lookup-tables.pdf

[image]

Would be great to see something like this in the Specter DIY

ChristopherA, 2020-09-08:
We are looking into doing this in LetheKit (we already support dice input that is compatible with Ian Coleman’s). If there is interest in supporting this then we should consider working on making all of these approaches of “deterministic randomness” a security reviewable standard.

bitcoinheiro, 2020-10-19:
I believe this issue has been closed with the new checksum fix feature. No need to use the random number generator at all. Great feature BTW, love it! Cheers

closed 2020-10-19 by bitcoinheiro