cryptoadvance/specter-diy #81: [Feature request] Import aezeed format to control the onchain part of the LND wallet

openoms, 2020-09-08:
LND uses the aezeed seed format for it's wallet: https://github.com/lightningnetwork/lnd/tree/master/aezeed
The wallet holds two balances:
* offchain / channel balance - 2of2 multisig according to the Lightning specification - this is not dealt with in this issue
* onchain - addresses derived only from the LND wallet seed (aezeed) which can be converted to a BIP32 Root Key. Generating the matching Master Public keys for native and nested SegWit addresses allows coin control with SpecterDesktop or Electrum: https://github.com/openoms/lightning-node-management/blob/master/RestoreLNDonchainFundsInElectrum.md#import-the-onchain-part-of-the-lnd-wallet-to-electrum-wallet

Currently the conversion is best done with:
https://guggero.github.io/cryptography-toolkit/#!/aezeed or 
https://github.com/guggero/chantools

Request:
* An option to import the aezeed format 24 word seed + passphrase to SpecterDIY
* display the Master Public keys for the accounts on:
`m/84'/0'/0'/0` (p2wkh)
`m/49'/0'/0'/0`  (np2wkh)

openoms, 2020-10-08:
Related issue in LND: https://github.com/lightningnetwork/lnd/issues/4678

And about extracting xpub from C-lighning underway: https://github.com/ElementsProject/lightning/issues/4110

schnuartz-ai, 2026-08-29:
**Current assessment:** Feature request, not implemented. Still feasible but no longer the recommended approach given the current architecture — see below.

**What I investigated:**
- `src/keystore/ram.py:54-72` `set_mnemonic` — the sole seed-import path; it only accepts BIP39 (`bip39.mnemonic_to_seed` at line 65, validated by `bip39.mnemonic_is_valid` at line 63).
- `src/specter.py:255-277` (`initmenu`) and `src/specter.py:299-323` (`import_mnemonic`) — import accepts digital/binary/text mnemonics, all BIP39-validated.
- `src/gui/screens/mnemonic.py` — `RecoverMnemonicScreen` is wired to `bip39.WORDLIST` / `bip39.find_candidates`.
- `src/apps/bip85.py` — closest existing analog for a "derive child keys + display xpubs + optionally load" app; uses `embit.bip85` and `cross_app_communicate(... "set_mnemonic " ...)`.
- embit (both the build venv and the bundled `f469-disco/libs/common/embit/`): no `aezeed`/`cipher_seed`/`lnd` module exists (grep returned nothing). Only bip39/slip39/bip32/bip85 ship.
- Git history: no commit references aezeed/lnd/lightning.
- Related issues: #276 (open, NFC), #31/#57 (closed).

**Current code:**
- Seed → root: `ram.py:65-67` `seed = bip39.mnemonic_to_seed(...); self.root = bip32.HDKey.from_seed(seed)`.
- The requested onchain xpubs (`m/84'/0'/0'`, `m/49'/0'/0'`) are already derivable via `keystore.get_xpub(derivation)` (`ram.py:117`, used by `manager.py:519-525` and the xpubs app). So display is trivial once a correct root key exists.

**Reproduction:** n/a (feature absent). Importing an aezeed phrase today would fail `bip39.mnemonic_is_valid` and raise `KeyStoreError("Invalid mnemonic")` at `ram.py:64`.

**Related issues/PRs:** #276 (open). No aezeed PRs.

**Implementation notes (if pursued):**
1. New `src/keystore/aezeed.py`: aezeed 2048-word list, word→11-bit unpack, XChaCha20-Poly1305 AEAD decrypt with passphrase-derived key, and the aezeed root-key derivation (NOT BIP39 PBKDF2 — reusing `bip39.mnemonic_to_seed` would produce a wrong root and silently lose funds).
2. Add an "Import aezeed (LND)" branch in `src/specter.py:import_mnemonic` (299-323) and `initmenu` (255-277), using `RecoverMnemonicScreen` with the aezeed wordlist and a new `keystore.set_aezeed(mnemonic, passphrase)` that sets `self.root` directly.
3. Reuse the existing `xpubs` app / `showaddr` (`manager.py:447`) to display the `m/84'/0'/0'` and `m/49'/0'/0'` xpubs.
4. Unit-test against published lnd/chantools aezeed→xprv vectors before shipping.

**Security review:** The critical correctness risk is the AEAD + root derivation: a buggy codec can silently derive the wrong root key → permanent loss of the onchain funds. Passphrase entry should reuse the existing secure PIN-keyboard path. RNG is unaffected. Given MCU constraints and the availability of off-device conversion (chantools, per the issue body), the lower-risk path is to keep aezeed→entropy conversion off-device and import via the audited BIP39 flow.

**Recommendation:** Do not implement on-device aezeed as a first-class seed format; instead document the off-device aezeed→BIP39 conversion (chantools) + standard recovery-phrase import, which already satisfies the "control the onchain part of the LND wallet" goal with the existing, audited code path.