cryptoadvance/specter-diy #82: Scanner doesn't start if pin is entered too fast

stepansnigirev, 2020-09-22:
We initialize and configure the QR code scanner after the pin screen.
It happens sometimes that at this moment the scanner is not fully booted yet, and then configuration silently fails.
We need to have a fixed minimal time delay before we start initializing the scanner. It can be done via asyncio task

Schnuartz, 2026-08-29:
I think that this bug only appears in the old Specter DIY with the waveshare Boards. 

And that it is not appearing in the Shield Versions.

schnuartz-ai, 2026-08-29:
**Current assessment:** Still relevant. The requested pre-init settling delay is not implemented — only an *after-init* delay exists. Hardware verification not performed, but the race is clearly present in the code path.

**What I investigated:**
- `src/hosts/core.py:130-139` `Host.enable` — calls `self.init()` first, then `await asyncio.sleep_ms(self.RECOVERY_TIME)` after. No delay before `init()`.
- `src/hosts/qr.py:124` `RECOVERY_TIME = 30` (comment: "time to wait after init"). `qr.py:517-543` `QRHost.init` immediately probes via `_update_scanner_model()` and `configure()`; on failure it silently falls back to PIN-trigger mode and lights LED3 (line 538-543) — this is the "configuration silently fails" behaviour named in the issue.
- `src/specter.py:175-191` `setup()` calls `unlock()` (PIN entry) then `main()`; `specter.py:57` sets `current_menu = self.initmenu`; `specter.py:243-247` `initmenu` runs `await host.enable()` for passive hosts — i.e. the scanner is configured right after the PIN is entered, matching the report.
- `qr.py:741-749` `_restart_scanner` ("fix scanner race condition") was added in #335 (commit `97ab4d4`, Dec 2025), but that race is between async `_restart_scanner` and `stop_scanning`, not the boot-settling race described here.
- `qr.py:278-308` `get_setting`/`set_setting` have retries (3 × 100 ms), so there is some resilience, but no explicit "wait for the scanner to be ready after power-on" before the first probe.
- Git history: `aa84f05` introduced `RECOVERY_TIME`; it has always been an after-init delay.
- Related: #126 (open, same "scanner doesn't initialize" symptom), #134, #273, #355, #314 (closed via #335).

**Current code:**
- `src/hosts/core.py:130-139`: `enable()` calls `init()` then `await asyncio.sleep_ms(self.RECOVERY_TIME)` after; no delay before init.
- `src/hosts/qr.py:517-543`: `init` probes + configures immediately; silent PIN-trigger fallback + LED3 on failure.

**Reproduction:** Not reproduced — hardware-only (Waveshare board + physical scanner + fast PIN entry). The race is evident from the code: no pre-init settling delay, and `enable()` is invoked immediately after PIN unlock.

**Related issues/PRs:** #126 (open), #314 (closed, #335 — different race), #134, #273, #355.

**Implementation notes:**
1. Add `BOOT_SETTLE_MS = 200` to `QRHost` and, in `init()` (`qr.py:517`), `time.sleep_ms(self.BOOT_SETTLE_MS)` before `_update_scanner_model()` — guarded by a new `_scanner_settled` flag so it only fires on first boot (not on reconfigures from the settings menu).
2. Alternatively, add a `BEFORE_INIT_TIME` sleep in `Host.enable` (`core.py:130`) guarded by `not self.initialized`, to benefit all hosts.
3. Consider surfacing the PIN-trigger fallback (qr.py:538-543) to the user instead of failing silently, which also addresses the "silently fails" wording of the issue.

**Security review:** No security impact. Scanner is an input peripheral; degradation is to PIN-trigger mode or no scanning. No effect on keys, signing, or RNG.

**Recommendation:** Add the small first-boot pre-init settling delay (option 1) and surface the fallback to the user; this is cheap, safe, and also helps #126.