cryptoadvance/specter-diy #85: Calculate 12th and 24th word in the specter DIY

lunaticoin, 2020-10-07:
Hi all. I have been playing with SpecterDIY and I really like how easy you can have your airgapped QR code HWW device running. 

The potential of the DIY is crazy and I would like to share a feature with you that I think would make it even more amazing. 
I like to create the seeds on my own using a coin and after the whole airgapped&manual process of calculating the 11 digits binari, translating it to decimal and finding the corresponding BIP39 words, I am not very happy having to jump to an 'offline' computer and using seedpicker to calculate the 24th word. I think a better way would be to use the Specter DIY to calculate either the 12th word or the 24th (on seedpicker you can only get the 24th). 

As you already do ECC in it, I suppose could be an easy (and superpowerful) feature to add. 
How does it sound?

stepansnigirev, 2020-10-08:
This sounds like a good idea :)
I was thinking about a way to verifiably include user entropy to the recovery phrase but didn't find a way to make a nice UX yet.
An easy fix would be to simply add a button "Fix checksum" that would appear when you enter 12 or 24 words.
When you click on the button it would replace the last word with the one that makes a valid recovery phrase.

Alternatively, we can add a checksum-fixing-word into autosuggestions. So when you are at the word 12 or 24 you would have normal autosuggestion plus one extra word that would fix the checksum. I would use a different background color, border or some other style to mark the words that make the valid mnemonic.

lunaticoin, 2020-10-09:
Sounds like a great way of doing it. If the user is already introducing the right word then fix checksum button wouldnt change anything to that user, as it is already good. 

The autocomplete could be confusing if the user want to introduce 24 words and then finds out you are sugesting a different '12th' from what he has but I think is a matter of trying and seeing.

But yay! The idea feels already like 🔝
Cant wait to test it when possible!

ChristopherA, 2020-10-09:
We are also considering doing this with #LetheKit (allow the last checksum word be calculate by the device), but for clarity, as I believe that there can be multiple words that are valid checksums, we need to be clear in a spec on which one to use. If you'd like to chat more about this issue, let's talk about it in the Airgapped Wallet Community discussions at https://github.com/BlockchainCommons/Airgapped-Wallet-Community/discussions and write a spec there, and maybe submit it as a BIP.

/cc @gorazdko

stepansnigirev, 2020-10-17:
@ChristopherA I think the correct way is to convert invalid mnemonic to entropy ignoring checksum validation, and replacing the checksum with the correct one. This method requires all words to be provided, is deterministic (only one last word is correct), and keeps maximum entropy from the user input.

lunaticoin, 2020-10-17:
Interesting. How would the wrong word be converted to entropy? Have you checked maybe how does seedpicker do it? I think it just uses three 0 bits to begin the 24th word

Cant wait to see this feature implemented on Specter

closed 2020-10-19 by stepansnigirev

stepansnigirev, 2020-10-19:
I made a pre-release that includes this feature, you can try it out:
https://github.com/cryptoadvance/specter-diy/releases/tag/v1.4.0-pre1

stepansnigirev, 2020-10-19:
Seedpicker is asking for 23 words, and the last word is probably zeroes + checksum.
For 24-word mnemonic it means you lose 3 bits of entropy, but if you would do the same for 12 word mnemonic it loses 7 bits.

#86 uses all words and only changing the last one with correct checksum.
This means the algorithm is deterministic and keeps as much entropy as it can.

lunaticoin, 2020-10-19:
Just tested the pre-release and LOVE it! 
I tried for 12, 18 and 24 words and it's great. Really like how it changes the result depending on which last word you input. 
Thank you for such an amazing feature. 
Don't trust, verify Hero Level.

PR #86 (2020-10-19): allow fixing invalid mnemonic
Fix #85 

Works on 12, 18 and 24 words

[image]