cryptoadvance/specter-diy #89: Switch to ur:crypto-psbt format

stepansnigirev, 2020-10-21:
https://github.com/cryptoadvance/specter-desktop/issues/513

mflaxman, 2021-11-07:
What's the status of this? I've been playing around with the bcur v2 spec and I can't say I understand why this is a useful upgrade. The code is an order of magnitude more complex, and intuitively I dislike that the QRs are not constant (as you iterate beyond `n` QRs they don't just repeat).

I noticed that in `specter-desktop` you used some unknown/minified js code to render the qr GIFs locally:
https://github.com/cryptoadvance/specter-desktop/pull/1237/files#diff-23225b8d95675fa8221e5d8719eefc5d87daba64df2a1a7400a0c503a3ef1789

stepansnigirev, 2021-11-15:
I understand why you don't like this standard - it is not human-readable and hard to implement. There is one thing that makes this standard valuable - ability to scan O(N) QR codes if you've missed one, so for large number of QR codes you don't have to scan over the whole sequence if one or two was missed. It provides better user experience for large multisig transactions, and for Liquid as well, but for most cases tx can fit in 1-5 QR codes.
We added support for it in Specter-Desktop because Keystone and Passport are using it (the minified library comes from Keystone).
On the Specter-DIY side, I will have to add support for it in order to maintain compatibility with other software wallets like Blue Wallet, Sparrow, and others. But in order to do that right, I need to properly implement that so we don't run into memory issues with the infinite flow of QR codes.
For communication between Specter-Desktop and Specter-DIY I want to keep using a simple `pMofN blob` format though.

PR #208 (2022-07-10): ur:crypto-psbt support
Modern ur:crypto-psbt standard support for QR codes. Experimental.
Close #89.
I had to write it from scratch as implementation from foundation devices uses too much memory.
Tested with small to mid-size transactions (up to 10 inputs) - works in most cases.
Sometimes it fails on large txs for mysterious reasons - decoder probably because of uart buffer overwrite, not sure why encoder fails. I need to investigate further.
Anyways, for most common txs it works, so makes sense to gather information when it fails and under what conditions.
closed 2022-07-10 by stepansnigirev