cryptoadvance/specter-diy #109: [Feature request] Create mnemonic using external entropy

valerio-vaccaro, 2020-12-02:
In order to create a more safe mnemonic specter-diy can use an external entropy source (like dice or playing cards) and create a mnemonic based on this data.

bitcoinheiro, 2020-12-02:
It's possible already. Just not with a wizard. But you can add your own external entropy when creating the seed

stepansnigirev, 2020-12-04:
In the "generate new key" menu you can click on any word and edit individual bits there.
The idea is that you can XOR hardware entropy with your coin-generated entropy, but you can also ignore hardware entropy completely and enter the coin flips directly.
See how it works here: https://twitter.com/StepanSnigirev/status/1318912100958412804?s=20

Another option is to pre-generate the recovery phrase by picking the words from bip39 dictionary and then fix it to be a valid bip39 mnemonic. In the "Enter recovery phrase" you can enter any recovery phrase and then the wallet will suggest you to fix the last word to make it valid.

closed 2020-12-04 by stepansnigirev