cryptoadvance/specter-diy #115: Crashes during Signing -  Memory allocation failed

flyingfaders, 2020-12-28:
When attempting to sign a txn via QR codes Specter-DIY crashes and displays the following error.

[image]

Setup:
3 of 5
Tnx: 3 inputs, 1 output (to self, consolidating) 

I didn't test using USB(HWI) as I'm trying to maintain an airgap but I was able to sign the transaction via USB using Screen  on an airgapped computer.

stepansnigirev, 2020-12-28:
Thanks for reporting!
I'll work on some optimizations for the next release.

QR code display implementation is not optimal at the moment, this can cause memory allocation errors when processing large transactions. Also, PSBT transaction is stored in MCU memory, but some parts of it can be moved to SDRAM.

stepansnigirev, 2021-01-14:
Should be fixed with https://github.com/cryptoadvance/specter-diy/commit/a5477ee2c08076eeff160b371433d578772bdab6
It will be available in the next release (next week)

closed 2021-01-15 by stepansnigirev

flyingfaders, 2021-01-26:
Works perfectly on the latest release. Thanks!