cryptoadvance/specter-diy #117: seed import fails after upgrade

hoverlord, 2020-12-30:
I generated a 24 word seed with specter diy in an older version. now i upgraded to 1.4.2 and tried to enter the seed phrase, but it won´t take the last word like it was and offers me to correct it to a different word. when i accept the corrected last word i can safe the seed phrase but specter desktop is not showing my funds for that key. is there a way to downgrade to 1.3.0 or anything else i can do to recover that exact seed phrase?

bitcoinheiro, 2020-12-30:
What if you don't accept to correct the last word and just continue?

hoverlord, 2020-12-30:
Thanks for the quick reply! I can´t continue without the last word being fixed, the "continue" button is greyed out.

stepansnigirev, 2020-12-30:
Are you sure the recovery phrase is correct? Maybe you did a typo there?
There is a way to downgrade to previous versions, but Specter DIY shouldn't generate invalid recovery phrase, we didn't change this part of the code.
I'll write a doc how to reset the device and disable the bootloader when I get back home.

hoverlord, 2020-12-30:
That´s great, thanks a lot! I tried to recover the original as well as the corrected seed phrase in blockstream green, which rejects both as wrong.

hoverlord, 2021-01-01:
Problem solved. It turns out I did write down two 12-word seeds which I can import perfectly, but mistakenly took them as one 24-word seed. Feeling quite stupid :)
Thanks for your help!

closed 2021-01-01 by hoverlord