cryptoadvance/specter-diy #118: device frozen after entering incorrect 24th mnemonic word

ashjkl159, 2021-01-07:
I created 24 word mnemonic on specter diy, wrote down, not saved to smart card, switched off, then switched on and started entering recovery phrase, 24th word I intentionally wrote wrong partially, i.e. say first 2 letters, then 'fix' push button appears next to it, i press 'fix' button before whole 24th word is out and device gets frozen, so after while i switched it off ... then switch it on and  got message critical error something went terribly wrong, device will be wiped
 
after i wiped device i started again and after entering recovery phrase, when i entered the whole 24th word wrong and then press 'fix' button then it all worked fine, so the problem appears only when you enter couple of wrong letters of the 24th word.
 
so if you can make it more idiot proof :-)

cheers

stepansnigirev, 2021-01-08:
Should be fixed in the next release with https://github.com/cryptoadvance/specter-diy/commit/463ea7aac15fd82a58ea87f6bfb349dab064696c

closed 2021-01-15 by stepansnigirev

ashjkl159, 2021-01-16:
CheersSent from ProtonMail Mobile

Closed #118.