cryptoadvance/specter-diy #120: cosigner XPUB validation

djpnewton, 2021-01-18:
Does specter DIY support cosigner XPUB validation?

I am trying to gather information for this: https://github.com/cryptoadvance/specter-desktop/pull/885

stepansnigirev, 2021-01-18:
When you import the wallet to DIY it asks you to confirm - shows policy, all xpubs and marks his xpub.
The flow is similar to coldcard and Cobo.
Recently we also added a toggle switch between xpubs and slip132 Zpubs, so you can double-check with other hardware wallets that cosigners are the same.

If the wallet is not imported it's still possible to sign, but DIY will not verify the change addresses.

[image]

djpnewton, 2021-01-18:
Thank you very much.

So I just ran the simulator to double check a couple of things but it looks like it can:

 - Display own xpub on demand                                - *yes*
 - Display cosigner xpubs                                          - *yes* (on wallet import)
 - Show Electrum `Zpub...` and `Ypub...` formats   - *yes*
 - Register xpubs inside the device                           - *yes* (on wallet import)

closed 2021-01-18 by djpnewton