cryptoadvance/specter-diy #126: After loading key from internal flash "Generating keys" not displayed, scanner doesn't initialize

dimaatmelodromru, 2021-03-03:
Sometimes

stepansnigirev, 2021-03-04:
I can confirm the issue. Happens to me sometimes as well, but I didn't figure out why.

I was thinking about adding an option to reinitialise the scanner from the settings, it could help at least to fix the scanner without reloading the whole device.

schnuartz-ai, 2026-08-29:
**Current assessment:** UNCERTAIN — likely a combination of two intermittent issues, neither fully resolved. The "scanner doesn't initialize" symptom matches the boot-time scanner race documented in #82 (no pre-init settling delay), which the recent scanner PR #335 did **not** fix (it addressed a different race and a first-boot-only factory reset). The "'Generating Keys' not displayed" symptom points at a synchronous loader followed immediately by a blocking key-derivation with no yield to the event loop. Both are intermittent ("Sometimes"), consistent with the report. Hardware verification not performed; this assessment is from code inspection.

**What I investigated:**
- The load-key-from-flash flow: `src/specter.py:243-297` `initmenu` shows the menu; on `menuitem == 2` it calls `res = await self.keystore.load_mnemonic()` (`specter.py:280`), then `self.init_apps()` (`specter.py:284`) and returns `self.mainmenu` (`specter.py:285`). `FlashKeyStore.load_mnemonic` (`src/keystore/flash.py:254-268`) optionally calls `select_file` then `self.set_mnemonic(data.decode(), "")` (`flash.py:267`).
- Where "Generating keys..." is shown: `RAMKeyStore.set_mnemonic` (`src/keystore/ram.py:54-72`) calls `self.show_loader(title="Generating keys...")` at `ram.py:59`, then **immediately** runs the blocking `seed = bip39.mnemonic_to_seed(self.mnemonic, password)` (`ram.py:65`, PBKDF2-HMAC-SHA512, 2048 iterations) and `bip32.HDKey.from_seed(seed)` (`ram.py:66`) with no `await`/`asyncio.sleep_ms` between the loader call and the heavy computation.
- How the loader renders: `gui.show_loader` (`src/gui/async_gui.py:33-37`) calls `self.scr.show_loader(text, title)`. `Screen.show_loader` (`src/gui/screens/screen.py:59-67`) creates a `Modal`, sets the text, and calls `update()` twice. `update()` is `display.update(dt)` (`src/gui/core.py:20-22`). So the loader text is pushed synchronously, but whether the physical display fully flushes before the blocking PBKDF2 monopolises the CPU depends on the display driver; the cooperative event loop (`AsyncGUI.update_loop`, `async_gui.py:116-119`) cannot run during the blocking derivation. This is the most plausible cause of "Generating Keys not displayed": the render is requested but the CPU is then blocked before/while the flush completes, and by the time control returns the loader is stale or already hidden.
- The scanner init path and the #82 race: `initmenu` enables passive hosts at `specter.py:244-247` (`if host.button: await host.enable()`). QR/SD/USB all have `button` set (`src/hosts/qr.py:126`, `sd.py:16`, `usb.py:17`), so the QR scanner is initialised here, **before** the user picks "Load key". `Host.enable` (`src/hosts/core.py:130-139`) calls `self.init()` first, then `await asyncio.sleep_ms(self.RECOVERY_TIME)` **after** — there is no delay before `init()`. `QRHost.init` (`src/hosts/qr.py:517-543`) immediately probes via `_update_scanner_model()` (`qr.py:522`) and `configure()` (`qr.py:534`); on failure it silently falls back to PIN-trigger mode and lights LED3 (`qr.py:538-543`) — exactly the "configuration silently fails" behaviour in #82.
- Whether recent scanner work fixed this: PR #335 (commit `97ab4d4`, "Add full M3Y support and fix QR scanning issues for GM65 and M3Y", Dec 2025) added `_restart_scanner` (`qr.py:741-749`, comment "fix scanner race condition") — but that race is between the async `_restart_scanner` and `stop_scanning` during **animated QR scanning**, not the boot-settling race. It also added `_factory_reset_scanner_on_boot` (`qr.py:611-619`) gated by `_boot_reset_pending` (`qr.py:149-159`), which is only true on the **first-ever boot** (when the marker file `.qr_factory_reset_done` is absent, `qr.py:152-155`); on every subsequent boot `_boot_reset_pending` is `False` and `init()` probes immediately with no settling delay. `get_setting`/`set_setting` (`qr.py:278-308`) retry 3×100 ms, which gives some resilience but not a power-on-ready wait. So the recurring boot race is not resolved.
- `QRHost.RECOVERY_TIME = 30` (`qr.py:124`, comment "time to wait after init") confirms the existing delay is post-init only.
- Git history: no commit addresses the "Generating keys" loader timing (`git log --grep "generating"` empty). The scanner history is `97ab4d4` (#335), `031d285` (#299), `d838664` (command mode). `src/hosts/core.py` `RECOVERY_TIME` has always been an after-init delay.
- Related issues: #82 (open, same scanner race, "Scanner doesn't start if pin is entered too fast"), #314 (closed via #335, different M3Y issue), #355 (open, GM65 SW 137/134 problems), #273 (open, scanner returns error from Sparrow), #134 (open, add QR scanner settings). A maintainer (stepansnigirev) commented on #126 on 2021-03-04 confirming the intermittent behaviour and suggesting a settings-menu reinit option — that now exists as the "Reset QR Module" button in `QRHost.settings_menu` (`qr.py:666-699`, `_factory_reset_scanner`), which is a mitigation but not a fix for the boot race.

**Current code:**
- `src/keystore/ram.py:54-66`:
  ```python
  def set_mnemonic(self, mnemonic=None, password=""):
      if mnemonic == self.mnemonic and password != "":
          self.show_loader()
      else:
          self.show_loader(title="Generating keys...")
      """Load mnemonic and password and create root key"""
      if mnemonic is not None:
          self.mnemonic = mnemonic.strip()
          ...
      seed = bip39.mnemonic_to_seed(self.mnemonic, password)
[... cut ...] ed as "not recommended for real funds" in `docs/security-model.md`; the loader issue is purely cosmetic/UX for that mode.

**Recommendation:** Keep open. Treat the scanner symptom as a duplicate-of/variant-of #82 and fix the boot-settling race there (which also resolves this report's scanner symptom); separately add a small yield/sleep in `set_mnemonic` (`ram.py:59-66`) so the "Generating keys..." loader actually renders before the blocking PBKDF2. Link #82 as the primary scanner-tracking issue.