cryptoadvance/specter-diy #129: Compilation error

matthgyver, 2021-03-20:
Hi, 

When I'm trying to compile from source just cloned from this repo, I got this error :
```
arm-none-eabi-ld: build-STM32F469DISC/scard/reader.o:(.rodata.scard_conn_params_type+0x0): multiple definition of `scard_conn_params_type'; build-STM32F469DISC/scard/scard.o:(.rodata.scard_conn_params_type+0x0): first defined here
arm-none-eabi-ld: build-STM32F469DISC/scard/connection.o:(.rodata.scard_conn_params_type+0x0): multiple definition of `scard_conn_params_type'; build-STM32F469DISC/scard/scard.o:(.rodata.scard_conn_params_type+0x0): first defined here
```

Have you an idea please ?

Thank's

miketlk, 2021-03-21:
Hi,
Could you please check the version of the toolchain you are using, running something like:

```shell
arm-none-eabi-gcc --version 
```

matthgyver, 2021-03-21:
Hi @miketlk 
My version is : ``arm-none-eabi-gcc (Arch Repository) 10.2.0``

miketlk, 2021-03-22:
Thanks, @matthgyver
It looks like GCC 10.x has a stricter treatment of the "external" keyword (or the lack of it) for constants.
 
Fixed in 65e719a1c758167d0309411d09b663d6d25112f9

closed 2021-03-22 by miketlk