cryptoadvance/specter-diy #132: PIN code font size can be bigger

kirillkovalenko, 2021-03-23:
It fills like PIN code numbers are too small comparing to the screen size.

stepansnigirev, 2021-03-25:
Good point. I slightly increased it in https://github.com/cryptoadvance/specter-diy/commit/1fb727ca414e5ecd117dcdda7b0ba5b046f42288, this is the largest available built-in font.

If we would want to increase it further we would need to add another font. I'll keep it for later when we'll work on the UX improvements.

kirillkovalenko, 2021-03-26:
> I slightly increased it in 1fb727c, this is the largest available built-in font.

I've tested 1.5 firmware and looks way better now.  Thank you.

> If we would want to increase it further we would need to add another font. 

@stepansnigirev  How difficult would it be to implement application-wide font size setting (including a new font) ?

stepansnigirev, 2021-03-26:
Not very difficult, there is a tool that can convert fonts from ttf to lvgl fonts:
https://lvgl.io/tools/fontconverter

And then we can add bindings to micropython to access these fonts here:
https://raw.githubusercontent.com/diybitcoinhardware/f469-disco/master/usermods/udisplay_f469/lv_mpy_example.c
(search for roboto_mono_16 to get an idea how it's done)

The system-wide style is set here, so the theme can be tweaked with minimal effords:
https://github.com/cryptoadvance/specter-diy/blob/master/src/gui/common.py#L16-L112
There are also a few things that are hard-coded in the screen classes that need to be moved to the theme instead.

Another thing I wanted to do for a very long time is to add icons for error messages and buttons. This can be done with some icon font.

I am postponing these things because I am thinking about migrating from lvgl v6 to lvgl v7. It has cleaner UX and much better styling opportunities. And also I would like to spend some time on redesign of the UX in general. Now we reached the point when we have a pretty solid architecture of the wallet, so it's clear what the UI needs to do and what widgets we commonly use.

If you have any suggestions on the UX improvements I'll be happy to hear.

P.S. Upgrade to v1.5.1 please, I broke the tx screen in v1.5.0 with the last-minute change...

schnuartz-ai, 2026-08-29:
**Current assessment:** Likely fixed for the specific PIN-font ask; the broader "application-wide font size setting" follow-up raised in the comments is tracked separately by #236.

**What I investigated:**
- The commit referenced by the maintainer: `1fb727c` "increase PIN font size" (2021-03-25), which only touches `src/gui/screens/input.py` (+6/-1).
- The current PIN screen: `PinScreen` at `src/gui/screens/input.py:208`, with the bip-39 password screen at `input.py:111`.
- The reporter confirmed in the thread that firmware 1.5 "looks way better now".
- The maintainer noted this is the largest built-in LVGL font and that going larger requires generating a custom font (lvgl font converter) and adding micropython bindings (the `f469-disco` usermods `lv_mpy_example.c` pattern, e.g. `roboto_mono_16`).
- Related: #236 (open, "Add a setting to customize fonts size") is the generalised version of the follow-up question.

**Current code:**
- `src/gui/screens/input.py:208` `PinScreen` uses the font increased by `1fb727c`. No further font-size change has landed since.
- The system style/theme is centralised in `src/gui/common.py:16-112` (as the maintainer described), which is where a future custom-font setting would plug in.

**Reproduction / validation:** Not reproduced on hardware. The relevant change is a single confirmed commit (`1fb727c`) that the reporter validated on firmware 1.5. The specific complaint ("PIN numbers too small") was addressed within the constraints of the built-in fonts.

**Related issues/PRs:** #236 (open — generalised font-size setting). PR `1fb727c`.

**Implementation notes:** The only remaining work is the generalised font-size setting (+ a new custom font), which belongs under #236 rather than this issue.

**Security review:** None — pure UI/font sizing.

**Recommendation:** This specific issue appears resolved by `1fb727c` (reporter-confirmed). I would consider closing it and letting #236 track the generalised custom-font-size feature.

Schnuartz, 2026-08-31:
Marking that as fixed. 

Even though there are intersting findings from Stepan regarding fonts and icons. I will add this information to [#236](https://github.com/cryptoadvance/specter-diy/issues/236)

closed 2026-08-31 by Schnuartz