cryptoadvance/specter-diy #139: DIY HSM

polto, 2021-04-01:
Not without tradeoffs, but Specter DIY could be turned into a DIY HSM module.

Similar to what coldcard does with https://ckbunker.com/ , users can create a policy allowing a USB connected Specter DIY to automatically sign PSBT matching a particular policy. 

https://coldcardwallet.com/docs/hsm/rules
https://coldcardwallet.com/docs/hsm/protocol

Such DIY HSM can also be part of a multisig.

openoms, 2021-04-03:
My dream is to have coinjoin transactions signed automtically by a HWW: https://github.com/JoinMarket-Org/joinmarket-clientserver/issues/767#issuecomment-746474249

schnuartz-ai, 2026-08-29:
**Current assessment:** Feature request — not recommended as a default capability, because automatic (unattended) PSBT signing directly conflicts with the device's core verify-every-transaction security model.

**What I investigated:**
- Whether any HSM / auto-sign / policy mechanism exists: `grep -rn "hsm\|HSM\|policy\|auto.*sign\|automatic.*sign" src/` found only UI "policy" labels (wallet policy strings in `src/apps/wallets/screens.py:27,129-148`) — no signing-policy engine. `git log --grep "hsm\|policy.*sign\|auto.*sign"` returned nothing.
- How signing works today: every PSBT goes through `confirm_transaction` → `confirm_wallets` → `confirm_transaction_final` (`src/apps/wallets/manager.py:323-371`) with explicit user confirmation on the device screen before `sign_psbtview` runs (`manager.py:318-320`). There is no code path that signs without user confirmation.
- USB host capability: `src/hosts/usb.py` is a USB-VCP CDC device-mode host for receiving commands/PSBTs and returning signed results (`process_command`, `usb.py:66-93`). It is a transport, not an always-on auto-signer. There is no policy rule engine (cf. Coldcard's `ckbunker` ruleset referenced in the issue).
- Security model: `docs/security-model.md` is built around the user verifying every transaction on the trusted display before signing. Unattended policy-based signing removes exactly this human-verification control.

**Current code:**
- Signing is gated behind `confirm_transaction` (`manager.py:323`); no bypass exists.
- No policy/rule structures, no auto-sign loop.

**Reproduction / validation:** n/a (feature absent). Confirmed by source search.

**Related issues/PRs:** #107 (open — Yubikey; also a different secure-element path), #138 (open — TOTP/2FA-for-signing, mentioned in #139's body), #235 (open — stateless mode, related to always-on use). The linked JoinMarket coinjoin auto-sign wish (in comments) is the same concept.

**Implementation notes (if pursued):**
1. An HSM mode is a large architectural addition: a policy DSL (allowed derivations, amount limits, whitelisted destinations, rate limits, number of co-signers), a persistent policy store, a USB daemon that signs matching PSBTs without the confirmation screens, and a secure way to enable/disable the mode.
2. It must be opt-in and ideally time-bounded / session-bounded, with the policy itself reviewed on-screen at enablement (reusing `confirm_transaction`-style UX). Otherwise it reduces the device to a hot key.
3. The safest variant keeps per-transaction confirmation but adds *batch* or *unattended-under-policy* as a separate, explicitly-dangerous mode gated behind a second unlock.

**Security review:** This is the central security trade-off: the device's value is that every signing operation is human-verified on a trusted display. Automatic policy-based signing converts it into a network-adjacent hot signer; a policy bug or a USB host compromise can then sign arbitrary transactions within the policy. This should not be implemented as a silent/default capability. If implemented, the policy engine itself needs rigorous testing and the threat model in `docs/security-model.md` must be rewritten.

**Recommendation:** Keep open as a discussion item, but do not implement unattended auto-signing as described — it undermines the device's primary security property. If pursued, scope it to a heavily-guarded, opt-in, session-bounded policy mode with on-screen policy review, and update the security model first.