cryptoadvance/specter-diy #140: Wallet not permanent

634-5789, 2021-04-06:
After initially installing via USB and generating a wallet, the device seems not to remember the wallet after a reboot (via reset button or powering off). Only key management (generate and recovery) and settings options are available. PIN code works, recovery is successful but the wallet is gone again after a reboot. Am I doing anything wrong?

Firmware 1.5.1

bitcoinheiro, 2021-04-06:
That's how the wallet works. It does not store the seed unless you store it
manually. Not recommended to store on the device tho, since it's not a
secure device. Specially if connected via USB

634-5789, 2021-04-06:
Thank you for your comment. I was not aware of this.

closed 2021-04-06 by 634-5789