cryptoadvance/specter-diy #170: Can't Export Signed Transaction via QR Code without Scanner?

mflaxman, 2021-08-10:
I love the Specter-DIY with a camera for QR code scanning and have built many of them, but I've also been playing around with Specter-DIY's without a camera. These have the benefit of not requiring any assembling and having very few parts to source, just grab the STM32F469I, a MicroSD card and a Mini-USB cable and you're good to go!

Without a camera, I'm successfully able to export a public key record from DIY, include it in a multisig wallet and then sign a transaction from that multisig wallet (once it has funds deposited). However, I'm doing all of it via SD card. It'd be nice after signing a transaction to be able to relay it back to Specter-Desktop/Sparrow via QR code using the DIY's screen. There's no reason to require a physical camera in order to display a QR code of a signed transaction. I can display the public key record via QR without a camera.

I know this is an unusual use-case, and having a camera for the QR workflow is strictly superior, but the lower the bar is for people to get DIY's the more likely they are to use them (and then want to upgrade them to full camera support).

Thanks!

stepansnigirev, 2021-08-11:
Makes sense, I'll do that.

stepansnigirev, 2021-08-28:
Added in the latest release

closed 2021-08-28 by stepansnigirev