cryptoadvance/specter-diy #186: import multisig wallet - "My key" not found - watch only

chri2, 2021-10-07:
I tried to import a multisig wallet per json and wondered why I always ended up with a watch only wallet.

Looking at the specter-diy master public key for multisig and one of the master public keys of the imported watch only wallet I found them to be the same.

Both start with `[aabbccdd/48h/0h/0h/2h]` followed by the same master public key.

The `aabbccdd` part differed in the both keys.

I didn't find reference to the format, but it seems that the `aabbccdd` part is kind of a serial number of the device the key has been generated on.

I edited my wallet.json to make that part match the part my specter-diy showed to me and imported the multisig wallet again. 

This time specter-diy recognized that it had already the private key for that one public key and saved the multisig wallet - not as 'watch only', but as I'd expected as a normal wallet I could sign with.

My question is whether it wouldn't be better to ignore the suspected 'serial number' `aabbccdd`-'serial number'-part since it wouldn't be an unusual use case to import a multisig wallet to a replacement device already holding the correct private key. A replacement device would in most cases have a different `aabbccdd`-'serial number'-part I suspect.

stepansnigirev, 2021-10-10:
Check out key expressions in descriptors (bip-380):
https://github.com/bitcoin/bips/blob/master/bip-0380.mediawiki#key-expressions

Fingerprint is the first 4 bytes of hash-160 of the root public key in hex.
For example if you have a root key from mnemonic `abandon*11 about`, the root xprv is `xprv9s21ZrQH143K3GJpoapnV8SFfukcVBSfeCficPSGfubmSFDxo1kuHnLisriDvSnRRuL2Qrg5ggqHKNVpxR86QEC8w35uxmGoggxtQTPvfUu`, corresponding public key is `03d902f35f560e0470c63313c7369168d9d7df2d49bf295fd9fb7cb109ccee0494`, it's hash160 is `73c5da0a03d2d0803b731f04242bb40ced2f8bbc` and therefore the fingerprint is `73c5da0a`.
So to make sure your software wallet uses correct derivation path in PSBT you should include fingerprint and derivation like this:
segwit xpub:
`[73c5da0a/84h/0h/0h]xpub6CatWdiZiodmUeTDp8LT5or8nmbKNcuyvz7WyksVFkKB4RHwCD3XyuvPEbvqAQY3rAPshWcMLoP2fMFMKHPJ4ZeZXYVUhLv1VMrjPC7PW6V`
and corresponding receiving descriptor will be
```
wpkh([73c5da0a/84h/0h/0h]xpub6CatWdiZiodmUeTDp8LT5or8nmbKNcuyvz7WyksVFkKB4RHwCD3XyuvPEbvqAQY3rAPshWcMLoP2fMFMKHPJ4ZeZXYVUhLv1VMrjPC7PW6V/0/*)
```

chri2, 2021-10-14:
Just to fill in the information: I ran into the issue, because I tried to generate a multisig wallet in specter using as one device an electrum wallet. On the electrum wallet I only got the xpub or Zpub not including the xfp which is needed by other software to import the xpub to form a multisig wallet. Long story short, I just added it. Later I played with the specter-diy and imported the seed from electrum to the specter-diy. On the specter-diy I tried to rebuild the multisig wallet using the data of electrum with a freely invented xfp which specter-diy _correctly_ identified as not fitting its seed and derived keys.

[Electrum has got an open issue](https://github.com/spesmilo/electrum/issues/5715) related to this problem even though they only mention coldcard as a problematic device in a multisig setup. Will add the information there that their issue might also inflict problems  when using specter-diy as a replacement for a former electrum wallet.