cryptoadvance/specter-diy #197: new unrecognizable device on spectrum desktop

antonvier, 2022-01-20:
Hi I'm trying to add specter-DIY as new device on specter desktop, I enabled usb communication in DIY, I did the steps mentioned in udev, lsusb shows 'Bus 001 Device 006: ID 0483:374b STMicroelectronics ST-LINK/V2.1 ', the file manager shows the DIS_F469NI...

What am I forgetting?

thanks

stepansnigirev, 2022-01-20:
You are using a wrong port - microUSB port is used for communication with the computer, miniUSB is only for debugging and initial firmware installation.
Don't forget to switch the power jumper so the board powers from microUSB.

antonvier, 2022-01-20:
Oh God... I'm embarrassed, I knew it, but I didn't even remember.

Thanks again for the fantastic work Stepan, and forgive me for this inattention.

closed 2022-01-21 by stepansnigirev