cryptoadvance/specter-diy #211: Compact SeedQR sometimes fails

stepansnigirev, 2022-07-25:
Maybe some magic bytes at start or at the end are breaking it.

stepansnigirev, 2022-07-25:
Add more debug info:
[image]

Check this:
[image]

Failing on this QR:
[image]

kravens, 2022-07-25:
25x25 version of the same SeedQR has no issues importing, seems indeed only Compact QR related.
[image]
Here is the same 12-word seed as 25x25, which imports just fine.

https://github.com/SeedSigner/seedsigner/blob/dev/docs/seed_qr/README.md#recoverability
Since Compact SeedQR's use binary format instead of a human readable digit-stream, it is likely some strange unintended character that causes the bug.

kdmukai, 2022-07-25:
Interesting. zxing.org was fine decoding it. Same for SeedSigner itself. Any clue which bytes were tripping up the DIY?

For testing on SeedSigner I wrote a script to randomly generate new seeds and their resulting Compact SeedQRs and fed those back into our decoder. That's how I uncovered the issue with `\x00` chars which required some tweaking of an updated version of `zbar` to fix (if I remember right). Once that char was handled, I ran the script out to maybe 100k random seeds and didn't have any further decoding problems so I figured the SeedSigner implementation was pretty safe at that point.

[image]

stepansnigirev, 2022-08-13:
It was `\r` (`0x0d`) - it is used as EOL to separate multiple QR codes. Fixed in #213 by adding a "raw" mode when we don't expect animated QR codes.

PR #213 (2022-08-20): fix processing of \r when importing binary seedqr
Close #211 
Until now we assumed that all messages from the host are text-based and end with `\r` - QR scanner is adding `\r` at the end of the data, so we were using it to split data coming from the scanner when multiple QR codes are scanned one after another.

This approach doesn't work when we start working with binary data. SeedQR integration is the first time we are handling QR codes in binary form.

This PR adds a possibility to tell the scanner that we are not expecting animated QR codes and ask for raw data coming from the scanner witho [... cut]
closed 2022-08-20 by stepansnigirev