cryptoadvance/specter-diy #212: Add transcription mode to SeedQR +

refazi, 2022-08-11:
The seedqr "show recovery phrase" menu is great, but it's not easy to transcribe into a paper card or metal.
The SeedSigner has a transcription mode with a grid - it would be really great to have Specter support it as well!

stepansnigirev, 2022-08-20:
For now, I added a gap between squares that you can toggle (#215).
I think in the future it would be nice to add a proper grid and the ability to zoom in and move around the QR code. LVGL supports all that, so shouldn't be that complicated.

schnuartz-ai, 2026-08-29:
# Issue #212 — Add transcription mode to SeedQR

**Current assessment:** PARTIALLY FIXED / FEATURE REQUEST - STILL VALUABLE. A basic "transcribe" toggle (a gap between QR squares) was implemented in August 2022 via merged PR #215 and is present in the current code, wired into the mnemonic-export flow. The full SeedSigner-style workflow requested in the issue — an alignment **grid**, **zoom**, and **pan** for transcribing onto paper/metal — is NOT implemented. The verification step proposed in the latest issue comment (scan-and-compare the transcribed backup) is also NOT implemented.

**What I investigated:**
- Full issue body + both comments (maintainer @stepansnigirev 2022-08-20; @Schnuartz 2026-07-30).
- `git show ee91ca1` (the PR #215 implementation commit, 2022-08-21).
- Current `src/gui/screens/qralert.py`, `src/gui/components/qrcode.py`, `src/keystore/ram.py`.
- `gh issue view 215` (the merged PR), `gh issue list --search seedqr`.
- grep for "seedqr", "transcri", "seed.*qr" across `src/`.

**Current code:**
The transcribe gap-toggle is live end-to-end:

- `src/keystore/ram.py:379` — mnemonic export menu offers `"SeedQR (digits)"`, `"Compact SeedQR (binary)"`, `"Plaintext"`.
- `src/keystore/ram.py:396` — `await self.show(QRAlert(title="Your mnemonic as QR code", message=msg, qr_message=qr_msg, transcribe=True))` — `transcribe=True` is hardcoded for all three SeedQR encodings.
- `src/gui/screens/qralert.py:16` — `QRAlert.__init__` accepts `transcribe=False`.
- `src/gui/screens/qralert.py:24-26` — when `transcribe`, a `"Toggle transcribe"` button is added below the message.
- `src/gui/screens/qralert.py:28-29` — `toggle_transcribe` flips `self.qr.spacing` between `0` and `3`.
- `src/gui/components/qrcode.py:61-72` — the `spacing` property setter applies the gap via LVGL style: `qr_style.body.border.width = spacing; self.qr.set_style(0, qr_style)`. Default `_spacing = 0`.

What is **missing** vs. the issue's request and the maintainer's stated roadmap:
- No alignment **grid** overlay on the QR modules (only a uniform inter-module border/gap).
- No **zoom** into the QR (the existing `toggle_fullscreen` in `qrcode.py:209-225` only resizes the widget to 480x800, not a pixel-level zoom for transcription).
- No **pan**/navigation across a zoomed QR.
- No **verification** workflow (scan the user's transcribed backup and compare to the original) as proposed in the 2026-07-30 comment.

The maintainer's own 2022-08-20 comment acknowledges this is interim: *"For now, I added a gap between squares that you can toggle (#215). I think in the future it would be nice to add a proper grid and the ability to zoom in and move around the QR code. LVGL supports all that, so shouldn't be that complicated."*

**Reproduction / validation:** INFERRED from code reading (no hardware/SDL display available to click the button). The code path is unambiguous: `ram.py:396` passes `transcribe=True`, `qralert.py:24` renders the button, `qralert.py:29` toggles spacing, and `qrcode.py:68-71` applies `body.border.width`. The gap mechanism itself was added in the `lv.pximg` usermod (f469-disco commit `0910add` "add support for text.color and body.border.width (gap between pixels) in lv.pximg", the parent of the submodule bump in PR #215).

**Related issues/PRs:**
- #215 (MERGED, 2022) "add transcribe mode for seedQR (gap between squares in QR code)" — the partial implementation; PR body: *"Tackles #212, currently only adds a gap between QR squares."*
- #211 (CLOSED) "Compact SeedQR sometimes fails" — scanning-side robustness, not transcription display.
- #288 (CLOSED) "Issue with False Seed Words from Compact Seed QR" — Compact SeedQR decoding bug.
- Commit `071208a` "add [compact]seedQR and plaintext support to import mnemonics from QR and SD" — the original SeedQR import feature.
- Commit `6f89f70` (#213) "fix processing of \r when importing binary seedqr".
- Commit `031d285` (#299) "Fix for newer GM65 Scanners scanning CompactSeedQR + Add About Screen".

**Implementation notes (if still relevant):**
To deliver the full request, the natural extension points are:
1. **Grid overlay** — `src/gui/components/qrcode.py` `QRCode` class. Add an LVGL container/layer drawing grid lines on top of `self.qr` (the `lvqr.QRCode` object). The existing `spacing` setter at `qrcode.py:67-72` is the hook point; a grid can be toggled alongside or instead of the gap.
2. **Zoom/pan** — `qrcode.py:209-225` `toggle_fullscreen()` already resizes the widget; replace/augment with a zoom state (scale factor + scroll offset) applied to `self.qr.set_size(...)` and `self.qr.align(...)`. LVGL supports `lv.obj.set_style_transform_scale` (v9) / zoom on the parent. The `display_unixport/display.py` SDL init in f469-disco already accepts a `zoom=` kwarg.
3. **Verification workflow** (per the 2026-07-30 comment) — new screen after the QR display: invoke the existing QR-scan host path (`QRHost` in `src/hosts/`) to capture the user's transcribed backup, decode via the same `bip39`/SeedQR decoder used in `src/keystore/ram.py:386-395`, compare byte-for-byte against `qr_msg`, and render a module-diff overlay. Keep the comparison on-device (`src/platform.py` — no network in simulator is real, and on hardware USB/QR hosts are user-driven), and do not persist the scanned copy.

**Security review:**
Issue-scoped notes (concrete, not invented):
[... cut ...]
- No new secret-storage or network surface is introduced by the gap toggle. The proposed scan-back verification reuses the existing `QRHost` untrusted input channel; decoded data must be treated as untrusted (same as existing QR import) and scrubbed after comparison.

**Recommendation:** PARTIALLY FIXED — the gap toggle from #215 is merged and in `qralert.py`/`qrcode.py`, but the grid/zoom/pan and verification workflow remain open and valuable; keep the issue open and scope it to those remaining items.