cryptoadvance/specter-diy #219: unixport on Apple M-chips doesn't compile

stepansnigirev, 2022-08-27:
Compiles fine with micropython master

```
CC ../../../usermods/secp256k1/mpy/libsecp256k1.c
../../../usermods/secp256k1/mpy/libsecp256k1.c:496:48: warning: passing 'char *' to parameter of type 'unsigned char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
    int res = secp256k1_ec_privkey_negate(ctx, vstr.buf);
                                               ^~~~~~~~
../../../usermods/secp256k1/secp256k1/include/secp256k1.h:641:20: note: passing argument to parameter 'seckey' here
    unsigned char *seckey
                   ^
../../../usermods/secp256k1/mpy/libsecp256k1.c:578:51: warning: passing 'char *' to parameter of type 'unsigned char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
    int res = secp256k1_ec_privkey_tweak_add(ctx, priv2.buf, tweakbuf.buf);
                                                  ^~~~~~~~~
../../../usermods/secp256k1/secp256k1/include/secp256k1.h:680:20: note: passing argument to parameter 'seckey' here
    unsigned char *seckey,
                   ^
../../../usermods/secp256k1/mpy/libsecp256k1.c:1230:25: warning: assigning to 'unsigned char *' from 'char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
            bfactors[i] = vstr.buf;
                        ^ ~~~~~~~~
../../../usermods/secp256k1/mpy/libsecp256k1.c:1472:56: warning: passing 'char *' to parameter of type 'unsigned char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
    int res = secp256k1_surjectionproof_serialize(ctx, vstr.buf, &l, ptr);
                                                       ^~~~~~~~
../../../usermods/secp256k1/secp256k1/include/secp256k1_surjectionproof.h:93:18: note: passing argument to parameter 'output' here
  unsigned char *output,
                 ^
../../../usermods/secp256k1/mpy/libsecp256k1.c:1559:46: warning: passing 'char *' to parameter of type 'unsigned char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
    int res = secp256k1_rangeproof_sign(ctx, vstr.buf, &prooflen,
                                             ^~~~~~~~
../../../usermods/secp256k1/secp256k1/include/secp256k1_rangeproof.h:252:18: note: passing argument to parameter 'proof' here
  unsigned char *proof,
                 ^
../../../usermods/secp256k1/mpy/libsecp256k1.c:1724:48: warning: passing 'char *' to parameter of type 'unsigned char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
    int res = secp256k1_rangeproof_rewind(ctx, vbf_out.buf, &value_out,
                                               ^~~~~~~~~~~
../../../usermods/secp256k1/secp256k1/include/secp256k1_rangeproof.h:208:18: note: passing argument to parameter 'blind_out' here
  unsigned char *blind_out,
                 ^
../../../usermods/secp256k1/mpy/libsecp256k1.c:1725:29: warning: passing 'char *' to parameter of type 'unsigned char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
                            msg.buf, &msglenout,
                            ^~~~~~~
../../../usermods/secp256k1/secp256k1/include/secp256k1_rangeproof.h:210:18: note: passing argument to parameter 'message_out' here
  unsigned char *message_out,
                 ^
../../../usermods/secp256k1/mpy/libsecp256k1.c:1888:61: warning: passing 'char *' to parameter of type 'unsigned char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
    int res = secp256k1_rangeproof_rewind_preallocated(ctx, vbf_out.buf, &value_out,
                                                            ^~~~~~~~~~~
../../../usermods/secp256k1/mpy/config/rangeproof_preallocated/rangeproof_preallocated.h:10:17: note: passing argument to parameter 'blind_out' here
 unsigned char *blind_out, uint64_t *value_out, unsigned char *message_out, size_t *outlen, const unsigned char *nonce,
                ^
../../../usermods/secp256k1/mpy/libsecp256k1.c:1889:29: warning: passing 'char *' to parameter of type 'unsigned char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
                            msg.buf, &msglenout,
                            ^~~~~~~
../../../usermods/secp256k1/mpy/config/rangeproof_preallocated/rangeproof_preallocated.h:10:64: note: passing argument to parameter 'message_out' here
 unsigned char *blind_out, uint64_t *value_out, unsigned char *message_out, size_t *outlen, const unsigned char *nonce,
                                                               ^
9 warnings generated.
CC ../../../usermods/udisplay_f469/display_unix.c
CC ../../../usermods/udisplay_f469/display_unix_sdl.c
In file included from ../../../usermods/udisplay_f469/display_unix_sdl.c:1:
In file included from ../../../usermods/udisplay_f469/lv_sdl_hal/SDL/modSDL.c:3:
In file included from ../../../usermods/udisplay_f469/lv_sdl_hal/SDL/SDL_mouse.h:25:
In file included from /usr/local/include/SDL2/SDL.h:38:
In file included from /usr/local/include/SDL2/SDL_cpuinfo.h:99:
In file included from /Library/Developer/CommandLineTools/usr/lib/clang/13.1.6/include/immintrin.h:13:
In file included from /Library/Developer/CommandLineTools/usr/lib/clang/13.1.6/include/x86gprintrin.h:15:
/Library/Developer/CommandLineTools/usr/lib/clang/13.1.6/include/hresetintrin.h:42:27: error: invalid input constraint 'a' in asm
  __asm__ ("hreset $0" :: "a"(__eax));
                          ^
In file included from ../../../usermods/udisplay_f469/display_unix_sdl.c:1:
In file included from ../../../usermods/udisplay_f469/lv_sdl_hal/SDL/modSDL.c:3:
[... 38 lines cut ...]
    return (__m64)__builtin_ia32_paddw((__v4hi)__m1, (__v4hi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/usr/lib/clang/13.1.6/include/mmintrin.h:394:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_paddd((__v2si)__m1, (__v2si)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/usr/lib/clang/13.1.6/include/mmintrin.h:416:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_paddsb((__v8qi)__m1, (__v8qi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/usr/lib/clang/13.1.6/include/mmintrin.h:439:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_paddsw((__v4hi)__m1, (__v4hi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/usr/lib/clang/13.1.6/include/mmintrin.h:461:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_paddusb((__v8qi)__m1, (__v8qi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/usr/lib/clang/13.1.6/include/mmintrin.h:483:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_paddusw((__v4hi)__m1, (__v4hi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/usr/lib/clang/13.1.6/include/mmintrin.h:504:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_psubb((__v8qi)__m1, (__v8qi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[1]: *** [build/udisplay_f469/display_unix_sdl.o] Error 1
make: *** [unix] Error 2
```

moneymanolis, 2023-03-19:
I also tried to run the simulator today on my new Mac (ARM64). Managed to build micropython (master) and could run `simulate.py` with it, but of course it then complained about not having lvgl etc. since we "bake" additional libraries into the micropython binary.

Would be great if we could upgrade the Micropython version to make the simulator work again. I can also test to build other commits below master of Micropython if this helps.

schnuartz-ai, 2026-08-29:
# Issue #219 — unixport on Apple M-chips doesn't compile

**Current assessment:** PARTIALLY FIXED / HARDWARE VERIFICATION NEEDED. Several build-side fixes for macOS/Apple Silicon have landed since the issue was opened (a MicroPython submodule bump with an `__APPLE__` patch, a macOS-aware `MPY_CFLAGS` in the top-level Makefile, and a documented Docker `--platform linux/x86_64` workaround for reproducible builds). However, the *specific* compiler error shown in the issue body — x86 SIMD intrinsics (`mmintrin.h`, `hresetintrin.h`) pulled in via `display_unix_sdl.c` → `SDL2/SDL.h` → `SDL_cpuinfo.h` → `immintrin.h` on an ARM64 Mac — is not directly addressed by a source change in this repo. I could not compile on Apple Silicon to confirm, so whether the current tree builds cleanly on an M-chip Mac is unverified.

**What I investigated:**
- Full issue body + 1 comment (@moneymanolis, 2023-03-19).
- Top-level `Makefile`, `build_firmware.sh`, `docs/build.md`, `docs/reproducible-build.md`, `docs/simulator.md`.
- `f469-disco/usermods/udisplay_f469/display_unix_sdl.c` and `lv_sdl_hal/SDL/SDL_mouse.h`.
- `git log --all` in specter-diy and in the `f469-disco` and `f469-disco/micropython` submodules; `git show` of the M1/macOS commits.
- `gh issue list --search "mac|apple|m1|arm64"`.
- `git merge-base --is-ancestor` to confirm which fixes are in the audited HEAD (`3f3c831`).

**Current code:**

*Root cause of the original error.* The failing translation unit is `f469-disco/usermods/udisplay_f469/display_unix_sdl.c:1`:
```c
#include "lv_sdl_hal/SDL/modSDL.c"
```
`modSDL.c:3` includes `SDL_mouse.h`, whose includes at `SDL_mouse.h:25` pull `<SDL2/SDL.h>` → `SDL_cpuinfo.h` → `immintrin.h` → `mmintrin.h`/`hresetintrin.h`. On the reporter's machine these resolved to **x86** intrinsics headers from `/Library/Developer/CommandLineTools/.../immintrin.h` (visible in the issue's clang `13.1.6` paths), which are invalid on ARM64 and produce the `invalid conversion between vector type '__m64' ...` and `invalid input constraint 'a' in asm` errors. This is fundamentally an SDL2-header / clang-target mismatch (the reporter had an x86 SDL2 install at `/usr/local/include/SDL2`, likely from a Rosetta/homebrew-x86 setup), not a Specter logic bug.

*Fixes that ARE in the audited HEAD (`3f3c831`).* Confirmed via `git merge-base --is-ancestor ... HEAD` (exit 0 = ancestor):
1. `Makefile:6-10` (from `a239557`, "fix: macos build (#325)", 2025-11-24) makes the GCC-only warning flags conditional:
   ```
   ifeq ($(shell uname),Linux)
       MPY_CFLAGS ?= -Wno-dangling-pointer -Wno-enum-int-mismatch
   else
       MPY_CFLAGS ?=
   endif
   ```
   This removes clang-unsupported `-W` flags on macOS that previously broke the build before even reaching SDL. Present at `Makefile:6-10`.
2. `docs/reproducible-build.md:40-47` (from `0f1f837`, "add m1 note for reproducible build", 2022-11-07) documents the Docker workaround for M1: `docker build ... --platform linux/x86_64` and `docker run --platform linux/amd64 ...`.
3. f469-disco submodule at `db3ce3e` includes commit `662e6ed` "support apple m1" (2022-08-25), which bumps the `micropython` submodule to `4c4c72ef9`. That Micropython commit patches `py/persistentcode.c`:
   ```
   -#if defined(__i386__) || defined(__x86_64__) || defined(_WIN32) || defined(__unix__)
   +#if defined(__i386__) || defined(__x86_64__) || defined(_WIN32) || defined(__unix__) || defined(__APPLE__)
   ```
   enabling `mp_raw_code_save_file` on Apple. (Note: this fixes a *different* Apple build path — persistent code saving — not the SDL intrinsics error in the issue body.)

*Fixes that are NOT in the audited HEAD.* `git merge-base --is-ancestor` returns 1 (not an ancestor) for `8c8c6db` ("Update Makefile for Micropython v1.25", 2025-06-08). Its Makefile changes (new `build-standard` path, `mpconfigport_specter.h` injection) are absent from `Makefile:74-83`. The f469-disco commits `7c48e0f` ("upgrade Unix SDL2 driver for LVGL v9.3.0") and `14cc010` ("avoid building modSDL.c when display module is disabled") are also NOT ancestors of the current `db3ce3e` (they appear only in `--all` history). The latter is especially relevant: it wraps `#include "lv_sdl_hal/SDL/modSDL.c"` in `#if MODULE_DISPLAY_ENABLED`, which would let a no-display build skip the problematic SDL include entirely — but it is not in this HEAD, and `display_unix_sdl.c` at HEAD is still the unconditional 3-line file shown above.

**Reproduction / validation:** NOT REPRODUCED. I am on Linux x86_64 (`/home/finn/specter-diy-build`); I cannot compile the unixport on Apple Silicon. The prebuilt `bin/micropython_unix` here runs fine (see #189), which confirms the x86 path but says nothing about the M-chip path. The @moneymanolis comment (2023-03-19) reports they could build micropython master on ARM64 but then hit missing baked-in libs (lvgl), i.e. the SDL/native-module problem persisted at least into early 2023.

**Related issues/PRs:**
- #325 (the PR behind `a239557`, merged 2025-11-24) "fix: macos build" — the `MPY_CFLAGS` conditional.
- #144 (CLOSED) "[mac] make disco fails: mpy-cross" — a separate macOS native-build failure (cross-compiler, not the unixport/SDL issue).
- f469-disco `662e6ed` / micropython `4c4c72ef9` "fix apple m1" — the persistentcode patch.
- f469-disco `14cc010` (not in HEAD) "avoid building modSDL.c when display module is disabled" — would sidestep the SDL include when display is off.
[... cut ...] and does not change bytecode semantics.

**Recommendation:** PARTIALLY FIXED / HARDWARE VERIFICATION NEEDED — the Makefile (`a239557`/#325) and MicroPython `__APPLE__` patch (`662e6ed`) resolve several macOS build problems, and the Docker x86 workaround is documented, but the original SDL x86-intrinsics error in `display_unix_sdl.c` is not directly fixed in this HEAD; needs a confirmatory `make unix` on Apple Silicon (with native ARM64 `brew install sdl2`), and porting `14cc010` would harden it.