cryptoadvance/specter-diy #220: Compile from source but not the same hash

StingJohnson, 2022-10-06:
hi developers,
my eyes have been on this project for months, really interesting. 
i am following the build.md and i am building it on a ubuntu machine and i have the bin file produced. 
**but the sha256 of this bin file doesn't match any of the released binary.** 
my hash is: 
874eb42b46c76296e1fde55ca06395089eaa15c74ee704ca6c8230b2136e7f2d  specter-diy.bin\
Is it meant to be like this? 
Hope to get replied. 
Thanks a ton!

stepansnigirev, 2022-10-06:
Depending on the version of the compiler and libraries on your system the binary will be slightly different. Therefore the hash will be also different.

AFAIK it should be possible to get deterministic build, maybe with docker, could be already deterministic if you use nix package manager.

Unfortunately I don't have experience with deterministic builds, so if you know how it's normally done that would be very helpful.

StingJohnson, 2022-10-07:
thanks for the prompt reply.
If I am not mistaken, those releases prior to v1.4.0 didn't have nix with them, did they?

StingJohnson, 2022-10-12:
I am surprised that the developer is not only a genius in creating such a wallet but a DOER! Highly respect!

stepansnigirev, 2022-10-12:
Thanks for bringing up this issue! The next release will be reproducible and verifiable :)
If everything goes well I'll make a new release over the weekend.

closed 2022-10-16 by stepansnigirev