cryptoadvance/specter-diy #221: Support partial derivations in psbt

stepansnigirev, 2022-10-11:
In some cases software wallets don't add a full derivation path to the psbt, but only last non-hardened derivation after known xpub. For example if normal derivation looks like `m/84h/0h/0h/0/3` software that is using only xpub may provide only `m/0/3` part. And fingerprint can be either a fingerprint of the xpub (btcpay) or a fingerprint filled with zeroes (blue wallet).
Makes sense to check if existing wallets can derive correct keys even with a short derivation. I think we can limit support to 2-index derivations.
I thought we had this supported but it looks like it's broken now, so also makes sense to add tests for it.

tadeubas, 2025-05-09:
I think it is related, on v1.9.0 when clicking on "**Generate new key  > Done > Master public keys > Single key** (Show derivation path _on_, Use SLIP-132 _off_)" gives a descriptor without branch/output address info: `[643d3b25/84h/0h/0h]xpub6CKuFBGfGT7RECCeDcwpnTwmGLU4hVAbG6tUKyHJsjt9REHPU3d9GQnvw1kXNcnaWNH3X9m47FbWAjqEWxyHKzzT52zDJwsmBZ98xcHc4iD`

When clicking on "**Wallets > Settings > Show detailed information** (Canonical xpub _on_)" it gives the same info but with `/<0;1>/*` at end. However when clicking "**Wallets > Settings > Export wallet descriptor > Show QR Code or Save to SD card**" it gives the below JSON that has a wrong/incomplete branch/output address info with `/0/*` only:
```
{"descriptor": "wpkh([643d3b25/84h/0h/0h]xpub6CKuFBGfGT7RECCeDcwpnTwmGLU4hVAbG6tUKyHJsjt9REHPU3d9GQnvw1kXNcnaWNH3X9m47FbWAjqEWxyHKzzT52zDJwsmBZ98xcHc4iD/0/*)#aftg30ne", "label": "Default"}
```

**Additional info**: When creating a single/multi signature with the descriptor (master public key w/ derivation path) without branch/output address on Specter Desktop, it will export the JSON/PDF with wrong/incomplete branch/output address info `/0/*`.

schnuartz-ai, 2026-08-29:
# Issue #221 — Support partial derivations in psbt

**Current assessment:** PARTIALLY FIXED. The core feature requested — matching a PSBT that carries only the non-hardened tail of a derivation (e.g. `m/0/3`) against a wallet whose xpub is at account level — is present in the current code path, via embit's `Key.check_derivation` "short derivation path" branch (`f469-disco/libs/common/embit/descriptor/arguments.py:290-302`) and the Blue Wallet zero-fingerprint workaround (`src/apps/wallets/manager.py:590-596`, `fill_zero_fingerprint`). I verified this by reading the logic; I did not execute it against a real short-derivation PSBT because the existing test does not exercise that case. Two gaps remain: (1) the test `test_no_derivation` does NOT actually use a short/PSBT-side derivation, so the short-path branch has no regression coverage; (2) a separate, newer concern raised in the @tadeubas comment (May 2025) — the wallet **export** producing only `/0/*` instead of `/<0;1>/*` — is a distinct, still-open bug in `src/apps/wallets/wallet.py:75`.

**What I investigated:**
- Full issue body + 1 comment (@tadeubas, 2025-05-09, with a THUMBS_UP from the maintainer).
- `src/apps/wallets/wallet.py` (`owns`, `fill_scope`, `get_derivation`, `get_descriptor`, `from_descriptor`, `export_menu`).
- `src/apps/wallets/manager.py` (`preprocess_psbt`, `fill_zero_fingerprint`, `find_wallet_from_address`).
- embit bundled at `f469-disco/libs/common/embit/descriptor/arguments.py` (`AllowedDerivation`, `Key.check_derivation`) and `descriptor.py` (`Descriptor.owns`, `Descriptor.check_derivation`); `bip32.py` (`HDKey.my_fingerprint`); `psbt.py` (`DerivationPath`).
- Git history: `git log --all --grep "partial|derivation|psbt|AllowedDerivation"` in both the repo and the `f469-disco` submodule; `git show` of `c91a36d`, `054e4cd`, `9e67558`, `ed77919`, `7665a63`; `git log -S` on `fill_zero_fingerprint` and `short derivation path`.
- Tests: `test/integration/tests/test_with_rpc.py` (`test_no_derivation`), `test/tests/test_wallets.py`, `f469-disco/tests/tests/test_psbt.py`.
- `gh issue list --search "psbt derivation"` → #221 (open), #80 (closed, custom derivation paths), #186 (closed).

**Current code:**

*Origin of the feature.* Commit `c91a36d` "allow partial derivations of known wallets in psbt (#163)" (2021-06-10) refactored `Wallet.owns` to delegate to `self.descriptor.owns(psbt_scope)` and added the `test_no_derivation` test. `git show c91a36d` confirms `Wallet.owns` went from a manual scriptpubkey+derivation check to a one-liner calling embit's `Descriptor.owns`.

*How a short/PSBT-side derivation is matched today.* `Descriptor.owns` (`f469-disco/libs/common/embit/descriptor/descriptor.py:203-233`) iterates `psbt_scope.bip32_derivations`, and for each extended key in the descriptor calls `Key.check_derivation(der)` (`arguments.py:290-302`):
```python
def check_derivation(self, derivation_path):
    rest = None
    # full derivation path
    if self.fingerprint == derivation_path.fingerprint:
        origin = self.derivation
        if origin == derivation_path.derivation[: len(origin)]:
            rest = derivation_path.derivation[len(origin) :]
    # short derivation path
    if self.my_fingerprint == derivation_path.fingerprint:
        rest = derivation_path.derivation
    if self.allowed_derivation is None or rest is None:
        return None
    return self.allowed_derivation.check_derivation(rest)
```
- **btcpay case** (fingerprint = the xpub's own fingerprint): the "short derivation path" branch fires, `rest = derivation_path.derivation` (e.g. `[0, 3]`), then `AllowedDerivation.check_derivation` (`arguments.py:69-91`) matches it against `[[0,1], None]` → `(idx=3, branch_idx=0)`. `HDKey.my_fingerprint` is `hash160(sec)[:4]` of the xpub itself (`bip32.py:64-68`), so this matches an account-level xpub fingerprint. By inspection this is exactly the #221 scenario.
- **blue wallet case** (fingerprint = `00000000`): `self.my_fingerprint` is the real hash160, so neither `self.fingerprint` nor `self.my_fingerprint` equals `b"\x00\x00\x00\x00"`. This is handled separately by `WalletManager.fill_zero_fingerprint` (`manager.py:590-596`), invoked in `preprocess_psbt` for every input (`manager.py:665`) and output (`manager.py:726`) BEFORE `fill_scope`/`owns` runs:
```python
def fill_zero_fingerprint(self, scope):
    """Blue Wallet hack - zeroes in fingerprint should be checked and replaced by our fingerprint"""
    for pub in scope.bip32_derivations:
        if scope.bip32_derivations[pub].fingerprint == b"\x00\x00\x00\x00":
            # check if pubkey matches
            if self.keystore.get_xpub(scope.bip32_derivations[pub].derivation).key == pub:
                scope.bip32_derivations[pub].fingerprint = self.keystore.fingerprint
```
  This was added in `ed77919` "add bw hack with zero fingerprint (#176)" (2021-08-24). Caveat: it only fires for the **device's own keys** (it calls `self.keystore.get_xpub(...)` and compares the pubkey), so a multisig cosigner's xpub delivered with a zero fingerprint would NOT be patched. The regular (non-Liquid) manager only; `grep` found no equivalent in `src/apps/wallets/liquid/manager.py`.

*Wallet construction with no derivation in descriptor.* `Wallet.from_descriptor` (`wallet.py:306-316`) detects when all keys are extended with `allowed_derivation is None` and assigns `AllowedDerivation.default()` (=`[[0,1], None]`) so the device can still derive receive/change:
```python
[... cut ...]
- No new secret handling. The export fix (#2 above) only changes which public descriptor is written; it does not leak xprvs (the descriptor is public by construction).

**Recommendation:** PARTIALLY FIXED — the PSBT-side partial-derivation matching the issue asks for is implemented (btcpay via `arguments.py:298-299`, blue wallet via `manager.py:590-596`) but has no test covering the short path, and the @tadeubas export-only-`/0/*` bug (`wallet.py:75`) is still open. Suggest: (a) add the short-derivation integration test, (b) fix the export to include change, then close.