cryptoadvance/specter-diy #223: UIUX: Improve entering recovery phrase screen

moneymanolis, 2022-10-15:
Two things would massively improve the UX here:
- Having a "Previous word" button besides the "Next word" button, jumping to the previous word. Few users will find out that you can actually change a word again by the `<` sign - and this would also erase a lot of entered words when going back from, say, word 10 to word 2 to change it. The alternative here would be to enable to jump to a previous word by the touch display, but I suppose this is much more difficult to implement @stepansnigirev?
- A cursor would help to orientate much better where one is currently entering what.

stepansnigirev, 2022-10-15:
I think jumping to the cell using touch screen makes most sense. I think it's better than having a "previous word" button. Cursor also makes sense to implement. I didn't do that because right now the cursor is always at the end.

moneymanolis, 2024-01-27:
I've thought about this screen more. 

We should split the UX improvements into two parts. First part would make the "enter your phrase" UX better and the second part would be to improve the UI if you want / need to correct a word. Let's start with the first part. I think these relatively easy (?) changes would go a long way: 

- Renaming "Next word" to "Auto-complete" + highlighting the currently used row in `RecoverMnemonicScreen`

schnuartz-ai, 2026-08-29:
# Issue #223 — UIUX: Improve entering recovery phrase screen

**Current assessment:** FEATURE REQUEST - STILL VALUABLE (DUPLICATE/continuation of #45). The recovery-entry screen (`RecoverMnemonicScreen`) still lacks every concrete ask in this issue: there is no "Previous word" button, no touch-to-jump-to-cell, and no cursor/row highlight for the active word. The autocomplete half (landed in #45 via PR #66) is done; the navigation/hand-off half that #223 specifically adds is not. The "Next word" button still double-duties as an auto-complete accept. No code changes have touched the navigation behavior since the issue was opened.

**What I investigated:**
- Full issue body + 2 comments (@stepansnigirev, 2022-10-15; @moneymanolis, 2024-01-27).
- Cross-referenced #45 (open) — the parent issue, including my prior audit comment on #45 (posted 2026-08-29) which established that autocomplete is done but landscape/swipe/"Previous word" are not.
- `src/gui/screens/mnemonic.py` (`RecoverMnemonicScreen`, full file 1-386).
- `src/gui/components/mnemonic.py` (`MnemonicTable`, full file 1-85).
- `src/gui/components/` for `HintKeyboard`.
- Git history: `git log --all --grep "recovery|mnemonic|previous|keyboard"` and `git log --all -- src/gui/screens/mnemonic.py src/gui/components/mnemonic.py`.
- `gh issue list --search "recovery"` → #45, #223 (both open), plus #349/#385 (unrelated feature requests).

**Current code:**

*The recovery keyboard has no "Previous word".* `RecoverMnemonicScreen.__init__` (`mnemonic.py:186-235`) builds a `HintKeyboard` with this button map (`mnemonic.py:187-224`):
```python
self.kb.set_map([
    "Q","W","E","R","T","Y","U","I","O","P","\n",
    "A","S","D","F","G","H","J","K","L","\n",
    "Z","X","C","V","B","N","M", lv.SYMBOL.LEFT, "\n",
    lv.SYMBOL.LEFT + " Back", "Next word", lv.SYMBOL.OK + " Done", "",
])
```
Row 4 has exactly three controls: `Back`, `Next word`, `Done`. There is no "Previous word" / forward-word navigation button. The only back-navigation is `lv.SYMBOL.LEFT` (row 3), which is a per-character backspace handled at `mnemonic.py:327-328` (`self.table.del_char()`), not a per-word jump.

*No touch-to-cell jump on the recovery screen.* `MnemonicTable` (`components/mnemonic.py:5-85`) exposes only `add_char`, `del_char`, `autocomplete_word`, `get_last_word`, `set_mnemonic`, `update`. There is no `goto_word(idx)` / `set_active_word(idx)` / highlight API, and no touch event handler on the table in `RecoverMnemonicScreen` (contrast with `NewMnemonicScreen.on_word_click` at `mnemonic.py:105-117`, which DOES let the user tap a cell to edit — but that's the *new*-mnemonic screen, not recovery). So Stepan's preferred solution (touch-to-cell) is also absent.

*No active-row/word highlight.* `MnemonicTable.update` (`components/mnemonic.py:40-47`) just writes the word strings into cells; it never restyles the cell for the currently-typed word. There is no cursor object. This matches the issue's "cursor would help to orientate" ask and @moneymanolis's (a) "highlighting the currently used row".

*"Next word" still conflates with auto-complete.* `RecoverMnemonicScreen.callback` (`mnemonic.py:329-334`):
```python
elif c == "Next word":
    word = self.table.get_last_word()
    if self.lookup is not None and len(word) >= 2:
        candidates = self.lookup(word, 2)
        if len(candidates) == 1:
            self.table.autocomplete_word(candidates[0])
```
and `get_mnemonic` (`mnemonic.py:274-278`) enables "Next word" only when `len(candidates) == 1 or word in candidates`. This is exactly the @moneymanolis (b) concern: "Next word" performs an auto-complete accept rather than purely advancing. The suggested split (auto-complete only via the suggestion buttons; "Next word" only advances) has NOT been applied — the diff @moneymanolis proposed is not in the tree.

*Git history confirms no navigation work.* `git log --all -- src/gui/screens/mnemonic.py` shows only LVGL-9 migration, bip85, seedQR, mnemonic-fix, PIN-required, and "fix Done when mnemonic is not complete" commits — none add previous-word/touch-jump/row-highlight. `git log --all --grep "previous"` (case-insensitive) returns nothing. The `git log --all --grep "keyboard"` hits are a button-miss handler and an LVGL migration, not word navigation.

**Reproduction / validation:**
- Code inspection only. I did not run the LVGL flow on hardware/simulator. The recovery screen relies on physical-keyboard LVGL widgets; the absence of the buttons/methods listed above is definitive from the source, but the *feel* of the existing flow (e.g. whether the `<` backspace is discoverable) is a UX judgment I can only infer.
- The stale-state glitch from #45 (Back → re-enter shows old input) is not in scope here; #223 does not mention it. As noted in my #45 audit, `RecoverMnemonicScreen.__init__` builds a fresh `MnemonicTable` per entry (`mnemonic.py:175` → `super().__init__("", title)` → `MnemonicTable(self)`), so the reported glitch is likely already resolved, but unverified on hardware.

**Related issues/PRs:**
- #45 (OPEN) "Improve recovery screen (bip39 words)" — the parent; this issue is its continuation. My audit comment on #45 (2026-08-29) already tabulated: landscape ❌, suggestions ✅, swipe ❌, 4-char-auto-proceed partial, sound ❌, cursor ❌.
- #66 (the PR behind `0bedf5f`/`67be6db` "add mnemonic autocomplete") — delivered the autocomplete that both #45 and #223 build on.
[... cut ...] the existing `checker(mnemonic)` gate at `mnemonic.py:289` as the single authority.

**Recommendation:** FEATURE REQUEST - STILL VALUABLE — DUPLICATE/continuation of #45; none of #223's asks (Previous-word button, touch-to-cell jump, cursor/row highlight, Next-word/auto-complete split) are implemented in `src/gui/screens/mnemonic.py` / `src/gui/components/mnemonic.py`. Suggest consolidating into #45 and implementing the touch-to-cell navigation + active-row highlight as the highest-value subset.