cryptoadvance/specter-diy #224: Misleading battery status

moneymanolis, 2022-10-15:
After fully charging the battery is never showing a full bar and it **very** quickly deteriorates from green to orange. I am not sure whether this is due to my device or a more general problem, e.g. rounding errors.
Using Specter Shield.

stepansnigirev, 2022-10-15:
The battery status is implemented here:
https://github.com/cryptoadvance/specter-diy/blob/master/src/platform.py#L267

Right now it's a simple linear interpolation based on some values from the datasheet.

If you want to make a proper display of the battery status you can calibrate it. For that I'd suggest writing a script that will:
- turn on the display (to make power consumption larger)
-  read the battery level every 10 seconds or so in an infinite loop
- after every measurement open a file on the SD card in append mode and add measured value there together with time.time() value

Then you can charge the battery to 100%, and keep the script running until the device turns off.
Then we can plot the results and see how we should change the function to return better estimates.

Charging is close to zero when the device is on, depending on the device can be even negative.
It should be possible to optimize the power consumption but I didn't put any effort into that.
For example, currently the display is fully refreshed even if nothing happens on the screen, but most screens are static. So one optimization would be to refresh the display only when touch is detected.

schnuartz-ai, 2026-08-29:
# Issue #224 — Misleading battery status

**Current assessment:** STILL RELEVANT (HARDWARE VERIFICATION NEEDED) for the Specter Shield + battery add-on. The charging-symbol half of the issue is DONE (a `CHARGE` glyph is shown when charging, `src/gui/components/battery.py:40-42`, driven by `src/platform.py:423`). The "misleading level" half — the battery never shows full and drops green→orange too fast — is NOT fixed: the `BATTERY_TABLE` voltage→level mapping (`src/platform.py:395-401`) and the `Battery.LEVELS` color thresholds (`battery.py:7-13`) are unchanged since the battery monitor was first added (`cdb6b94`, 2021-01-25) and were never calibrated against a real LiPo discharge curve. The maintainer's own comment proposes a calibration script; that script was never written/run, and no calibration data exists in the repo. On a bare STM32F469-DISCO (no Shield) `i2c` stays `None`, `get_battery_status` returns `(None, None)`, and no battery icon is shown, so the issue does not manifest there.

**What I investigated:**
- Full issue body + 1 comment (@stepansnigirev, 2022-10-15).
- `src/platform.py` (`get_battery_status`, `BATTERY_TABLE`, `i2c`, `usb_connected`).
- `src/gui/components/battery.py` (full file 1-44).
- `src/gui/async_gui.py` (`set_battery_callback`, `update_battery`), `src/gui/screens/screen.py` (`Battery` widget placement), `src/specter.py` (callback registration).
- `boot/main/boot.py` and `boot/debug/boot.py` (i2c init, fuel-gauge setup, poweroff).
- Git history: `git log --all --grep "battery|charging|power|fuel|i2c"`; `git show cdb6b94`; `git log -- src/platform.py`, `-- src/gui/components/battery.py`, `-- boot/main/boot.py`.
- `gh issue list --search "battery"` → #224 (open) only; #159 (closed, unrelated restart issue).
- Grepped `src/` for `battery|charging|power|USB|LED|fuel|batt|i2c` to map all battery-related sites.

**Current code:**

*Voltage → level table (the misleading part).* `src/platform.py:395-424`:
```python
BATTERY_TABLE = [
    (4.2,  100),
    (4.0,  75),
    (3.85, 50),
    (3.75, 35),
    (3.6,  0),
]

def get_battery_status():
    # simulator or no i2c
    if i2c is None:
        return None, None
    try:
        # check if battery monitor exists
        if 112 not in i2c.scan():
            return None, None
        voltage = int.from_bytes(i2c.mem_read(2, 112, 8),'little')*2.44e-3
        level = 0
        for i, (v, lvl) in enumerate(BATTERY_TABLE):
            if voltage > v:
                # max voltage
                if i == 0:
                    level = lvl
                    break
                # linear interpolation
                prevV, prevLvl = BATTERY_TABLE[i-1]
                level = int(lvl + (prevLvl-lvl)*(voltage-v)/(prevV-v))
                break
        charging = (int.from_bytes(i2c.mem_read(2, 112, 6),'little') < 8192)
        return level, charging
    except Exception as e:
        print(e)
        return None, None
```
The fuel gauge sits at I2C address `112` (0x70) — a MAX170xx-class monitor. Voltage is read from register 8 (LSB × 2.44 mV); charging state from register 6 (`< 8192` ⇒ charging). The level is a piecewise-linear interpolation of `BATTERY_TABLE`.

*Why it reads "misleading".* `Battery.LEVELS` (`battery.py:7-13`) maps the computed level to a glyph+color:
```python
LEVELS = [
    (95, lv.SYMBOL.BATTERY_FULL,  "00D100"),  # green
    (75, lv.SYMBOL.BATTERY_3,     "00D100"),  # green
    (50, lv.SYMBOL.BATTERY_2,     "FF9A00"),  # orange
    (25, lv.SYMBOL.BATTERY_1,     "F10000"),  # red
    (0,  lv.SYMBOL.BATTERY_EMPTY, "F10000"),  # red
]
```
A freshly charged LiPo sits at ~4.2 V only while on the charger; off-charger it falls rapidly to ~4.0–4.1 V. With the current table, 4.2 V → 100 (only shown while charging), 4.1 V → ~87, 4.0 V → 75, and anything below 4.0 V interpolates toward 50 → `BATTERY_2` (orange). So the green→orange transition happens at the 4.0 V / 75% boundary, which a real LiPo crosses within minutes of coming off charge — exactly the reporter's "very quickly deteriorates from green to orange." "Never showing a full bar" off-charger is because 4.2 V→100 requires the open-circuit voltage of a just-finished charge, which the cell relaxes below almost immediately. The table is a rough datasheet guess, not a measured curve; it has never been tuned (`git log -- src/platform.py` shows no edit to `BATTERY_TABLE` or `get_battery_status` since `cdb6b94`).

*The charging symbol IS implemented.* `battery.py:26-44`:
```python
def update(self):
    if self.VALUE is None: ... return
[... 16 lines cut ...]
*Wiring.* `boot/main/boot.py:18-23` initializes `i2c = pyb.I2C(1); i2c.init()` and, if `112 in i2c.scan()`, writes `0b00010000` to register 0 to start measurements; `boot/main/boot.py:64` injects it as `platform.i2c = i2c`. `src/specter.py:116-117` registers `get_battery_status` at a 3000 ms interval. `src/platform.py:34` declares `i2c = None` (overwritten by boot.py); on a bare DISCO the `112 not in i2c.scan()` check at `platform.py:409` returns `(None, None)` and `battery.py:27-31` hides the widget, so the bug is Shield-specific. `usb_connected` (`platform.py:390-393`) reads `USB_VBUS` but is not used for the battery display.

**Reproduction / validation:**
- NOT REPRODUCED. I have no Specter Shield with battery and no hardware fuel gauge; on Linux the simulator path hits `i2c is None` (`platform.py:405`) and returns `(None, None)`, so the battery widget is invisible and the misleading-level path cannot be observ [... cut ...] a security standpoint.

**Recommendation:** STILL RELEVANT — HARDWARE VERIFICATION NEEDED; the charging-symbol half is DONE (`battery.py:40-42`), but the misleading-level half is unfixed — `BATTERY_TABLE` (`platform.py:395-401`) was never calibrated (unchanged since `cdb6b94`, 2021-01-25). Fix is blocked on collecting a real discharge curve on a Specter Shield; until then the table is a datasheet guess. Relevant only to the Shield+battery add-on; bare DISCO shows no battery icon (`i2c is None`).