cryptoadvance/specter-diy #233: Suggest user to format SD card if it can't be read by Specter DIY

aglkm, 2022-11-17:
Specter DIY can read SD cards formatted to fat 16/32. 
It would be nice if the device could detect that a card has a different format and suggest a user to re-format it within the device.

Or/and a meaningful error message that the device can't read SD card and a suggestion with the ways how to fix it (re-format, replace the manufacturer of the card, too large, too small and etc.)

SaroC, 2022-12-11:
This maybe could be useful to fix some issue. I flashed the initial fw into the DISCO and I have an error only when the sd card is inside. I tryed to format th 32GB card several time without results.

schnuartz-ai, 2026-08-29:
# Draft comment for #233 — Suggest user to format SD card if it can't be read by Specter DIY

## Current assessment

**STILL RELEVANT (UX/robustness gap).** Not implemented on HEAD (`3f3c831`). `SDCard.mount()` calls `os.mount(self._sd, "/sd")` with no `try/except`, so any mount failure (wrong filesystem, ENODEV, card too large/unrecognised, corrupt MBR) propagates as an uncaught `OSError` and surfaces to the user as a raw traceback ("Something unexpected happened..."). There is no detection of "card present but unmountable" and no prompt offering to reformat. The one existing use of `os.VfsFat.mkfs` is for the SDRAM ramdisk, not the SD card. A real-world user hit exactly this in the comment on this issue and in #44 (`OSError: [Errno 19] ENODEV` with a 32 GB card).

## What I investigated

- `src/platform.py` — `SDCard` class (`mount`, `unmount`, `is_present`, `__enter__`/`__exit__`), `mount_sdram` (the only `mkfs` call), `file_exists`.
- `src/hosts/sd.py` — `SDHost.reset_and_mount` (calls `platform.sdcard.mount()` with no error handling).
- `src/keystore/sdcard.py` — `SDKeyStore` calls `platform.sdcard.mount()` directly in `save_mnemonic`, `load_mnemonic`, `delete_mnemonic`, `is_key_saved` (all unguarded).
- `src/apps/wallets/wallet.py`, `src/apps/xpubs/xpubs.py` — further `mount()`/`is_present` call sites.
- `grep` for `ENODEV`, `errno`, `Errno`, `VfsFat`, `mkfs`, `format`, `reformat` across `src/`.
- Git history: `git log -- src/platform.py`, `git log --grep="format|mount|ENODEV"`.
- Related issues: #44 (open), #141 (open).

## Current code

`SDCard.mount` performs the mount with no exception handling (`src/platform.py:74-84`):

```python
def mount(self):
    """Mounts SD card"""
    if not self.is_present:
        raise RuntimeError("SD card is not present")
    if self._sd is None:
        return
    if self._led is not None:
        self._led.on()
    self._sd.power(True)
    os.mount(self._sd, "/sd")          # <-- can raise OSError(ENODEV)/EIO/etc., uncaught
    self._mounted = True
```

`unmount` syncs first then `os.umount` (`src/platform.py:95-107`):

```python
def unmount(self):
    """Unmounts SD card"""
    if not self._mounted:
        return
    self._mounted = False
    if self._sd is None:
        return
    os.sync()
    os.umount("/sd")
    self._sd.power(False)
    ...
```

The only in-repo use of FAT filesystem creation is for the SDRAM ramdisk, not the SD card (`src/platform.py:246-257`):

```python
def mount_sdram():
    path = fpath("/ramdisk")
    ...
    else:
        bdev = sdram.RAMDevice(512)
        os.VfsFat.mkfs(bdev)           # <-- mkfs exists in the API but is never offered for /sd
        os.mount(bdev, path)
    return path
```

Callers invoke `mount()` directly and let exceptions bubble to the generic handler in `specter.py`, which prints a traceback (`src/specter.py:146-152`). Example from `SDHost.reset_and_mount` (`src/hosts/sd.py:26-33`):

```python
def reset_and_mount(self):
    if self.f is not None:
        self.f.close()
        os.remove(self.fram)
[... 10 lines cut ...]
## Reproduction / validation

Not reproduced on hardware (hardware unavailable). The failure mode is reported by two independent users: the comment on this issue ("I tryed to format the 32GB card several time without results") and the ENODEV traceback in #44 (`OSError: [Errno 19] ENODEV` at `platform.py mount_sdcard` → `keystore/sdcard.py is_key_saved`). The code confirms there is no format/autorecover path: `mount()` raises, and the generic exception handler in `src/specter.py:146-152` displays a traceback rather than a helpful prompt.

## Related issues/PRs

- **#44** (open) — SD card support tracking; the ENODEV comment there is the same symptom (32 GB card, FAT32, `Errno 19`).
- **#141** (open) — "xPub & µSD" — adjacent SD UX issue.
- No commits in history touch SD mount error handling or formatting (`git log --grep="format|mount|ENODEV"` returns only unrelated formatting/format commits and the `mount sd before file_exists` fix #164, which is about ordering, not error handling).

## Implementation notes

1. **Catch mount errors in `SDCard.mount`.** Wrap `os.mount(self._sd, "/sd")` (`src/platform.py:83`) in `try/except OSError`. On failure, raise a typed exception (e.g. `SDCardMountError`) carrying the errno, instead of letting a raw `OSError` escape. Keep `_mounted = False` and power down the SD on the error path (currently `unmount` is the only place `_sd.power(False)` is called — add it to the failure cleanup).
2. **Offer to reformat.** When the error is mount-related (ENODEV `19`, EIO `5`, or `OSError` from `os.mount`), present a `Prompt`: "Can't read SD card. Format it as FAT32? This will erase all data." On confirm, call `os.VfsFat.mkfs(self._sd)` then retry `os.mount`. The `os.VfsFat.mkfs` API is already used at `src/platform.py:255` for the ramdisk, so no new dependency is needed; `self._sd` (`pyb.SDCard()`) is a block device suitable for `mkfs`.
3. **Surface meaningful messages.** Map common errnos to hints, matching the issue's ask ("re-format, replace the manufacturer, too large, too small"): ENODEV → "Card not detected or unsupported — try a smaller card (≤32 GB) or reformat on a computer as FAT32."; EIO → "Card read failed — try reformatting or replacing the card.". Put this mapping in `SDCard.mount` or a small helper.
[... cut ...] akage path.

## Recommendation:

STILL RELEVANT — wrap `os.mount` in `SDCard.mount` (`src/platform.py:83`) with `try/except OSError`, raise a typed recoverable error, and add a user-confirmed "Format as FAT32?" prompt that calls `os.VfsFat.mkfs(self._sd)` (already used for the ramdisk at `src/platform.py:255`) before retrying the mount. Add errno→hint mapping (ENODEV/EIO) to satisfy the issue's "meaningful error message" ask. Hardware verification recommended for the mkfs-on-`pyb.SDCard()` path.