cryptoadvance/specter-diy #236: Add a setting to customize fonts size

aglkm, 2022-11-26:
A request to add a setting to be able to customize fonts size.
Currently there is no possibility to increase fonts size in the specter-diy gui.

schnuartz-ai, 2026-08-29:
# #236 — Add a setting to customize fonts size

## Current assessment

**STILL RELEVANT — FEATURE REQUEST - STILL VALUABLE.** There is no font-size setting. All fonts are hardcoded to one of three sizes (16/22/28 px) across the theme and per-screen styles. The LVGL build already compiles a wider range of font sizes (12/16/22/28 for both Roboto and Roboto Mono, plus unscii_8 and square1–10), so a runtime size setting is feasible without new font assets — but constrained to those pre-compiled sizes (custom intermediate sizes would require the lvgl font converter + a rebuild).

## What I investigated

- Issue body + comments (no comments). Request: "add a setting to be able to customize fonts size… there is no possibility to increase fonts size in the specter-diy gui."
- `src/gui/common.py` (`init_styles`) — the central theme/style setup.
- All `font` references across `src/gui/` (screens and components).
- The LVGL font set actually compiled into the build: `f469-disco/usermods/udisplay_f469/lv_conf.h` and `f469-disco/usermods/udisplay_f469/micropython.mk`.
- The settings persistence mechanism (`src/specter.py` `load_settings`/`save_settings`) for where a new setting would be stored.
- Git history: `--grep font` and `src/gui/common.py` log (notably the recent LVGL 9.x migration `195a50c` and the PIN font commit `1fb727c`).
- Related issues: #132 (OPEN, PIN font), #293 (OPEN, language selection), #298 (OPEN, language button).

## Current code

**Centralized style init** — `init_styles(dark=True)` in `src/gui/common.py:16` sets the LVGL theme default font to `lv.font_roboto_22` (`common.py:19` and `:31`) and defines named styles with fixed fonts:

- `common.py:80` — `th.style.ta.oneline.text.font = lv.font_roboto_28` (text area / input)
- `common.py:107` — `styles["title"].text.font = lv.font_roboto_28`
- `common.py:112` — `styles["hint"].text.font = lv.font_roboto_16`

**Per-screen hardcoded fonts:**

- `src/gui/screens/input.py:241-244` and `:254` — PIN/keyboard buttons use `lv.font_roboto_28` (this was the `1fb727c` "increase PIN font size" change addressing #132).
- `src/gui/screens/transaction.py:42,46,51,56,61` — mono_28, mono_22, roboto_22 for transaction amounts/labels.
- `src/gui/components/mnemonic.py:13` — `lv.font_roboto_22` for mnemonic cells.
- `src/gui/components/qrcode.py:36` — `lv.font_roboto_16` for QR footer text.

So the app uses exactly three body sizes (16, 22, 28) today; #132 already pushed the PIN buttons from default to 28.

**Compiled-in fonts (confirmed in the LVGL config of the f469-disco submodule):**

`f469-disco/usermods/udisplay_f469/lv_conf.h:287-315`:
```c
#define LV_FONT_ROBOTO_12    1
#define LV_FONT_ROBOTO_16    1
#define LV_FONT_ROBOTO_22    1
#define LV_FONT_ROBOTO_28    1
#define LV_FONT_UNSCII_8     1
#define LV_FONT_CUSTOM_DECLARE ... font_roboto_mono_{12,16,22,28} ...
```
and `micropython.mk:43-46,78-81` compiles the mono variants. Available to Python at runtime: `lv.font_roboto_12/16/22/28`, `lv.font_roboto_mono_12/16/22/28`, `lv.font_unscii_8`, `lv.font_square_*`. Sizes 12 and the mono set are currently unused by the app.

**Settings persistence (where a setting would live):** `Specter.load_settings`/`save_settings` (`src/specter.py:515-533`) already JSON-serialize a global settings dict to `SETTINGS_DIR/global.settings`, encrypted with `keystore.settings_key`. An experimental-settings UI exists (`experimental_settings`, `specter.py:535-562`) showing the pattern for adding a new toggle/field. `BaseApp.GLOBAL` (`src/app.py:19`) exposes settings to all apps.

**API note (LVGL 9 migration):** commit `195a50c` (Dec 2025) migrated core GUI to LVGL 9 (`theme_default_init`, `style.set_*`, `screen_active()`). Any font-setting work must follow the LVGL 9 style API used in the migrated files.

## Reproduction / validation

- **Verified by reading source**: all font references above are hardcoded (file:line quoted). No `font_size`/`font_scale` setting or dynamic font selection exists (exhaustive grep of `src/gui/`).
- **Verified by reading source**: the lv_conf.h + micropython.mk compile roboto 12/16/22/28 and mono 12/16/22/28 into the firmware, so a runtime choice among these needs no new assets.
- Not run on hardware; font glyph coverage for the smaller (12) or larger custom sets at the device's 480×800 resolution not visually verified.

## Related issues/PRs

- **#132** (OPEN) — "PIN code font size can be bigger" — partially addressed by `1fb727c` (PIN buttons → roboto_28) but the issue is still open; a global font-size setting would close it.
- **#293** (OPEN) — "Implement language selection in the settings" — related i18n/settings work; a font setting and language setting would share the settings UI and may interact (some scripts need larger glyphs).
- **#298** (OPEN) — "Implementing a language button" — same area.

## Implementation notes

Feasible with no new font assets (the 12/16/22/28 sets are already compiled in). Suggested approach:

1. **Add a `font_scale`/`font_size` setting** to global settings (`Specter.GLOBAL`), saved via `save_settings` (`specter.py:526`) and loaded at boot via `load_settings` (`specter.py:515`). Add a settings menu item next to `experimental_settings` (`specter.py:535`) — e.g. a `DevSettings`/`HostSettings` control offering "Small / Medium / Large" mapping to base sizes 16 / 22 / 28 (and 12 for the hint style at "Small").
2. **Introduce a font table in `src/gui/common.py`** keyed by the setting, e.g.:
   ```python
   FONT_SETS = {
[... cut ...]
No security impact. Fonts are a pure UI concern; the setting is encrypted at rest with the rest of global settings (`settings_key`). No user-controlled data flows into the font objects (they are static LVGL symbols), so there is no parsing/overflow surface introduced.

## Recommendation

STILL RELEVANT — practical, low-risk enhancement. The compiled font set already supports 3–4 sizes, so no new assets are needed for a discrete size setting; wire a `font_size` field into global settings (`specter.py:515-533`) and route `init_styles` (`common.py:16`) plus the per-screen hardcoded fonts through a font table. Would also close #132.

Schnuartz, 2026-08-31:
I think the information from Stepan Snigirev about that in #132 is interesting for that.

There was the question:

> @stepansnigirev How difficult would it be to implement application-wide font size setting (including a new font) ?

And Stepan answered with:

> Not very difficult, there is a tool that can convert fonts from ttf to lvgl fonts:
> https://lvgl.io/tools/fontconverter
> 
> And then we can add bindings to micropython to access these fonts here:
> https://raw.githubusercontent.com/diybitcoinhardware/f469-disco/master/usermods/udisplay_f469/lv_mpy_example.c
> (search for roboto_mono_16 to get an idea how it's done)
> 
> The system-wide style is set here, so the theme can be tweaked with minimal effords:
> https://github.com/cryptoadvance/specter-diy/blob/master/src/gui/common.py#L16-L112
> There are also a few things that are hard-coded in the screen classes that need to be moved to the theme instead.
> 
> Another thing I wanted to do for a very long time is to add icons for error messages and buttons. This can be done with some icon font.
> 
> I am postponing these things because I am thinking about migrating from lvgl v6 to lvgl v7. It has cleaner UX and much better styling opportunities. And also I would like to spend some time on redesign of the UX in general. Now we reached the point when we have a pretty solid architecture of the wallet, so it's clear what the UI needs to do and what widgets we commonly use.
> 
> If you have any suggestions on the UX improvements I'll be happy to hear.
> 
> P.S. Upgrade to v1.5.1 please, I broke the tx screen in v1.5.0 with the last-minute change...