cryptoadvance/specter-diy #238: Add support for base43 encoding

stepansnigirev, 2022-12-06:
It is used in Electrum

Jason-Flem, 2022-12-07:
hi, Stephen, I do feel the pain when i was trying to set up electrum with my specter diy.
It would be a huge plus if you can make it work.
Again, thanks million for you hard work.

schnuartz-ai, 2026-08-29:
# #238 — Add support for base43 encoding

## Current assessment

**STILL RELEVANT — FEATURE REQUEST - STILL VALUABLE (but narrow scope).** Base43 encoding is not implemented anywhere in the firmware or its `embit` library — confirmed by exhaustive grep across `src/`, `f469-disco/libs/common/embit/`, and `f469-disco/libs/common/`. The codebase supports three QR payload encodings today (raw binary, base64, Blockchain-Commons UR/BCUR including `crypto-psbt`), plus a binary "Compact SeedQR" mnemonic export, but **not** Electrum-style base43. The issue request ("It is used in Electrum") is specifically about interoperability with Electrum's base43-encoded payloads (most notably Electrum's encrypted seed backups and some legacy QR formats). The request is still valid as a feature gap; the main caveat is that base43's practical use has narrowed over time (UR/BCUR has largely displaced it for PSBT interchange), so the value today is mainly mnemonic/seed-import interop with Electrum rather than PSBT transport.

## What I investigated

- Issue body ("It is used in Electrum") + one user comment asking for Electrum interop (no maintainer response, no labels).
- `src/qrencoder.py`: the QR encoder class hierarchy (`QREncoder`, `Base64QREncoder`, `LegacyBCUREncoder`, `CryptoPSBTEncoder`).
- `src/hosts/qr.py`: scanner host, including the recent CompactSeedQR/GM65 RAW-mode fix (commit `031d285`, #299).
- `src/helpers.py`: stream helpers (`b2a_base64_stream`, `a2b_base64_stream`).
- `src/keystore/ram.py`: mnemonic export menu (SeedQR digits / Compact SeedQR binary / plaintext).
- `f469-disco/libs/common/embit/` and `f469-disco/libs/common/`: any base43 / electrum-encrypted-seed module.
- Git history: `git log --oneline --all --grep="base43\|b43\|compactseedqr\|compact seed" -i` and per-file history of `src/qrencoder.py`, `src/hosts/qr.py`.
- PR search: `gh pr list --search "base43"` (none). Issue search: `gh issue list --search "base43 encoding electrum"` (none beyond #238; #57 about BCUR standards is closed).

## Current code

**Supported QR encodings** (`src/qrencoder.py`):

- `Base64QREncoder` (`qrencoder.py:73-82`) — `p1ofN <base64>` chunked PSBT; uses `b2a_base64_stream` (`helpers.py:136-143`).
- `LegacyBCUREncoder` (`qrencoder.py:84-104`) — `UR:BYTES/` legacy BCUR; uses `bcur_encode_stream`.
- `CryptoPSBTEncoder` (`qrencoder.py:106-147`) — `crypto-psbt` via `microur.encoder.UREncoder`.
- Plain `QREncoder` (`qrencoder.py:7-71`) — raw bytes passthrough (`read_write`).

**Decode side** (`src/apps/wallets/manager.py`): routes `UR:BYTES/` → `SIGN_BCUR` (`manager.py:160-163`, `bcur_decode_stream` at `226`), raw/base64 PSBT magic → `SIGN_PSBT` (`manager.py:164-176`), and address/wallet descriptors (`manager.py:180-191`). No base43 branch.

**Mnemonic QR export** (`src/keystore/ram.py:377-396`) offers three formats:

```python
# ram.py:379-395
v = await self.show(
        Menu([(1, "SeedQR (digits)"), (2, "Compact SeedQR (binary)"), (3, "Plaintext")],
        ...))
...
elif v == 1:
    nums = [bip39.WORDLIST.index(w) for w in self.mnemonic.split()]
    qr_msg = "".join([("000"+str(n))[-4:] for n in nums])   # SeedQR: 4-digit decimal per word
elif v == 2:
    qr_msg = bip39.mnemonic_to_bytes(self.mnemonic)          # Compact SeedQR: raw entropy bytes
elif v == 3:
    qr_msg = self.mnemonic
```

"Compact SeedQR" here is **raw binary entropy** (`bip39.mnemonic_to_bytes`), **not** base43. The #299 GM65 scanner fix (`031d285`) is about enabling RAW scanning mode for these binary QR codes on newer GM65 hardware — unrelated to base43.

**No base43 anywhere.** `grep -rniE "base43|b43" src/ f469-disco/` returns no application hits (the only `b43` matches are an unrelated STM32F469 SVD register name `f469-disco/debug/STM32F469.svd:3033` and false positives). `embit` (`f469-disco/libs/common/embit/`) has `base.py`, `base58.py`, `bech32.py`, `compact.py` — no base43 module. Electrum's encrypted-seed format (base43 over an AES+HMAC payload, optionally XPRV) is not parsed or produced.

## Reproduction / validation

- **Verified by reading source and grep**: no base43 encoder/decoder or Electrum-seed import path exists in `src/`, `f469-disco/libs/common/embit/`, or `f469-disco/libs/common/` (exhaustive case-insensitive grep for `base43`, `b43`, `electrum`, `encrypted.*seed`). Inferred: the feature is absent.
- **Verified by reading source**: "Compact SeedQR (binary)" produces raw entropy via `bip39.mnemonic_to_bytes` (`ram.py:391`), not base43; the #299 change (`git show 031d285 -- src/hosts/qr.py`) is a scanner RAW-mode/EEPROM fix, not an encoding addition.
- Not reproduced at runtime (no hardware/simulator run in this audit).

## Related issues/PRs

- **#57** (CLOSED) — "Cooperating with Blockchain Common's LetheKit team on QR standards, etc." — the project committed to the UR/BCUR direction for QR interchange, which is the modern replacement for ad-hoc encodings like base43. Context for why base43 was likely never prioritized.
- No open or closed PRs for base43 (`gh pr list --search "base43"` → empty).
- No other issues mentioning base43.

## Implementation notes

1. **Where base43 actually matters for interop.** Electrum uses base43 (alphabet `0123456789ABCDEFGHJKLMNPQRSTUVWXZabcdefghijklmnopqrstuvwxyz*$` — 43 chars) for its "Export → Encrypted hex"/"Encrypted seed" QR and for some legacy URI/PSBT forms. For Specter-DIY the realistic use cases are: (a) **import** an Electrum-encrypted seed QR to restore a wallet, and (b) option [... cut ...]
STILL RELEVANT — implement base43 encode/decode as a first-class QR format (`src/qrencoder.py` + `src/helpers.py`) and route Electrum base43 seed QRs to mnemonic import in `parse_stream` (`manager.py:142-193`). Scope to plaintext base43 seed import first; defer Electrum *encrypted*-seed import (passphrase + Electrum crypto stack) as a separate follow-up unless the maintainers want it. Note that for PSBT transport, base43 is legacy and the project already standardized on UR/BCUR (`crypto-psbt`, #57), so PSBT-over-base43 is not recommended.