cryptoadvance/specter-diy #256: RFC 6979, deterministic signing, Anti-Klepto / Anti-Exfiltration

thashm, 2023-07-02:
Hello, as title suggests, does the project use RFC 6979 standard to create ECDSA signatures or have any  Anti-Klepto / Anti-Exfiltration ability against nonce attacks?

stepansnigirev, 2023-08-13:
We already use RFC 6979 for signature generation.
Anti-Klepto would be nice to have, but will require an additional round of communication via QR codes, which makes UX much more complex. For now I don't have ETA for this feature.

schnuartz-ai, 2026-08-29:
# #256 — RFC 6979, deterministic signing, Anti-Klepto / Anti-Exfiltration

## Current assessment

**DUPLICATE of #234.** This issue asks whether the device uses RFC 6979 deterministic nonces and whether it implements Anti-Klepto / Anti-Exfiltration. The maintainer (stepansnigirev) already answered in the issue thread: *"We already use RFC 6979 for signature generation. Anti-Klepto would be nice to have, but will require an additional round of communication via QR codes, which makes UX much more complex. For now I don't have ETA for this feature."* The actionable implementation ticket for the missing part (anti-klepto) is **#234 — "Implement anti nonce exfiltration protocol"** (OPEN), which I have already audited in detail. #256 is the user-facing question; #234 is the implementation task covering the identical topic. They should be linked, and #256 can be closed as a duplicate of #234 once cross-referenced.

This comment deliberately adds **no new exploit detail** beyond what the issue and #234 already state publicly.

## What I investigated

- Issue body + the single maintainer comment (quoted above).
- Cross-referenced my prior audit of **#234** (see `draft_234.md` in this audit batch), which already traced the full signing path and the RFC 6979 / anti-klepto question end to end.
- Confirmed #234 and #256 are the only issues in the tracker on this topic (`gh issue list --search "anti-klepto nonce RFC 6979 deterministic"` → #234 only; #256 found via direct view).
- Signing entry points and the libsecp256k1 nonce function binding (summarized below from #234's verified findings).

## Current code

(Verified in `draft_234.md`; summarized here for completeness — all file:line refs re-confirmed against HEAD `3f3c831`.)

**RFC 6979 — present.** The micropython `secp256k1` binding calls libsecp256k1's default nonce function, which *is* RFC 6979:
- `f469-disco/usermods/secp256k1/mpy/libsecp256k1.c:420-448` — `ecdsa_sign(msg, sk)` → `secp256k1_ecdsa_sign(ctx, &sig, msg, sk, usecp256k1_nonce_function, data)` (nonce callback = `secp256k1_nonce_function_default` = RFC 6979; the comment at `libsecp256k1.c:306` confirms "same as `secp256k1_nonce_function_rfc6979`").
- `f469-disco/usermods/secp256k1/mpy/libsecp256k1.c:904-920` — `ecdsa_sign_recoverable` likewise uses the same nonce function.
- The pure-Python bootloader fallback (`bootloader/tools/core/secp256k1/key.py:397` `deterministic_k`) is an explicit RFC 6979 implementation.
- Application signing calls (`src/keystore/ram.py:74-88`: `sign_psbt`, `sign_input`, `sign_hash`, `sign_recoverable`) and embit (`f469-disco/libs/common/embit/ec.py:216-230`, `psbtview.py:806-823`) all route to these bindings.

**Anti-Klepto / host nonce commitment — absent.** No `antiklepto`, `commit`, `host_random`, `host_nonce`, or `nonce_data` parameter anywhere in `src/` or `f469-disco/libs/common/embit/` (exhaustive grep; re-verified). The signing flow accepts only the message hash and derives the nonce internally; there is no host→device commitment round. This matches the maintainer's comment.

## Reproduction / validation

- **Verified by reading source** (re-confirmed at HEAD `3f3c831`): RFC 6979 is the nonce function for both ECDSA and recoverable ECDSA via the libsecp256k1 binding; no anti-klepto/host-commitment code or parameter exists. Not run on hardware; inferred from the C binding the firmware compiles.
- Full evidence and line references are in `draft_234.md` (issue #234 audit).

## Related issues/PRs

- **#234** (OPEN) — "Implement anti nonce exfiltration protocol" — the implementation ticket for the missing anti-klepto half of this question. Already audited; classification STILL RELEVANT — FEATURE REQUEST - STILL VALUABLE. Contains the full implementation plan (nonce `data` argument plumbing, embit `ec.py`/`bip32.py`/`psbtview.py` changes, host protocol in `src/hosts/` + `manager.py`, Schnorr `aux_rand`).
- No other issues on this topic.

## Implementation notes

See `draft_234.md` (issue #234) for the full, evidence-based implementation plan. In short: libsecp256k1's nonce function already accepts a `data` pointer that is mixed into RFC 6979, and the micropython binding already exposes it (`ecdsa_sign(msg, secret, callback, data)` at `f469-disco/usermods/secp256k1/mpy/libsecp256k1.c:420-448`), so no C change is needed for ECDSA — the work is in embit (`ec.py`, `bip32.py`, `psbtview.py`) to thread a `host_commitment`/`nonce_data` parameter down from `src/apps/wallets/manager.py`'s signing flow, plus a new host commitment round in `src/hosts/`. Recoverable signing and Schnorr (`aux_rand`) need binding additions. Nothing new to add here beyond #234's plan.

## Security review

- **Confirmed strength:** RFC 6979 deterministic nonces are correctly in place, so a *malicious host* cannot bias the nonce to leak the secret key (the host supplies only the message hash).
- **Confirmed gap (the issue's concern, already documented in #234):** determinism alone does not defend against a *malicious device/firmware* choosing biased nonces that leak the secret key via published signatures (the "nonce exfiltration" one-way channel). Anti-klepto closes this by mixing host-supplied committed randomness into the nonce. Not implemented.
- This comment discloses **no new vulnerability detail**; the above is the publicly stated motivation for #234 and for the maintainer's own comment on this issue.

## Recommendation

DUPLICATE of #234 — close #256 as a duplicate and link it to #234 (the implementation ticket). RFC 6979 is already implemented and verified; the anti-klepto / host-commitment protocol is the open work tracked under #234 (see `draft_234.md` for the full plan and security analysis).

Schnuartz, 2026-10-09:
So I had a look. And with another round Stepan means not that we would just have one more QR to scan but

[image]