cryptoadvance/specter-diy #257: Unable to use it with Nunchuk wallet as coordinator

bitcoinheiro, 2023-07-11:
I was unable to scan a multisig descriptor provided by Nunchuk, received an error message:
[image]

Also, when trying to read an address to check if it was part of the wallet, this is the message I got:
[image]

I'm running the latest version of the Specter DIY firmware and of Nunchuk wallet

BenGWeeks, 2023-12-19:
Did you resolve this?

schnuartz-ai, 2026-08-29:
# Issue #257 — Unable to use it with Nunchuk wallet as coordinator

**Current assessment:** PARTIALLY FIXED. The descriptor-scanning failure is most likely resolved by PR #301 (commit `a5c9926`, Oct 2025), which expanded multisig-descriptor detection. The address-verification failure, however, is NOT resolved: the device still refuses to verify an address that lacks an explicit `index=` query parameter, which is a plausible cause of the second error screenshot in the report. Hardware/QR-scanner validation of part 1 is still recommended.

---

## What I investigated

- Fetched the issue (opened 2023-07-11, OPEN, one follow-up comment from Dec 2023 with no resolution). Two distinct failures are reported: (1) "unable to scan a multisig descriptor provided by Nunchuk" with an error, and (2) "when trying to read an address to check if it was part of the wallet" with an error.
- The descriptor-import path: `WalletManager.parse_stream` → `ADD_WALLET` → `parse_wallet` → `Wallet.parse` → `Descriptor.from_string` (embit).
- The address-verification path: `parse_stream` → `VERIFY_ADDRESS` → `find_wallet_from_address`.
- Git history of `src/apps/wallets/manager.py`, in particular commit `a5c9926` ("Expand detection of possible multisig descriptors & Add data input parsing CI tests", #301).
- embit descriptor parsing capabilities for the `wsh(sortedmulti(...))` form Nunchuk emits.

---

## Current code

### Part 1 — descriptor detection (manager.py:177-184)

`parse_stream` decides whether incoming QR/USB data is a wallet descriptor:

```python
# src/apps/wallets/manager.py:178-184
# Check for an & Symbol, typically used when descriptor supplied with a name
# Also check the most common descriptor types for multisig wallets
common_descriptor_markers = [b"&", b"tr(", b"wsh(", b"sh("]
if any(marker in data for marker in common_descriptor_markers) and b"?" not in data:
    # rewind
    stream.seek(0)
    return ADD_WALLET, stream
```

Before #301 the check was only (confirmed via `git show a5c9926^:src/apps/wallets/manager.py`):

```python
# OLD (pre-#301)
if b"&" in data and b"?" not in data:
    return ADD_WALLET, stream
```

Nunchuk exports multisig wallets as `wsh(sortedmulti(...))` **without** a `Name&` prefix, so the old single-`&` check would not classify it as `ADD_WALLET`. The current code recognizes `wsh(`, `sh(`, and `tr(` markers, so a bare Nunchuk `wsh(sortedmulti(...))` descriptor is now detected.

### embit descriptor parsing (descriptor.py:286-366)

The vendored embit `Descriptor.read_from` handles `tr(`, `sh(wsh(`, `wsh(`, `sh(wpkh`, `wpkh(`, `pkh(`, `sh(` and delegates the body to `Miniscript.read_from` (which includes `sortedmulti`). A Nunchuk-style `wsh(sortedmulti(2,...))` is parseable.

### Part 2 — address verification (manager.py:249-263)

```python
# src/apps/wallets/manager.py:249-263
elif cmd == VERIFY_ADDRESS:
    data = stream.read().decode().replace("bitcoin:", "")
    # should be of the form addr?index=N or similar
    if "index=" not in data or "?" not in data:
        raise WalletError("Can't verify address with unknown index")
    addr, rest = data.split("?")
    args = rest.split("&")
    idx = None
    for arg in args:
        if arg.startswith("index="):
            idx = int(arg[6:])
            break
    w, _ = self.find_wallet_from_address(addr, index=idx)
    await show_screen(WalletScreen(w, self.network, idx))
    return True
```

The check at line 252-253 **requires both** `index=` and `?`. An address sent as a bare `bc1q...`, `bitcoin:bc1q...`, or `bitcoin:bc1q...?index=...` missing the `?`/`index=` pair raises `WalletError("Can't verify address with unknown index")`.

[... 27 lines cut ...]

---

## Implementation notes

For the still-open address-verification half (part 2), a minimal improvement would let the device identify the owning wallet when `index=` is absent by scanning within each wallet's gap-limit range:

1. In `WalletManager.process_host_command` → `VERIFY_ADDRESS` (manager.py:249-263): when `index=` is missing, fall back to a bounded search rather than raising.
2. Add a method on `Wallet` such as `find_address(addr, network, gap_limit=None)` that iterates branch `0..num_branches-1` and index `0..gaps[branch]-1` (capped by `GAP_LIMIT`) calling `get_address` and comparing — reusing the existing `gaps` bookkeeping (wallet.py:184-202) so the scan stays bounded.
3. In `find_wallet_from_address` (manager.py:565-588) add an `index=None, paths=None, scan=False` path that loops over `self.wallets` and calls the new bounded search when `scan` is set.
4. Keep the explicit `index=` fast path as-is; only invoke the bounded scan when the caller omitted the index.

This mirrors how the gap-limit is already used for PSBT inputs (manager.py:685-689, wallet.py:184-202) and avoids unbounded derivation scans.

---

## Security review

- The descriptor path receives untrusted host data and passes it to `Descriptor.from_string`; embit validates structure and raises `DescriptorError` on malformed input, which `parse_wallet` wraps in `WalletError` (manager.py:532-542). No injection risk identified in the changed detection code — it only routes the stream to the existing parser.
- A bounded address scan (implementation notes above) must be capped by `GAP_LIMIT` (wallet.py:23, `=20`) to avoid resource exhaustion from a host-supplied address that matches nothing (an attacker could otherwise force an open-ended derivation loop). The existing `gaps` array provides the natural bound.
- The `b"?" not in data` guard at manager.py:181 prevents a BIP21-style `addr?index=` payload from being misrouted into `ADD_WALLET`; this is preserved.

---

## Recommendation: PARTIALLY FIXED — descriptor scanning is most likely fixed by #301 (needs hardware confirmation); the address-verification-without-`index=` failure remains open and is a concrete, low-risk follow-up.