cryptoadvance/specter-diy #259: Miniscript Fragment C has no Attribute Name

Rob1Ham, 2023-08-13:
Running some tests, I have a miniscript output descriptor that will not be recognized by the Specter DIY:

```{"label":"Test_Miniscript_Wallet","descriptor":"wsh(c:andor(multi(1,[d83593a2/48'/1'/0'/2']tpubDF4vhecRfUyYF5AEzfcFC7oNapKKVm24wdJmkjZ6QZefJsT6wAvFPotgAAPSCmWTDkysMHk9kv5pzwUDTNpDbAUj4vP6arLVt6YAysoan16/0/*,[9b5f84f8/48'/1'/0'/2']tpubDEDgqmrhRpSYBpSA7dbxg7kZYhsPgACWGPcLavmyJVqdh2ERL3ZNAnUu1HHbAVYK3Rcj25A1pQMWPDCcsbGeGdq7DsAxNVZnRFatNTsAKXw/0/*),pk_k([621f3bec/48'/1'/0'/2']tpubDEXD6B4sX85AyA3WJv1rB3NnADe8EvujnudmLTY8dmdgVKyUst3R65KQqAVyxY5q5USsMh9iqgGkDMtMzfV7zvRtUhV8timsH3H37P5C4Nt/0/*),pk_k([5d2e766d/48'/1'/0'/2']tpubDEdap7gaFhs35jNgpQhVadeKREP1XzPrD2DZDxbi3HRQL8CUDM1XWVsUYVD4HwWmECoe9hBFZsZT3w7ap282jVMA6XfF3VjUuYb2UGUrgze/0/*)))"}```

Upon trying to upload this output descriptor, it shows the following error:

[image]

[image]

stepansnigirev, 2023-08-13:
Thanks!
This should fix it: https://github.com/diybitcoinhardware/embit/pull/50

Rob1Ham, 2023-08-13:
Thanks for the quick response! I'm working on building the firmware myself, including the updated `embit` library to reproduce and confirm the bug is fixed.