cryptoadvance/specter-diy #263: Do not sign a psbt if specter does nit own any input

pythcoiner, 2023-11-24:
I experienced this during Liana wallet testing ([see here](https://github.com/wizardsardine/liana/issues/788)), specter will sign a PSBT even if no input is owned (in my case I load the wrong key).
Is there some reason to let sign such PSBT?
Shouldn't specter at least warn the user he's going to sign this king PSBT?

stepansnigirev, 2023-12-10:
Specter-DIY actually counts number of signatures that it is adding to the transaction. If 0 signatures are added it will show an error screen with the text like "We didn't add any signatures".

It's very strange that it was able to sign the transaction without any warnings if the key is not in the setup. Would be nice to have a test vector to reproduce it.

stepansnigirev, 2025-11-05:
plz double check if it's still the case and provide test vector

closed 2025-11-06 by pythcoiner