cryptoadvance/specter-diy #271: Cannot register descriptor

pythcoiner, 2023-12-10:
While i try to register this descriptor 
```
wsh(or_i(and_v(v:thresh(1,pkh([5eb6aff1/48'/1'/0'/2']tpubDEmCTbWN3ZHypBRHCvdHhm4hEJbCZ6EhgT8Krfemmvmo3AqZee3dk9GgVYH9bmsyptreUwUXob4JARKXhdnYJqkJdut7CsAC8npuTRncVi3/<4;5>/*),a:pkh([94ed3672/48'/1'/0'/2']tpubDE1HLabeQUBLMogm7FD2zhh2Y9rkhM8sHhBE2HGdc9jB5yodsrezgsqdwZMKWaTSPmxBEQ2NrAAU1kTX254rzKnZM5xvPee6DP1mTjjtR7A/<4;5>/*)),older(65535)),or_i(and_v(v:thresh(2,pkh([5eb6aff1/48'/1'/0'/2']tpubDEmCTbWN3ZHypBRHCvdHhm4hEJbCZ6EhgT8Krfemmvmo3AqZee3dk9GgVYH9bmsyptreUwUXob4JARKXhdnYJqkJdut7CsAC8npuTRncVi3/<2;3>/*),a:pkh([94ed3672/48'/1'/0'/2']tpubDE1HLabeQUBLMogm7FD2zhh2Y9rkhM8sHhBE2HGdc9jB5yodsrezgsqdwZMKWaTSPmxBEQ2NrAAU1kTX254rzKnZM5xvPee6DP1mTjjtR7A/<2;3>/*)),older(55000)),or_i(and_v(v:thresh(1,pkh([6abb52a9/48'/1'/0'/2']tpubDFZTCVU1Sa9nJXCxx97UFvGausHQPFjJyaiDbdr8GNqjCLKwYc8ihegK7yJdcizs9HMbiGA7ke1HiCENVHaERvNANHW7U2Wo2qnRsuqB52r/<4;5>/*),a:pkh([d4ab66f1/48'/1'/0'/2']tpubDEXYN145WM4rVKtcWpySBYiVQ229pmrnyAGJT14BBh2QJr7ABJswchDicZfFaauLyXhDad1nCoCZQEwAW87JPotP93ykC9WJvoASnBjYBxW/<4;5>/*)),older(45000)),or_d(multi(2,[6abb52a9/48'/1'/0'/2']tpubDFZTCVU1Sa9nJXCxx97UFvGausHQPFjJyaiDbdr8GNqjCLKwYc8ihegK7yJdcizs9HMbiGA7ke1HiCENVHaERvNANHW7U2Wo2qnRsuqB52r/<0;1>/*,[d4ab66f1/48'/1'/0'/2']tpubDEXYN145WM4rVKtcWpySBYiVQ229pmrnyAGJT14BBh2QJr7ABJswchDicZfFaauLyXhDad1nCoCZQEwAW87JPotP93ykC9WJvoASnBjYBxW/<0;1>/*),and_v(v:thresh(3,pkh([6abb52a9/48'/1'/0'/2']tpubDFZTCVU1Sa9nJXCxx97UFvGausHQPFjJyaiDbdr8GNqjCLKwYc8ihegK7yJdcizs9HMbiGA7ke1HiCENVHaERvNANHW7U2Wo2qnRsuqB52r/<2;3>/*),a:pkh([d4ab66f1/48'/1'/0'/2']tpubDEXYN145WM4rVKtcWpySBYiVQ229pmrnyAGJT14BBh2QJr7ABJswchDicZfFaauLyXhDad1nCoCZQEwAW87JPotP93ykC9WJvoASnBjYBxW/<2;3>/*),a:pkh([5eb6aff1/48'/1'/0'/2']tpubDEmCTbWN3ZHypBRHCvdHhm4hEJbCZ6EhgT8Krfemmvmo3AqZee3dk9GgVYH9bmsyptreUwUXob4JARKXhdnYJqkJdut7CsAC8npuTRncVi3/<0;1>/*),a:pkh([94ed3672/48'/1'/0'/2']tpubDE1HLabeQUBLMogm7FD2zhh2Y9rkhM8sHhBE2HGdc9jB5yodsrezgsqdwZMKWaTSPmxBEQ2NrAAU1kTX254rzKnZM5xvPee6DP1mTjjtR7A/<0;1>/*)),older(35000)))))))#5h3txqua
```
using Liana wallet via usb, i got hint ' Processing host data, please wait until the process is complete...' then nothing, no 'wallet' added on specter side and liana get info as registration success.

stepansnigirev, 2023-12-10:
Pretty large descriptor :) Can you try loading the wallet from SD card, does that work?

To import wallet from SD card you need to create a text file with content `addwallet <wallet_name>&<wallet_descriptor>` and open it from the wallet.

This can be either an issue with USB, or with memory allocation. I'll investigate.

seedor-io, 2024-11-14:
I'm facing a similar issue. I am trying to register a large decaying multisig (Liana) with my Specter Shield running 1.90.
Smaller Descriptors were no problem but this one is a 3 of 5  decaying to 2 of 5 decaying to 1 of 5 and when I load it via SD card I get the following error.

[image]

Ending on MemoryError: memory allocation failed allocating 2368 bytes

Here I anonymized the xpubs. The text file looked like this because the standard `addwallet <wallet_name>&<wallet_descriptor>` command didn’t work for me, but this format always did:

`{"descriptor": "tr(xpub_/<0;1>/*,{{and_v(v:multi_a(2,[f68f701d/48'/0'/1'/2']xpub_2/<2;3>/*,[f4c7191f/48'/0'/0'/2']xpub_3/<2;3>/*,[7a9b94bf/48'/0'/0'/2']xpub_4/<2;3>/*,[b826e13b/48'/0'/0'/2']xpub_5/<2;3>/*,[7986d2f7/48'/0'/0'/2']xpub_1/<2;3>/*),older(52600)),and_v(v:multi_a(1,[f68f701d/48'/0'/1'/2']xpub_2/<4;5>/*,[f4c7191f/48'/0'/0'/2']xpub_3/<4;5>/*,[7a9b94bf/48'/0'/0'/2']xpub_4/<4;5>/*,[b826e13b/48'/0'/0'/2']xpub_5/<4;5>/*,[7986d2f7/48'/0'/0'/2']xpub_1/<4;5>/*),older(65535))},multi_a(3,[f68f701d/48'/0'/1'/2']xpub_2/<0;1>/*,[f4c7191f/48'/0'/0'/2']xpub_3/<0;1>/*,[7a9b94bf/48'/0'/0'/2']xpub_4/<0;1>/*,[b826e13b/48'/0'/0'/2']xpub_5/<0;1>/*,[7986d2f7/48'/0'/0'/2']xpub_1/<0;1>/*)})#4l4aafpg", "label": "Liana_Chungus"}`

Registering this descriptor on Coldcard Mark4, Q and BitBox02 worked.

pythcoiner, 2024-11-15:
if it's a problem about the memory, one quick fix could be to allow import wallet policies (BIP388), as keys are not repeated there.

stepansnigirev, 2025-11-05:
TODO: optimize miniscript and descriptors to use less memory, add this descriptor as test case.
Also makes sense to implement bip388 but it will solve only half of the problem

schnuartz-ai, 2026-08-29:
**Current assessment**

STILL RELEVANT. The issue (registering a large Liana miniscript descriptor fails on the device) is confirmed as an on-device memory problem. The maintainer (@stepansnigirev) acknowledged it in a 2025-11-05 TODO comment ("optimize miniscript and descriptors to use less memory, add this descriptor as test case"). No fix has landed in the repo as of HEAD `3f3c831`. The descriptor parses correctly on desktop CPython, so this is a heap/fragmentation problem on the STM32 target, not a descriptor-grammar bug.

**What I investigated**

- Fetched the full issue body + 4 comments. The original reporter (USB/Liana) gets stuck at "Processing host data, please wait…"; @seedor-io reproduced it via SD-card import with `MemoryError: memory allocation failed allocating 2368 bytes`; @pythcoiner suggested BIP388 wallet policies; the maintainer's latest comment (2025-11-05) lists "optimize miniscript/descriptor memory" + "implement bip388 (solves only half)" as TODO.
- Reproduced parsing of the exact descriptor from the issue on desktop with the pip-installed `embit==0.8.0` (the vendored copy in `f469-disco/libs/common/embit` is Micropython-only and requires the C secp256k1 usermod, so I used the API-equivalent installed package for a logic/memory-structure probe — the parsing code is byte-for-byte identical; see "Reproduction / validation" caveat).
- Measured the in-memory structure: 37 miniscript nodes, 12 key *occurrences* but only **4 unique xpubs** (each xpub is repeated across the `or_i`/`thresh`/`and_v` branches), `num_branches = 2` (from the `<4;5>` / `<2;3>` / `<0;1>` multi-path sets), compiled witness script = 379 bytes, `str(descriptor)` = 1943 chars.
- Instrumented the `Miniscript.keys` property: it is invoked **74 times during a single `Descriptor.from_string()`** and ~37 times for each later access (`d.keys`, `d.num_branches`, `d.full_policy`). The property is recomputed every call with no caching.
- Confirmed BIP388 / wallet policies are **not** implemented anywhere (`grep -rn "bip388\|BIP388\|WalletPolicy\|wp_"` → no matches in `src/` or vendored `embit/`).
- `git log --all --grep="271"` and `gh pr list --search 271` → no PR/commit references this issue.

**Current code**

The parsing/registration path (USB and SD share the same code):

- `src/apps/wallets/manager.py:241-248` — `ADD_WALLET` branch reads the descriptor and calls `self.parse_wallet(desc)`.
- `src/apps/wallets/manager.py:532-542` — `parse_wallet()` → `self.WalletClass.parse(desc)` → raises `WalletError("Can't parse descriptor…")` on any exception (this is where `MemoryError` surfaces to the user).
- `src/apps/wallets/wallet.py:294-316` — `Wallet.parse()` splits on `&`, calls `from_descriptor()`, which calls `cls.DescriptorClass.from_string(desc)` (the embit `Descriptor`), then iterates `descriptor.keys` and sets `allowed_derivation`.
- `f469-disco/libs/common/embit/descriptor/descriptor.py:286-292` — `Descriptor.from_string()` builds the full object tree.
- `f469-disco/libs/common/embit/descriptor/descriptor.py:49` — `for k in self.keys:` in `Descriptor.__init__` (first `keys` traversal, to set `taproot`).
- `f469-disco/libs/common/embit/descriptor/miniscript.py:20-25` — the hot spot:
  ```python
  @property
  def keys(self):
      return sum(
          [arg.keys for arg in self.args if isinstance(arg, Miniscript)],
          [k for k in self.args if isinstance(k, Key) or isinstance(k, KeyHash)],
      )
  ```
  `sum(list, list)` is **O(n²)** in the number of args (each `+` copies the accumulator) and is recomputed on every access (no memoization). For the Liana tree this is called 74× during parse alone, each call allocating several temporary lists.
- `f469-disco/libs/common/embit/descriptor/descriptor.py:276-283` — `Descriptor.keys` delegates to `self.miniscript.keys` (so every `Descriptor.keys` access triggers a full O(n²) re-walk).
- `src/apps/wallets/manager.py:435-445` — `confirm_new_wallet()` calls `w.get_key_dicts(network)` and `w.full_policy`.
- `src/apps/wallets/wallet.py:234-258` — `get_key_dicts()` builds a list of dicts for **every key occurrence** (12 here, not the 4 unique xpubs) and computes SLIP132/canonical base58 strings for each.
- `f469-disco/libs/common/embit/descriptor/descriptor.py:152-159` — `full_policy` builds the 1943-char `str(self.miniscript)` then does `s.replace(str(k), chr(65+i))` once per key occurrence (12 string replacements, each producing a new ~2 KB string).
- `f469-disco/libs/common/embit/descriptor/descriptor.py:36-40` — `Descriptor.__init__` also calls `len(k.branches)` for every key to validate branch counts.

**Reproduction / validation**

- TESTED (desktop, `embit==0.8.0`): the issue's exact descriptor parses successfully, `derive(0, branch_index=0)` succeeds, `full_policy` returns a 246-char abstracted policy (12 letters A–L, i.e. it does **not** deduplicate the 4 unique xpubs). So the grammar and logic are correct.
- TESTED (instrumentation): `Miniscript.keys` invoked 74× during `from_string`; 37 nodes, 12 key occurrences, 4 unique xpubs.
- NOT REPRODUCED on-device: I cannot run the STM32 build here, so the `MemoryError: …allocating 2368 bytes` reported by @seedor-io is taken from the issue as given. The 2368-byte failure is consistent with heap fragmentation (a small allocation failing because no contiguous block is free even though total free bytes may be higher) rather than a single oversized allocation — which matches the observed pattern of many mid-sized tempora [... cut ...] script.py:596-614`) bounds `k`. The memory exhaustion is a availability issue on the hardware wallet (a large/miniscript descriptor can be supplied by the host and wedge wallet import until the device reboots), not a key-handling or signing-bypass flaw. No new private security detail.

Recommendation: STILL RELEVANT — prioritize memoizing/deduplicating `keys` in the vendored embit and add the issue's descriptor as a frozen regression test; BIP388 is a worthwhile follow-up but not a complete fix.

lawnodes, 2026-10-03:
Confirming this is still on issue, also over SD card.