cryptoadvance/specter-diy #276: [Feature request] Add NFC

tscharly, 2024-06-08:
NFC would be great

Azzbo77, 2024-06-23:
Specter shield has this

stepansnigirev, 2024-06-23:
Spectre shield has wired communication, not NFC.
NFC can be done with standard Arduino shields like [this one](https://www.adafruit.com/product/789).

Changes in specter-javacard are required though...

schnuartz-ai, 2026-08-29:
**Current assessment**

FEATURE REQUEST — STILL VALUABLE (but blocked on upstream `specter-javacard`, not on specter-diy firmware code). The issue asks for NFC as an alternative transport for the JavaCard secure element that stores the key. The maintainer (stepansnigirev) confirmed in the issue thread that the current Specter shield uses **wired** communication, not NFC; NFC is feasible with off-the-shelf Arduino NFC shields (e.g. Adafruit PN532-based shield #789) but requires changes in `specter-javacard` (the on-card applet code) first. No NFC/contactless code exists in the firmware today, and none is needed *in specter-diy* until the card side supports it.

**What I investigated**

- Fetched the full issue body and 2 comments. Title: "[Feature request] Add NFC". Body: "NFC for communication with smartcard would be great". Azzbo77 commented that "Specter shield has this" (incorrect — see below). stepansnigirev corrected: "Specter shield has wired communication, not NFC. NFC can be done with standard Arduino shields like [Adafruit #789]. Changes in specter-javacard are required though…".
- Inspected the device-side smartcard stack to confirm the current transport is wired only:
  - `src/keystore/javacard/util.py:11-24` — `get_reader()` builds an `sc.Reader(...)` hardwired to STM32 GPIO pins: `ioPin=Pin.cpu.A2`, `clkPin=Pin.cpu.A4`, `rstPin=Pin.cpu.G10`, `presPin=Pin.cpu.C2`, `pwrPin=Pin.cpu.C5`. These are the ISO-7816 contact-pad signals (I/O, CLK, RST, presence, VCC) routed to the smartcard socket on the shield — i.e. a **contact** reader, not a contactless/NFC antenna.
  - `f469-disco/usermods/scard/protocols.h:16-25` — the `uscard` usermod protocol enum exposes only `T0_protocol`, `T1_protocol`, `protocol_any`, `protocol_na`. T=0/T=1 are the **contact** ISO-7816 transmission protocols. There is no `T=CL` / ISO 14443 / contactless protocol, and no NFC stack.
  - Searched the scard usermod for "contactless"/"NFC"/"RFID"/"ISO14443"/"T=CL" → zero hits. The reader is contact-only.
  - `src/keystore/javacard/applets/applet.py:14-37` — `Applet` talks APDUs over `self.conn.transmit(apdu)`. APDU-level logic is transport-agnostic, but the connection underneath is the wired `uscard` connection.
- `git log --all --grep="NFC" -i` and `--grep="276"` → no commits. `git log --all -- f469-disco/usermods/scard/` → usermod history exists but no NFC additions.
- Related issues: `gh issue list --search "NFC"` → only #276 itself (no other NFC requests). No PRs targeting NFC.

**Current code**

- Smartcard transport (wired ISO-7816, contact): `src/keystore/javacard/util.py:15-23` (reader pin map), `f469-disco/usermods/scard/protocols.h:16-25` (T=0/T=1 only).
- Smartcard keystore: `src/keystore/memorycard.py:19-38` — `NAME = "Smartcard"`, `NOTE` mentions "PIN-protected external smartcard (requires devkit)" and "device can only operate when the smartcard is inserted" — confirms a seated/contact card model.
- Host classes (for context, not the NFC path): `src/hosts/{usb,qr,sd}.py` — USB/QR/SD are the *coordinator-facing* hosts; the smartcard is a *keystore*, not a host. NFC for the smartcard would be a keystore-layer change, not a host-layer change.
- No NFC reader driver, no ISO 14443 stack, no antenna/GPIO config for NFC anywhere in `f469-disco/usermods/` or `src/`.

**Reproduction / validation**

- VERIFIED (transport is wired): confirmed by reading the reader pin assignment (`util.py:15-23`) and the protocol enum (`protocols.h:16-25`). Not a runtime reproduction; a static-code finding.
- VERIFIED (no NFC code present): `grep -rni "contactless|NFC|RFID|ISO14443|T=CL"` across `f469-disco/usermods/scard/` and `src/keystore/` returns no source-code hits (only binary/PDF false positives under `docs/` and `shield/`, which are images/datasheets, not code).
- NOT INFERRED beyond the above.

**Related issues/PRs**

- #276 itself (OPEN, 2024-06-08, no labels).
- `gh issue list --search "NFC"` → no other NFC issues.
- The `specter-javacard` repo (referenced by the maintainer) is a separate repository holding the on-card applet; it is not vendored here, so its NFC-readiness could not be inspected in this audit.

**Implementation notes** (if relevant)

This is not a specter-diy firmware change alone; it needs coordinated work across three layers. Sketch of what "add NFC" would entail, from the device-firmware side that *this* repo owns:

1. **Add a contactless transport to the `uscard` usermod** (`f469-disco/usermods/scard/`): today `protocols.h:16-25` only defines T=0/T=1 (contact). An NFC path needs an ISO 14443 (T=CL) protocol wrapper and a reader driver for an NFC frontend (e.g. NXP PN532 or ST25R3911) driven over SPI/I2C/UART. This is the largest firmware-side component.
2. **Wiring/HAL config**: `src/keystore/javacard/util.py:15-23` hardcodes the contact reader's GPIO pins. Add a second reader instance (or make the reader selectable) for the NFC frontend's bus pins. The F469 discovery board exposes SPI/I2C/UART that can drive an external NFC shield.
3. **Connection selection in the keystore**: `src/keystore/memorycard.py:39` builds the applet over `get_connection()`; once two transports exist, the keystore should try contact, then contactless, or let the user choose.
4. **Upstream `specter-javacard` applet** (out of this repo): the on-card applet must be built for a contactless-capable JC card and its APDU interface must tolerate NFC framing/timings. This is the dependency the maintainer flagged in the issue.
[... cut ...]
Recommendation: FEATURE REQUEST — STILL VALUABLE, but the next action is not in this repo. The firmware-side prerequisites (a contactless transport in `f469-disco/usermods/scard/` and a second reader in `src/keystore/javacard/util.py`) are unblocked only once `specter-javacard` supports NFC. Suggest: (a) confirm with the `specter-javacard` maintainers whether NFC is on their roadmap; (b) if yes, file a companion tracking issue there and reference it here; (c) until then, leave #276 open as a low-priority feature request with an "upstream-blocked" label.