cryptoadvance/specter-diy #277: PSBT_TAP_KEY_SIG not supported

pythcoiner, 2024-06-22:
Signing w/ the taproot internal key looks not implemented, [cf here](https://github.com/diybitcoinhardware/f469-disco/blob/master/libs/common/embit/psbt.py#L353)

signing w/ taptree works fine, can be tested on liana w/ this [PR](https://github.com/wizardsardine/liana/pull/1103)

stepansnigirev, 2024-06-22:
When signing with internal key, embit finalizes the input right away. So instead of adding the signature to tap key signature field it creates final witness.
But I agree, we should implement handling this field anyway

pythcoiner, 2024-06-22:
will have a look!

stepansnigirev, 2025-11-05:
NOTE: implemented in embit (dev branch I guess) - need to test with specter-diy

pythcoiner, 2025-11-06:
see https://github.com/diybitcoinhardware/embit/commit/e5beacc11e58174fc11468ea0b420fe365ec1a5c

schnuartz-ai, 2026-08-29:
**Current assessment**

STILL RELEVANT in the vendored embit copy shipped with specter-diy, but the upstream fix exists and is small. The issue reports that signing with a Taproot **internal/key-path** key is not implemented: the device does not write the `PSBT_IN_TAP_KEY_SIG` (`0x13`) field, and the vendored embit has a `# TODO: 0x13 - tap key signature` at the parse site. Maintainer (stepansnigirev) confirmed in the issue (2024-06-22) that embit instead finalises the input right away (writes `final_scriptwitness`) and agreed the `0x13` field should be handled anyway. A later comment (2025-11-05) notes "implemented in embit (dev branch) — need to test with specter-diy", and pythcoiner linked the implementing commit `diybitcoinhardware/embit@e5beacc`. That commit is **not** present in the embit version vendored in this repo (the submodule is pinned to `db3ce3e` from 2023-12-03, predating the Nov-2025 fix).

**What I investigated**

- Fetched the full issue body and 4 comments. Title: "PSBT_TAP_KEY_SIG not supported". Body links the then-current `embit/psbt.py#L353` showing the missing `0x13` handling. stepansnigirev (2024-06-22): embit finalises immediately, "we should implement handling this field anyway". pythcoiner (2024-06-22): "will have a look!". stepansnigirev (2025-11-05): "implemented in embit (dev branch I guess) - need to test with specter-diy". pythcoiner (2025-11-06): linked `https://github.com/diybitcoinhardware/embit/commit/e5beacc11e58174fc11468ea0b420fe365ec1a5c`.
- Verified the vendored embit version: `f469-disco` submodule at `db3ce3e` ("update to latest embit", committed **2023-12-03**). `embit/` is **not** a nested submodule — it is vendored directly inside `f469-disco`, so it moves only when `f469-disco` is bumped.
- Confirmed the upstream fix commit `e5beacc` ("tap_key_sig", authored by pythcoiner) adds: (a) `self.taproot_key_sig = None` to `PSBTInput.__init__`; (b) a write path `ser_string(stream, b"\x13" + self.taproot_key_sig.serialize())` under a `# PSBT_IN_TAP_KEY_SIG` comment; (c) changes the key-path signing branch in `sign_input_with_tapkey` from leaving a `# TODO: maybe better to put into internal key sig field` comment to `inp.taproot_key_sig = sig`.
- Checked the vendored copy against that fix:
  - `f469-disco/libs/common/embit/psbt.py:353` — still `# TODO: 0x13 - tap key signature` immediately before the `0x14` (tap script sig) branch. **No `0x13` parse/write branch.**
  - `f469-disco/libs/common/embit/psbt.py` — `grep "taproot_key_sig"` → **zero hits**. The field does not exist in the vendored `PSBTInput`.
  - `f469-disco/libs/common/embit/psbtview.py:654` — still `# TODO: maybe better to put into internal key sig field`, i.e. the exact line `e5beacc` replaced. Key-path signing at `psbtview.py:631-657` (`sign_input_with_tapkey`) writes only `inp.final_scriptwitness = Witness([wit])` and never populates a `taproot_key_sig` field.
  - `psbtview.py:780-804` (`sign_input` taproot branch) serialises only `final_scriptwitness` (`\x08`) and tap script sigs (`\x14`); no `\x13` write.
- Device signing path that invokes this: `src/apps/wallets/manager.py:772-799` `sign_psbtview` → per input `keystore.sign_input(...)` (`src/keystore/ram.py:74-78`) or `w.sign_input(...)` (`src/apps/wallets/wallet.py:260-290`), both of which call `psbtv.sign_input(i, root, sig_stream, ...)`. So the gap in embit is reachable from the device signing flow.
- `git log --all --grep="277"` / `--grep="TAP_KEY_SIG"` / `--grep="tap_key"` → no commits. No PR references #277. The `f469-disco` submodule has not been bumped past `db3ce3e`.
- Related issues: `gh issue list --search "taproot"` → #393 (taproot descriptor shows wrong), #326 (taproot PSBT "Unknown wallet in inputs" but signs ok), #302 (SIGHASH_ALL flag appended to Schnorr sigs), plus #277. These are distinct issues but all touch the taproot signing path; #302 in particular concerns the same `sign_input_with_tapkey`/sighash handling area.

**Current code**

Key-path (internal-key) Taproot signing as it stands today (vendored embit):

- `f469-disco/libs/common/embit/psbt.py:346-389` — PSBT input field parser. Line 353 carries the open TODO and the `0x13` key-path-sig case is **absent**; the next handled key is `0x14` (tap script sig) at lines 355-362. So a `PSBT_IN_TAP_KEY_SIG` arriving in an incoming PSBT is dropped into `self.unknown` (lines 391-394) and never used.
- `f469-disco/libs/common/embit/psbt.py` — no `taproot_key_sig` attribute on `PSBTInput`; `grep` confirms absence.
- `f469-disco/libs/common/embit/psbtview.py:631-657` — `sign_input_with_tapkey`: computes the tweak with `key.taproot_tweak(inp.taproot_merkle_root or b"")` (line 644), signs, and at line 655 sets `inp.final_scriptwitness = Witness([wit])`. Line 654 retains `# TODO: maybe better to put into internal key sig field`. The signature is finalised inline rather than recorded as `0x13`.
- `f469-disco/libs/common/embit/psbtview.py:780-804` — `sign_input` taproot branch: serialises `\x08` final_scriptwitness (lines 797-799) and `\x14` tap script sigs (lines 801-803); no `\x13` emission.
- Device flow: `src/apps/wallets/manager.py:772-799` `sign_psbtview`; `src/keystore/ram.py:77-78` and `src/apps/wallets/wallet.py:276-290` both delegate to `psbtv.sign_input`.

**Reproduction / validation**

[... cut ...] -650`). No new private security detail.

Recommendation: STILL RELEVANT — the upstream fix (`embit@e5beacc`) exists but is not vendored; the three TODO sites named in the issue are still open in the copy shipped with specter-diy (`psbt.py:353`, `psbtview.py:654`). Recommend vendoring/bumping embit to include `e5beacc`, then running the on-device taproot key-path test the maintainer flagged as the next step; coordinate with #302 since both touch the same `sign_input_with_tapkey` sighash handling.