cryptoadvance/specter-diy #278: Stepan's gpg key is expired

mflaxman, 2024-08-09:
```$ gpg --verify sha256.signed.txt 
gpg: Signature made Thu May 30 08:01:08 2024 CDT
gpg:                using ECDSA key 6F16E354F83393D6E52EC25F36ED357AB24B915F
gpg: Good signature from "Stepan Snigirev (Specter release signing key) <snigirev.stepan@gmail.com>" [expired]
gpg: Note: This key has expired!
Primary key fingerprint: 6F16 E354 F833 93D6 E52E  C25F 36ED 357A B24B 915F
```

FYI for the next release @stepansnigirev

schnuartz-ai, 2026-08-29:
**Current assessment**

LIKELY FIXED (and superseded by a key rotation). The issue reported that Stepan Snigirev's GPG release-signing key (`6F16 E354 F833 93D6 E52E C25F 36ED 357A B24B 915F`, "Stepan Snigirev (Specter release signing key)") had expired as of the 2024-05-30 signature on `sha256.signed.txt`. This specific key is no longer used to sign releases: starting with the **v1.10.3** release (2026-05-01) the project rotates to a new "Specter Signer 2026" key controlled by k9ert (fingerprint `9DC3 3CA8 3058 9DE3 B322 5C26 EEF5 756B 2EA4 2349`), and the docs/SECURITY policy were updated accordingly. The v1.10.3 release notes explicitly state the rotation ("v1.10.3 rotates the release signing keys (Ben removed, k9ert added)").

**What I investigated**

- Fetched the full issue body (no comments, no labels). Title: "Stepan's gpg key is expired". Body: a `gpg --verify sha256.signed.txt` transcript showing Stepan's key `6F16…B24B915F` produced a "Good signature … [expired]" / "Note: This key has expired!" for a signature made 2024-05-30, with an "FYI for the next release @stepansnigirev" note.
- Verified the current release-signing arrangement in-repo:
  - `SECURITY.md:32-35` — "The release hash manifests (`sha256.signed.txt`) are signed with the 'Specter Signer 2026' key, controlled by k9ert, fingerprint `9DC3 3CA8 3058 9DE3 B322 5C26 EEF5 756B 2EA4 2349` ([Ubuntu keyserver](…))."
  - `SECURITY.md:21-24` — the release-signer table now lists **k9ert** (`ECC0 B4AB D74E 716F 5ADE 0952 28B3 58A8 843B 0109`) and **Mike Tolkachev / miketlk** (`F2DB C4C6 14C1 13E2 B15F 879A DD5C 1264 EBD6 45BE`). Stepan's key is **not** in the table.
  - `docs/quickstart.md:10` — "Since v1.10.3 it is signed with the **'Specter Signer 2026'** key, controlled by k9ert (fingerprint `9DC3 3CA8 3058 9DE3 B322 5C26 EEF5 756B 2EA4 2349` …). Older releases were signed with [Stepan's PGP key](https://stepansnigirev.com/ss-specter-release.asc)."
  - `docs/security-model.md:98` — still links to Stepan's public key for older releases; not a regression, just legacy reference.
- Verified the actual release:
  - `gh release list` → latest is **v1.10.3** (2026-05-01), authored by **k9ert**. Assets include `initial_firmware_v1.10.3.bin`, `specter_upgrade_v1.10.3.bin`, and `sha256.signed.txt`.
  - v1.10.3 release notes: "v1.10.3 rotates the release signing keys (Ben removed, k9ert added) and ships an updated bootloader that embeds the new key set. Future releases will be signed with the new keys …". Listed PRs include `chore: Remove ben and add k9ert keys for releasing by @k9ert in #306` and `chore: Update bootloader submodule to latest by @k9ert in #350`.
  - So as of the latest release, `sha256.signed.txt` is signed by the new "Specter Signer 2026" key, not by Stepan's expired key. The exact complaint in #278 ("Stepan's key is expired") is therefore no longer applicable to current releases.
- `git log --all --grep="gpg" -i` / `--grep="signing key"` / `--grep="278"` → no commits referencing this issue directly. The key rotation landed via the bootloader submodule bump + #306/#350, not via a #278-specific change.
- Related/duplicate issues:
  - #185 (CLOSED, 2021-10-07, "gpg signing key expired?") — the *same* recurring problem (Stepan's key `6F16…` expired), earlier occurrence. Closed without a code change.
  - #291 (OPEN, 2025-01-22, "gpg signing key expired?") — reports a **different** key expired: "Specter Signer <noreply@specter.solutions>" fingerprint `785A 2269 EE3A 9736 AC1A 4F4C 864B 7CF9 A811 FEF7`, signature dated 2024-05-30. This is the pre-rotation "Specter Signer" key, not Stepan's `6F16…` key and not the new `9DC3…` "Specter Signer 2026" key. #291 is the one still potentially relevant (see below), not #278.

**Current code**

No application/firmware code is involved; this is a release-process/key-management issue. The relevant artefacts are documentation and release assets:

- `SECURITY.md:21-24` — current release-signer table (k9ert, miketlk; Stepan absent).
- `SECURITY.md:32-35` — current signing key for `sha256.signed.txt` ("Specter Signer 2026", k9ert, `9DC3 3CA8 3058 9DE3 B322 5C26 EEF5 756B 2EA4 2349`).
- `docs/quickstart.md:10` — user-facing verification instructions reflecting the v1.10.3 key change.
- `docs/security-model.md:98` — legacy reference to Stepan's key for old releases (historical, not current).
- Release v1.10.3 assets + notes (`gh release view v1.10.3`) — the rotation is live.

**Reproduction / validation**

- VERIFIED (new key documented): read `SECURITY.md:21-35` and `docs/quickstart.md:10`; both name the "Specter Signer 2026" key (k9ert, `9DC3…2349`) as the current signer.
- VERIFIED (rotation shipped): `gh release view v1.10.3` confirms the release is by k9ert and the notes state the key rotation and the bootloader embedding the new key set; relevant PRs #306 and #350 are listed.
- NOT INDEPENDENTLY VERIFIED: I did not run `gpg --verify` against the v1.10.3 `sha256.signed.txt` asset to confirm the signature is made with `9DC3…2349` and that that key is currently unexpired. The conclusion "Stepan's expired key is no longer used" is established from the docs and the release notes, not from a live GPG verification. A final close-out could verify the asset directly.
- INFERRED: the new "Specter Signer 2026" key itself is presumably valid through 2026+ given its name and the 2026-05 release; not independently checked.

**Related issues/PRs**

- #278 itself (OPEN, 2024-08-09, no labels).
[... cut ...]
Recommendation: LIKELY FIXED — the v1.10.3 release (2026-05-01) rotated the signing key to "Specter Signer 2026" (k9ert, `9DC3 3CA8 3058 9DE3 B322 5C26 EEF5 756B 2EA4 2349`), documented in `SECURITY.md:32-35` and `docs/quickstart.md:10`; Stepan's expired key (`6F16…B24B915F`) is no longer used. Suggest closing #278 as fixed, and re-evaluating the related-but-duplicate #291 (which concerns a different expired key `785A…`) against v1.10.3. Recommend a live `gpg --verify` of the v1.10.3 `sha256.signed.txt` as the final confirmation before closing.

Schnuartz, 2026-08-31:
Closing this as fixed.

The expired signing key reported in this issue (`785A 2269 EE3A 9736 AC1A 4F4C 864B 7CF9 A811 FEF7`) is no longer used for current releases.

Since **v1.10.3**, release hash manifests (`sha256.signed.txt`) are signed with the new **"Specter Signer 2026"** key:

`9DC3 3CA8 3058 9DE3 B322 5C26 EEF5 756B 2EA4 2349`

This is documented in `SECURITY.md` and `docs/quickstart.md`. The signing-key rotation was implemented through #306 and the subsequent bootloader update in #350.

The v1.10.3 `sha256.signed.txt` has also been independently verified with `gpg --verify`, producing a valid signature from the new "Specter Signer 2026" key.

Historical releases may of course still show the old key as expired when verified today, but current releases no longer use that key.

**Verified fixed, closing.**

closed 2026-08-31 by Schnuartz