cryptoadvance/specter-diy #279: Turn off SD card not respected

pvig0q1s, 2024-08-21:
Firmware 1.9.0 is not respecting the setting to turn off SD card. Have not tested with other versions.

Reproduce:

- Device Settings > Communication > SD card > Enabled SD card off
- Main Menu > Load key   _OR_   Main Menu > Import recovery phrase > Open SD card file,  both give the option to load a seed from SD.

On this note, for the second option (Main Menu > Import recovery phrase > Open SD card file), the 'Select a file' screen shows: 

.PSBT FILES - NO FILES
.TXT FILES
	seedword.txt
.JSON FILES - NO FILES

From a users perspective it is weird that these filetypes are shown here.

Azzbo77, 2025-12-23:
I think the main issue is solved with https://github.com/cryptoadvance/specter-diy/pull/338.

schnuartz-ai, 2026-08-30:
**Current assessment**

PARTIALLY FIXED. The issue reported two ways to load a seed from SD even after disabling "SD card" in *Device Settings > Communication*:
1. *Main Menu > Load key* — the `SDKeyStore` "Load key" button and its file picker.
2. *Main Menu > Import recovery phrase > Open SD card file* — the `SDHost` "Open SD card file" communication button.

A contributor comment points to PR #338 (merged 2026-01-25, commit `6bbec30`) as the fix. PR #338 added `and host.is_enabled` to the host filter in `import_mnemonic()` — this **fixes path 2** (the "Open SD card file" entry in the *Import recovery phrase* menu is now hidden when SD is disabled). It does **not** touch path 1: the `SDKeyStore` "Load key" button and its `select_file()` picker are driven by `platform.sdcard.is_present` (physical card presence), not by the `SDHost.is_enabled` setting, so they still offer SD-card seed files when the card is inserted but "disabled". The secondary UX complaint — that the *Import recovery phrase > Open SD card file* picker lists `.psbt`/`.txt`/`.json` extensions (so e.g. `.PSBT FILES` shows up when choosing a seedword) — is also unchanged by PR #338.

**What I investigated**

- Fetched the full issue (#279, OPEN, 2024-08-21, no labels, one comment). Body: firmware 1.9.0 ignores the "Enabled SD card off" setting; both *Load key* and *Import recovery phrase > Open SD card file* still allow loading a seed from SD; and the *Select a file* screen for the second path shows `.PSBT FILES - NO FILES`, `.TXT FILES` (seedword.txt), `.JSON FILES - NO FILES`, which the reporter finds odd.
- Comment (Azzbo77, CONTRIBUTOR, 2025-12-23): "I think the main issue is solved with #338."
- Inspected PR #338: `gh pr view 338` → title "Fix #329: Hide disabled QR scanner in recovery phrase import menu"; one file changed (`src/specter.py`, +1/-1); merged 2026-01-25. The diff modifies the `import_mnemonic()` host-menu list comprehension to filter `host.is_enabled`. PR #338's stated scope is the QR scanner (#329), but the same line covers `SDHost` for path 2 of #279.
- Verified the fix is present in audit head `3f3c831`: `git log -1 --format="%H %s" -L 301,301:src/specter.py` → `6bbec30 Fix #329: Hide disabled QR scanner ... (#338)`.
- Traced path 1 (unchanged): `initmenu()` builds the "Load key" button from `self.keystore.is_key_saved and self.keystore.load_button` (`src/specter.py:258-259`), with no `SDHost.is_enabled` gate; selecting it calls `self.keystore.load_mnemonic()` (`src/specter.py:280`). For `SDKeyStore`, `load_button = "Load key"` (`src/keystore/sdcard.py:26`), `is_key_saved` checks `platform.sdcard.is_present` (`src/keystore/sdcard.py:99`), and `select_file()` gates the SD section on `platform.sdcard.is_present` (`src/keystore/sdcard.py:139`) — all physical-presence checks, none consult the disabled setting.
- Traced path 2 (fixed): `import_mnemonic()` (`src/specter.py:299-302`) now filters `[(host, host.button) for host in self.hosts if host.is_enabled]`; `SDHost.button = "Open SD card file"` (`src/hosts/sd.py:16`); `Host.is_enabled` reads `self.settings.get("enabled", True)` (`src/hosts/core.py:63-65`). The setting UI itself states its scope: "This setting will completely enable or disable this communication channel and remove corresponding button from the main menu" (`src/hosts/core.py:90`) — i.e. it is documented as a *communication-channel* toggle, not a keystore-storage toggle.
- Traced the file-extension UX: `SDHost.get_data()` calls `self.select_file([".psbt", ".txt", ".json"])` (`src/hosts/sd.py:49`), and `select_file()` renders one section per extension regardless of caller context (`src/hosts/sd.py:71-100`). So the recovery-phrase import path inherits the generic PSBT/TXT/JSON picker.
- `git log --oneline --all -- src/keystore/sdcard.py` and `-- src/hosts/sd.py`: no commit referencing #279 or an SD-disable gate on the keystore side; the only recent SD-relevant change is #269 (overwrite prompt) and #338 (host filter, on `src/specter.py`).
- Related issues: `gh issue list --search "SD card"` → #44 (OPEN, "SD card support"), #232 (OPEN, "Password encrypted seed export to SD card"), #233 (OPEN, "Suggest user to format SD card"), #279 (this), #359 (OPEN, "SD Card Transaktion Data and Wallet Descriptors cannot be deleted"). #329 (CLOSED) is the QR-scanner twin of path 2 that #338 explicitly fixed.

**Current code**

- `src/specter.py:258-259` — path-1 button, no `is_enabled` gate:
  `if self.keystore.is_key_saved and self.keystore.load_button: buttons.append((2, self.keystore.load_button))`
- `src/specter.py:280` — path-1 handler: `res = await self.keystore.load_mnemonic()`
- `src/specter.py:301` — path-2 host filter (PR #338 fix): `buttons=[(host, host.button) for host in self.hosts if host.is_enabled]`
- `src/keystore/sdcard.py:26` — `load_button = "Load key"`
- `src/keystore/sdcard.py:99` — `if not platform.sdcard.is_present: return flash_exists` (physical presence, not setting)
- `src/keystore/sdcard.py:139` — `if platform.sdcard.is_present:` gates the SD section of `select_file()`
- `src/hosts/sd.py:16` — `button = "Open SD card file"`
- `src/hosts/sd.py:49` — `sd_file = await self.select_file([".psbt", ".txt", ".json"])` (generic extensions shown for recovery import)
- `src/hosts/sd.py:86-97` — `select_file()` builds a section per extension with a "No files" fallback
- `src/hosts/core.py:63-65` — `@property def is_enabled(self): return self.settings.get("enabled", True)`
[... cut ...]
Recommendation: PARTIALLY FIXED — PR #338 fixed path 2 (the *Import recovery phrase > Open SD card file* entry is now hidden when SD is disabled, `src/specter.py:301`), but path 1 (the `SDKeyStore` "Load key" button and SD section of its file picker, `src/specter.py:258-259` / `src/keystore/sdcard.py:131-145`) still operates on physical card presence only and ignores the disable setting; the `.psbt`/`.txt`/`.json` extension listing in the recovery-phrase picker (`src/hosts/sd.py:49`) is also unchanged. Suggest gating the `SDKeyStore` SD options on the same `enabled` setting (or a shared one) and narrowing the import extension list; then close.