cryptoadvance/specter-diy #280: Order of numerical bit values

pvig0q1s, 2024-08-21:
Order of bit values (when fixing seedwords through _Main Menu > Generate new key > click-a-word_) should be changed.

Currently, when using the feature two lines are shown under the seed word list:
	1   2   4   8  16  32
	64  128  256  512  1024
	
It should be like this:
	1024  512  256  128  64
	32   16   8   4   2   1
	
In practice, when a user activates a numerical field (when tossing coins for example) the order of input SHOULD not matter, but since we read binary from right to left the current positioning of the value fields feels weird. 

(Might be personal preference though).

schnuartz-ai, 2026-08-30:
**Current assessment**

STILL RELEVANT (minor UX, by the reporter's own "might be personal preference" caveat). The bit-value toggle keyboard in the *Generate new key > click-a-word* mnemonic-fixing screen still renders the values in **ascending** order — exactly the layout the issue asks to reverse. The button map at `src/gui/screens/mnemonic.py:93-94` is `["1", "2", "4", "8", "16", "32", "\n", "64", "128", "256", "512", "1024", ""]`, i.e. row 1 = `1 2 4 8 16 32`, row 2 = `64 128 256 512 1024`. The requested layout is the descending mirror: row 1 = `1024 512 256 128 64`, row 2 = `32 16 8 4 2 1`. The numeric interpretation code already iterates with `reversed(bits)` (`src/gui/screens/mnemonic.py:144-149`), so reversing only the *display* map (and keeping the index↔bit mapping consistent) is the whole change. No git history shows this was ever touched.

**What I investigated**

- Fetched the full issue (#280, OPEN, 2024-08-21, no labels, no comments). Body: in the click-a-word fixer, two lines are shown under the word list — `1 2 4 8 16 32` / `64 128 256 512 1024`; the reporter wants `1024 512 256 128 64` / `32 16 8 4 2 1` so that, when "tossing coins", the right-to-left binary reading order matches the visual field order. Reporter concedes "Might be personal preference though."
- Located the screen: the mnemonic-fix keyboard is built in `NewMnemonicScreen.__init__` (`src/gui/screens/mnemonic.py:91-102`); the button map is hardcoded at `src/gui/screens/mnemonic.py:93-94`.
- Confirmed the value computation: `change_word`'s toggle callback (`src/gui/screens/mnemonic.py:138-159`) reads `bits = [obj.get_btn_ctrl(i, lv.btnm.CTRL.TGL_STATE) for i in range(11)]` then `for i, bit in enumerate(reversed(bits)): num = num << 1; if bit: num += 1`. So button index 0 (currently labelled "1") is treated as the **least-significant** bit: `reversed(bits)` puts index 0 last, and the loop shifts left each step, so index 0 contributes `2^0`, index 1 → `2^1`, … index 10 → `2^10`. The labels already match this (index 0 = "1" = 2^0).
- `git log --oneline --all -- src/gui/screens/mnemonic.py`: most recent touch is `195a50c` (LVGL 9.x migration); `git log --all -S "1024" -- src/gui/screens/mnemonic.py` → only `dd77c90` (original "add user entropy for mnemonic generation") — the bit map has never been reordered.
- `gh issue list --search "bit"` / `--search "seedword"`: no other issue covers this ordering; #288 (CLOSED, "Issue with False Seed Words from Compact Seed QR") is a different SeedQR-decoding bug, not the bit-keyboard layout.

**Current code**

- `src/gui/screens/mnemonic.py:93-94` — the button map (ascending): `self.kb.set_map(["1", "2", "4", "8", "16", "32", "\n", "64", "128", "256", "512", "1024", ""])`
- `src/gui/screens/mnemonic.py:95` — `self.kb.set_ctrl_map([lv.btnm.CTRL.TGL_ENABLE for i in range(11)])` (11 toggle buttons)
- `src/gui/screens/mnemonic.py:133-136` — initial state from the word index: `self.kb.set_ctrl_map([lv.btnm.CTRL.TGL_ENABLE | (lv.btnm.CTRL.TGL_STATE if ((word_idx>>i)&1) else 0) for i in range(11)])` (bit `i` of `word_idx` maps to button index `i`)
- `src/gui/screens/mnemonic.py:144-149` — reconstruction: `bits = [obj.get_btn_ctrl(i, ...) for i in range(11)]; for i, bit in enumerate(reversed(bits)): num = num << 1; if bit: num += 1` (index 0 = LSB)

**Reproduction / validation**

- VERIFIED (by code reading): the displayed order at `src/gui/screens/mnemonic.py:93-94` is ascending `1…1024`, matching what the issue says is shown and asks to reverse. The reporter's screenshots are not in the issue, but the hardcoded map is unambiguous.
- INFERRED (not executed): reversing *only* the `set_map` strings to `["1024", "512", "256", "128", "64", "\n", "32", "16", "8", "4", "2", "1", ""]` would **break** the value computation, because the index→significance binding at lines 133-136 and 144-149 is positional (index 0 = bit 0 = 2^0) and would no longer match the relabelled "1024" at index 0. A correct fix must either (a) reverse the map **and** reverse the index↔bit mapping together, or (b) keep the map order but reverse only the visual layout. Not run on hardware/simulator.

**Related issues/PRs**

- #288 (CLOSED) — "Issue with False Seed Words from Compact Seed QR" (different root cause: SeedQR decoding, not the bit keyboard).
- No PR or commit touches this button map beyond its original introduction (`dd77c90`).

**Implementation notes** (if relevant)

The safe, minimal change is to reverse the map **and** the index significance together so the positional logic stays self-consistent. Concretely in `src/gui/screens/mnemonic.py`:

1. Reverse the label map (lines 93-94) to:
   `["1024", "512", "256", "128", "64", "\n", "32", "16", "8", "4", "2", "1", ""]`
2. Reverse the bit index used in the initial-state build (lines 133-136) so that the now-leftmost button "1024" is bit 10: replace `((word_idx>>i)&1)` with `((word_idx>>(10-i))&1)`.
3. Reverse the reconstruction (lines 144-149): iterate `bits` in forward order (drop `reversed`), since button index 0 is now the MSB — `for i, bit in enumerate(bits): num = num << 1; if bit: num += 1`.

Equivalently, keep the positional logic and only relabel, but then the displayed numbers must stay in the current index order (index 0 = "1"); that contradicts the request, so option (1–3) is the correct interpretation. A simulator test (`make simulate` > *Generate new key* > click a word > toggle "1024" alone) should yield a word whose wordlist index is 1024 to confirm.
[... cut ...]  above guard against that. No new vulnerability.

Recommendation: STILL RELEVANT — the bit-value keyboard at `src/gui/screens/mnemonic.py:93-94` still shows ascending `1…1024`; the requested descending layout is a small UX improvement (and the reporter flags it as possibly personal preference). If implemented, the label reversal must be paired with matching reversals of the index↔bit logic at `src/gui/screens/mnemonic.py:133-136` and `144-149`; otherwise the wrong word is selected. Low priority.

Schnuartz, 2026-08-30:
Yes, you are right. That makes sense to implement that way.

It was something I thought it should be changed as well.