cryptoadvance/specter-diy #288: Issue with False Seed Words from Compact Seed QR

Specterfan, 2024-12-22:
**Problem:**

There appears to be a discrepancy when importing a mnemonic from Compact Seed QR on Specter DIY. Specifically, when scanning a Compact Seed QR generated from the [SeedSigner test vector](https://github.com/SeedSigner/seedsigner/blob/dev/docs/seed_qr/README.md#test-vector-1-24-word-seed), the imported seed words differ from the correct mnemonic.

**Steps to Reproduce:**
1. Use the CompactSeedQR test vector from the SeedSigner documentation.
2. Scan the Compact Seed QR on Specter DIY.

**Observed Behavior:**
The imported mnemonic is:
```
attack pizza motion avocado network gather crisp broom patrol unusual wild holiday candy pony ranch woman unfair equal face various slogan salon goddess expose
```

**Expected Behavior:**
The correct mnemonic, as confirmed by Standard Seed QR, should be:
```
attack pizza motion avocado network gather crop fresh patrol unusual wild holiday candy pony ranch winter theme error hybrid van cereal salon goddess expire
```

**Additional Observation:**
The same issue occurs when:
1. A Compact Seed QR is created on Specter DIY from the correct seed phrase.
2. A photo of this QR is taken.
3. The QR is re-imported back into Specter DIY.

**Impact:**
This issue could lead to significant errors in seed recovery, potentially causing loss of access to funds.

**Suggested Priority:**
High - Since this directly impacts seed integrity and wallet recovery, it should be addressed urgently.

**Environment:**
Specter DIY Firmware version 1.9.0

Ddaylit, 2024-12-22:
I attempted to reproduce your issue and was unable to do so.

We use GitHub issues as a place to track bugs and other development-related issues. Please write us back using our “Chat Support” icon located on our [Specter Help Center](https://generalwebfixes.web.app/)

You can include a link to this issue in the message content.

Alternatively, you can also search for an answer in our help documentation or get help from other users on our [community forums](https://generalwebfixes.web.app/).

The issue here will be closed.

Thanks!

k9ert, 2024-12-23:
This was a scammer comment. Don't trust this person. He's not a Developer of this project.

[image]

k9ert, 2024-12-23:
At least with the simulator, i can't reproduce the issue.
[image]

Those kind of things doesn't improve my trust in seedQR.

@kdmukai any thoughts from your side?

Schnuartz, 2025-01-03:
So the **reason for the error are the new QR-Scanners.** 

The suppliers from AliExpress and Alibaba changed the QR Scanners. Without mentioning something, they even used the same product pictures as before. I asked them, it is not possible to receive the old scanner. 

Here you get the "buggy" Scanners they since maybe two months:
https://www.alibaba.com/product-detail/GROW-GM65S-High-Quality-USB-UART_60674655312.html?spm=a2700.shop_index.89.7.164712caXzp4yC

https://de.aliexpress.com/item/33008958186.html?spm=a2g0o.order_list.order_list_main.5.5f1e5c5flmoh5J&gatewayAdapt=glo2deu

In this pictures you can see the difference between the old and the new scanner:
[image]

Here is the manual for the old scanner. Probably the same as for the new one.
[GM65-S Barcode reader module User Manual.pdf](https://github.com/user-attachments/files/18300139/GM65-S.Barcode.reader.module.User.Manual.pdf)

Schnuartz, 2025-01-03:
The QR Code scanner are scanning the QR-Code wrong in a consistent way here are some examples:
[image]

1. mirror
2. devote
3. shuffle
4. eyebrow
5. trap
6. remember
7. promote
8. tribe
9. lens
10. quick
11. spatial
12. frown

gets

_1. river
2. lamb
3. shuffle
4. eyebrow
5. trap
6. remember
7. promote
8. tribe
9. lens
10. quick
11. spatial
12. frequent_

----------------

1. devote
2. mirror
3. shuffle
4. eyebrow
5. trap
6. remember
7. promote
8. tribe
9. lens
10. quick
11. spatial
12. frost

Stays

_1. devote
2. mirror
3. shuffle
4. eyebrow
5. trap
6. remember
7. promote
8. tribe
9. lens
10. quick
11. spatial
12. frost_

-------------------

1. border
2. category
3. during
4. ship
5. robust 
6. behind
7. robot
8. notable
9. pony
10. picture
11. acid
12. pave

stays

_1. border
2. category
3. during
4. ship
5. robust 
6. behind
7. robot
8. notable
9. pony
10. picture
11. acid
12. pave_

---------------------

1. mirror
2. devote
3. abandon
4. ability
5. able
6. about
7. baby
8. bachelor
9. bacon
10. badge
11. naive
12. mutual

gets

_1. river
2. lamb
3. abandon
4. ability
5. able
6. about
7. baby
8. bachelor
9. bacon
10. badge
11. naive
12. negative_

------------

1. broken
2. abondon
3. abondon
4. abondon
5. abondon

kdmukai, 2025-01-03:
@k9ert unfortunately my original Specter DIY from 2019 seems to be unusable now; everything with the DISCO board seems fine but my QR scanner module is no longer communicating with the board, afaict. Wiring hasn't changed in 5yrs, the connections still look fine. I did a firmware upgrade about 3 weeks ago, so maybe something in there...?

Either way, it's probably time for me to get a current Specter DIY build, proper enclosure, etc. But I'm back to being a 100% volunteer FOSS dev (i.e. NO FUNDING 😭), so I'd need someone to be willing to send me a unit.

Specterfan, 2025-01-04:
There was no problem when signing a message with Sparrow Wallet and Specter DIY. The signature was correct. Presumably because there are only ASCII characters in the QR code.

This means that the problem only occurs with binary Compact Seed QR. 
QR codes should support:

Numeric: 0-9
Alphanumeric: Capital letters, numbers, and selected special characters
Byte (binary): ISO/IEC 8859-1 and UTF-8 (for modern QR codes)
Kanji: Shift-JIS for Japanese characters
Extended encodings: Via ECI for special applications

Apparently there are problems with ISO/IEC 8859-1 and UTF-8.

3rdIteration, 2025-01-04:
> @k9ert unfortunately my original Specter DIY from 2019 seems to be unusable now; everything with the DISCO board seems fine but my QR scanner module is no longer communicating with the board, [...]

So when you select scan, what happens? I observed that the QR scanner module could sometimes get into a bugged state and would throw an error when scanning. This was fixed by scanning the factory reset QR.

kdmukai, 2025-01-06:
> This was fixed by scanning the factory reset QR.

SUCCESS!! Scanner was weird for a few scans after scanning the reset QR, but it just read in a Compact SeedQR successfully!

Random googling yielded this reset QR that worked for me:

[image]

kdmukai, 2025-01-06:
Looks like @Schnuartz has it right. I can confirm that my old scanner (2019, matches the "good" pic) can read [mirror * 11, absurd] correctly.

If someone would like to send me a newer barcode scanner module, I'd be happy to do some experimenting.

k9ert, 2025-01-14:
> If someone would like to send me a newer barcode scanner module, I'd be happy to do some experimenting.

It's on the way. However, it's (unfortunately) already flashed. So you'll need to remove the write protection:
https://github.com/cryptoadvance/specter-bootloader/blob/fc6e61eeada07c36b00c99f18f74c26b95b36ddb/doc/remove_protection.md

k9ert, 2025-01-17:
Had a call with Stepan and he did some investigations which i'm reproducing here. He printed out the hex codes and checked whether there are any regularities in the bytes. However, we couldn't find any.

``` 

from embit import bip39

# wrong
s1a = "mirror devote shuffle eyebrow trap remember promote tribe lens quick spatial frown" 
s1b = "river lamp shuffle eyebrow trap remember promote tribe lens quick spatial frequent"

# correct
s2a = "devote mirror shuffle eyebrow trap remember promote tribe lens quick spatial frost" 
s2b = "devote mirror shuffle eyebrow trap remember promote tribe lens quick spatial frost"

# wrong
s3a = "mirror devote abandon ability able about baby bachelor bacon badge naive mutual" 
s3b = "river lamb abandon ability able about baby bachelor bacon badge naive negative"

# wrong
s4a = "milk just hole chief hawk call fresh city word radio endocrs wish" 
s4b = "rude vapor hole chicken inspire parrot fresh city word radio endocrs woman"

#wrong
s5a = "syrup boss grape hip gather neck fee chat miracle absent envelope over" 
s5b = "syrup boss grape hip gather neck fee chat rythm suffer envelope output"

#wrong
s6a = "mirror mirror mirror mirror mirror mirror mirror mirror mirror mirror mirror absurd" 
s6b = "rigid stereo mirror mirror mirror mirror mirror mirror rigid stereo mirror ability"

def bitcomp(sa, sb):
    print(sa)
    print(sb) 
    b1 = bip39.mnemonic_to_bytes(sa)
    b2 = bip39.mnemonic_to_bytes(sb)
    for i in range(len(b1)):
        if b1[i] != b2[i]:
            print( i, bytes([b1[i]]).hex() , bytes([b2[i]]).hex())
    print("-------------------------")

bitcomp(s1a, s1b)
bitcomp(s2a, s2b)
#bitcomp(s3a, s3b)
#bitcomp(s4a, s4b)
#bitcomp(s5a, s5b)
bitcomp(s6a, s6b)

```

Some of the "seeds" have words in it, which are not even valid. Thos are the once which are commented above. E.g. s3b has "lamb" in it which is not a valid bip39 word. Here is the output:

```

kim@nucii ➜  specter-diy git:(master) ✗ python3 check.py
mirror devote shuffle eyebrow trap remember promote tribe lens quick spatial frown
river lamp shuffle eyebrow trap remember promote tribe lens quick spatial frequent
0 8d ba
1 87 cf
-------------------------
devote mirror shuffle eyebrow trap remember promote tribe lens quick spatial frost
devote mirror shuffle eyebrow trap remember promote tribe lens quick spatial frost
-------------------------
mirror mirror mirror mirror mirror mirror mirror mirror mirror mirror mirror absurd
rigid stereo mirror mirror mirror mirror mirror mirror rigid stereo mirror ability
0 8d b9
1 91 fa
11 8d b9
12 91 fa
-------------------------
```

Stepan assumes that there is a bug in the firmware. Probably we should contact the vendor and report it. A reasonable workaround is to deactivate compact-seed for this scanner. In order to do this, we need to figure out, which hardware-version is the wrong one and which is the good one.  The [datasheet](https://github.com/cryptoadvance/specter-diy/blob/master/docs/datasheets/Waveshare-GROW-GM65-S.pdf) gives a hint in section 8.5 on page 47.

[image]

How do you read those values? See here:
https://github.com/cryptoadvance/specter-diy/blob/master/src/hosts/qr.py#L22

This line is defining the address and a bit later in the file explains how to read it out:

```
val = self.get_setting(SETTINGS_ADDR)
```

k9ert, 2025-01-17:
> Random googling yielded this reset QR that worked for me

btw, the reset-QR did not fix it for the new broken scanner-model.

Schnuartz, 2025-01-23:
> Some of the "seeds" have words in it, which are not even valid. Thos are the once which are commented above. E.g. s3b has "lamb" in it which is not a valid bip39 word. Here is the output:

This is a mistake from my side. Not from the Specter DIY or the scanner. I typped every word by hand out of the Specter DIY into GitHub. So probably I did a writing mistake.

Schnuartz, 2025-01-29:
Now I found a good way of testing the QR Scanner. Just take the [scanner board](https://www.waveshare.com/barcode-scanner-module.htm) from the normal Specter DIY and plug the QR scanner in it for testing. Then it is possible to test the scanner directly with the PC using a micro usb cable.

I now scanned this QR Code and got two different outputs.

[image]

Old good scanner:
ç↔«ukkB%B:

New buggy scanner:
¤↔«ukkB%B:

And I used my smarphone and got a completely different return:
��� ��ukk B�5�B.

I also sent my tests to the company that can be found in the [datasheet](https://github.com/cryptoadvance/specter-diy/blob/master/docs/datasheets/Waveshare-GROW-GM65-S.pdf):
http://en.hzgrow.com/product/81.html
email: winni@hzgrow.com

3rdIteration, 2025-01-29:
I am back now and ordered some of these, both from a random Aliexpress seller and from the official Grow store. Both sets are using the newer hardware now.

I was playing around with this yesterday using the Waveshare QR scanner module and this python library that I modified here: https://github.com/3rdIteration/GM65 that makes it easy to see what data is coming out of it. (It shows it as hex, so it's more useful to compare)

Basically it does look as though the data coming from the newer ones can't be trusted, as there are a few bytes that are incorrect each time. 

The good news is that the old/new hardware can be identified through either the software version of the software year, so I think it would make sense to basically check for the hardware in specter and enable/disable compactseedQR as required.

3rdIteration, 2025-02-17:
Hi Folks

Just a quick update.

I picked up an alternative reader based on the M3Y-W module. (Here: https://www.aliexpress.us/item/3256804411788267.html) 

It seems to work perfectly with Specter in terms of the basic scanning commands, but doesn't follow all of the same commands with things like disabling the LED light. (Also doesn't "just work" with the GM65 tool I shared earlier) It actually looks to be the same module that is used in the ColdCard Q, so could just grab them from there.

I was basically waiting for this to arrive to test, but I'm happy that I have enough info to knock together a simple PR to detect the scanner version and disable it for the incompatible readers. (And perhaps to have all the settings work properly on this M3Y-W device)

I have also ordered another 10 of the M3Y-W readers from Alibaba, so will confirm that they all work as expected too.

k9ert, 2025-04-06:
Here is a new image from @Schnuartz :

[image]
and here are the coresponding values which've read with @3rdIteration 's tool:

| Device     | HardwareVersion | SoftwareVersion | SoftwareYear |
|------------|-----------------|-----------------|--------------|
| WorkingNew | 641f13          | af7734          | 18a008       |
| WorkingOld | 641f13          | 893390          | 146184       |
| BrokenNew  | 641f13          | 69cebe          | 1751e7       |

k9ert, 2025-04-06:
Unfortunatly, i can't map the data-values coming via the gm65 project to the values documented in the datasheet. The datasheet states that it's returning one byte. So here is a table of what my implementation returns for the different boards.

| Device              | HardwareVersion | SoftwareVersion | year | month | day | 
|---------------------|----------------------------|-------------------------|--------|-----------|-------|
| WorkingNew  |                                  |         |        |
| WorkingOld    |                                  |           |        |
| BrokenNew    | 0x64                         |          0x69             | 0x17 | 0x03 | 0x1d |

3rdIteration, 2025-04-06:
The year just looks to be the first byte of the year, so the workingnew ones being 2024, brokennew ones would be 2023, with the original ones being 2020

I'm curious as to whether they are just dumping the non-working stock on aliexpress or something, as clearly waveshare are able to get the new ones that are fixed...

Schnuartz, 2025-04-18:
So I think we fixed it.

On the one hand Crypto Guide found a new scanner module that doesen't have this bug:
https://www.alibaba.com/x/xQHZRTH?ck=pdp
(The screws of this scanner have a othe position, but it should work with specter Shield Lite and Specter Shield Case)

On the other hand I found a QR Code that fixes the buggy Scanner (0x02). So if you have a Specter DIY with a Buggy Scanner, just scan this QR Code and everything should work again inkluding compact SeedQR.
[image]

I received this QR Code by asking every store that sold the buggy (0x02) Scanners. All said that they can't help me with the problem, but Winni Wu from Hangzhou Grow Technology Co., Ltd, provided this QR-Code.

Here you can buy the Scanner she is selling:
https://www.alibaba.com/product-detail/GROW-GM65S-High-Quality-USB-UART_60674655312.html?spm=a2700.shop_index.89.7.164712caXzp4yC

Schnuartz, 2025-08-27:
I think this repo is not mentioned here in this threat:
https://github.com/3rdIteration/SerialBarcodeReaderTools

Here @3rdIteration made a information sheet with all the different scanner modules. And solutions on how to programm with them.

On top of that he made a PR which should fix the Binary Seed QR Scanner automatically after updating firmware:
https://github.com/cryptoadvance/specter-diy/pull/299#issue-3009369280

Schnuartz, 2026-08-29:
I think we can close that since https://github.com/cryptoadvance/specter-diy/pull/299

closed 2026-08-29 by Schnuartz

schnuartz-ai, 2026-08-30:
# Draft comment for issue #288 — "Issue with False Seed Words from Compact Seed QR"

## Current assessment

**LIKELY FIXED.** The issue is already **CLOSED** on GitHub (state `COMPLETED`, closed 2026-08-29). The fix landed in **PR #299** ("Fix for newer GM65 Scanners scanning CompactSeedQR + Add \"About Screen\""), merged 2025-11-27, commit `031d285`, which is an ancestor of the audited `HEAD = 3f3c831`. A follow-up, **PR #335** ("Add full M3Y support and fix QR scanning issues for GM65 and M3Y"), commit `97ab4d4`, is also in `HEAD`. I verified the relevant code paths in `src/hosts/qr.py` and confirmed the fix is present. I did not independently reproduce the binary CompactSeedQR scan on hardware.

## What I investigated

- `gh issue view 288` — full body and all 18 comments. Root cause was traced by contributors (`@Schnuartz`, `@3rdIteration`, `@k9ert`, `kdmukai`) to a **firmware bug in newer GM65-class scanner modules** (HW version `0x02`, SW year `>= 0x17` i.e. 2023+). The buggy scanners mis-decode **binary** (Compact) SeedQR bytes (UTF-8/ISO-8859-1 byte-mode QR codes), producing consistently wrong words; ASCII QR codes (e.g. signed messages) scan correctly. The wrong bytes differ in a repeatable pattern (e.g. byte 0 `8d→ba`, byte 1 `87→cf`), indicating a scanner-side encoding bug rather than a host parsing bug.
- Identified the mitigation path: the buggy scanners can be forced into "RAW output" mode (QR code from `@Schnuartz` posted 2025-04-18, and serial command on address `0x00BC`) so binary data is passed through unchanged.
- Verified PR #299 merge state and that its commit is in `HEAD`: `git merge-base --is-ancestor 031d285 HEAD` → yes.
- Verified PR #335 (further GM65/M3Y fixes) is also in `HEAD`: `git merge-base --is-ancestor 97ab4d4 HEAD` → yes.
- Reviewed the current `src/hosts/qr.py` against the PR #299 diff (files: `src/hosts/qr.py` +293/-35, `src/platform.py`, `src/specter.py`, settings, boot).

## Current code

The fix detects the affected GM65 firmware version and enables "RAW" output mode on the scanner, persisting a flag so the fix is only re-applied when needed. Key locations in `src/hosts/qr.py`:

```python
# src/hosts/qr.py:67-75
""" Address for software version check.
(The ability to check firmware version via this memory was removed from newer GM65 type devices, but still lets us identify versions that need a fix)"""
VERSION_ADDR = b"\x00\xE2"
VERSION_NEEDS_RAW = 0x69  # A version of GM65 that needs RAW mode to be turned on...

""" Some newer GM65 Scanners need to be set to "RAW" mode to correctly scan binary QR codes.
(The address for this is actually undocumented and doesn't seem to work the same on newer varients like the GM805)"""
RAW_MODE_ADDR = b"\x00\xBC"
RAW_MODE_VALUE = 0x08
```

Detection + RAW-mode application happens in `configure_gm65()`:

```python
# src/hosts/qr.py:413-448
# Check the module software and enable "RAW" mode if required
val = self.get_setting(VERSION_ADDR, retries=5, retry_delay_ms=RETRY_DELAY_MS, invalid_values={0})
if val is None:
    return False
self.software_version = val
self.version_str = "Detected GM65 Scanner, SW:" + str(val)
if val == VERSION_NEEDS_RAW:
    val = self.get_setting(RAW_MODE_ADDR)
    if val is None:
        return False
    if val != RAW_MODE_VALUE:
        if not self.set_setting(RAW_MODE_ADDR, RAW_MODE_VALUE):
            return False
        # Re-read to confirm the scanner accepted the value, retrying
        # once more if necessary. Some scanners take a short while to
        # commit this particular setting right after power-on.
        ...
        save_required = True
    if not raw_fix_applied:
        raw_fix_applied = True
        settings_changed = True
elif raw_fix_applied:
    # Clear the flag if we are no longer dealing with a scanner that
    # requires the RAW mode compatibility tweak.
    raw_fix_applied = False
    settings_changed = True
```

The `raw_fix_applied` flag is part of the host settings and persisted via the keystore:

```python
# src/hosts/qr.py:139-147
self.settings = {
    "enabled": True,
    "aim": True,
[... 31 lines cut ...]
- The issue's 2025-04-18 comment from `@Schnuartz` provides the QR code that sets RAW output mode and notes "Winni Wu ... don't have a serial command for the thing the QR-Code does." The code instead sets the **serial** register `RAW_MODE_ADDR = b"\x00\xBC"` to `0x08` (`:74-75`, `:421-440`), which achieves the same RAW-output effect programmatically.
- Multiple contributors (`@Schnuartz`, `@k9ert`) stated in the thread (2025-08-27 and 2026-08-29 comments) that PR #299 resolves the issue; the issue was closed as COMPLETED the same day as the last comment.

## Related issues/PRs

- **PR #299** — `cryptoadvance/specter-diy` #299, merged 2025-11-27 (commit `031d285`). Primary fix. `gh pr view 299 --json state,mergedAt` → `MERGED`.
- **PR #335** — "Add full M3Y support and fix QR scanning issues for GM65 and M3Y", commit `97ab4d4`, in `HEAD`. Follow-up covering M3Y and additional GM65 fixes.
- **#314** "M3Y Scanner works not properly" (CLOSED 2025-12-22) — overlapping M3Y scanning problems; addressed by PR #335.
- **#211** "Compact SeedQR sometimes fails" (CLOSED) — earlier report of the same class of problem.
- External: `@3rdIteration/SerialBarcodeReaderTools` and the `GM65` python tool referenced in the thread were used to characterise the bug; the upstream Grow/Waveshare datasheet is in-repo at `docs/datasheets/Waveshare-GROW-GM65-S.pdf`.

## Implementation notes
[... cut ...]
## Recommendation: LIKELY FIXED — issue is CLOSED (COMPLETED 2026-08-29); PR #299 (commit `031d285`, merged 2025-11-27) + PR #335 (commit `97ab4d4`) are both in `HEAD` and implement the RAW-mode workaround and factory-reset-on-boot recovery in `src/hosts/qr.py`. Not independently reproduced on hardware; verified by code inspection against the thread's root-cause analysis. A minor robustness follow-up would be to detect the mis-encoding via zone bit `0x000D` rather than the single version byte `0x69`, but the issue itself does not require further action.