cryptoadvance/specter-diy #291: gpg signing key expired?

purple608, 2025-01-22:
It seems like the key used to sign the releases is expired?

gpg: assuming signed data in 'SHA256SUMS'
gpg: Signature made Thu 30 May 11:42:49 2024 PDT
gpg:                using RSA key 785A2269EE3A9736AC1A4F4C864B7CF9A811FEF7
gpg: Good signature from "Specter Signer <noreply@specter.solutions>" [expired]
gpg: Note: This key has expired!
Primary key fingerprint: 785A 2269 EE3A 9736 AC1A  4F4C 864B 7CF9 A811 FEF7

schnuartz-ai, 2026-08-30:
# Draft comment for issue #291 — "gpg signing key expired?"

## Current assessment

**LIKELY FIXED (and duplicate of #278).** The expired key reported here — "Specter Signer" `<noreply@specter.solutions>`, fingerprint `785A 2269 EE3A 9736 AC1A  4F4C 864B 7CF9 A811 FEF7` — is **no longer used to sign releases**. Starting with the **v1.10.3** release (published 2026-05-01) the project rotates to a fresh **"Specter Signer 2026"** key controlled by k9ert (fingerprint `9DC3 3CA8 3058 9DE3 B322 5C26 EEF5 756B 2EA4 2349`). I verified this by downloading the v1.10.3 `sha256.signed.txt` release asset and running `gpg --verify` against the imported `9DC3…2349` key — it produces a "Good signature from Specter Signer 2026" with primary key fingerprint `9DC3 3CA8 3058 9DE3 B322 5C26 EEF5 756B 2EA4 2349` (signature dated 2026-04-24). The v1.10.3 release notes state explicitly: *"v1.10.3 rotates the release signing keys (Ben removed, k9ert added) and ships an updated bootloader that embeds the new key set."*

This issue is a duplicate of the broader recurring class tracked in #278 (Stepan's expired `6F16…` key) and #185 (earlier 2021 occurrence, closed). #291 specifically concerns a *third* key (`785A…`, the pre-rotation "Specter Signer" key) that was used to sign the v1.9.0 release on 2024-05-30; that key is also superseded by the v1.10.3 rotation.

## What I investigated

- `gh issue view 291` — body is a `gpg --verify` transcript: signature made Thu 30 May 2024, RSA key `785A2269EE3A9736AC1A4F4C864B7CF9A811FEF7`, "Good signature from 'Specter Signer <noreply@specter.solutions>' [expired]" / "Note: This key has expired!". No comments, no labels.
- Identified the v1.9.0 release as the source of that signature: `gh release view v1.9.0` → published 2024-05-30, by `stepansnigirev`; its `sha256.signed.txt` is signed with ECDSA key `6F16E354F83393D6E52EC25F36ED357AB24B915F` (Stepan's key, the #278 key) — so the v1.9.0 manifest itself was *not* signed by `785A…`. The `785A…` signature reported in #291 is dated the same day (2024-05-30) and likely belongs to a parallel/interim "Specter Signer" signing path that has since been retired. Either way, neither `785A…` nor Stepan's `6F16…` is the current signer.
- Confirmed the current signing arrangement in-repo:
  - `SECURITY.md:32-35` — *"The release hash manifests (`sha256.signed.txt`) are signed with the 'Specter Signer 2026' key, controlled by k9ert, fingerprint `9DC3 3CA8 3058 9DE3 B322 5C26 EEF5 756B 2EA4 2349`."*
  - `SECURITY.md:21-24` — release-signer table lists k9ert (`ECC0 B4AB D74E 716F 5ADE 0952 28B3 58A8 843B 0109`) and miketlk (`F2DB C4C6 14C1 13E2 B15F 879A DD5C 1264 EBD6 45BE`); neither Stepan nor the `785A…` "Specter Signer" key is present.
  - `docs/quickstart.md:10` — *"Since v1.10.3 it is signed with the 'Specter Signer 2026' key, controlled by k9ert (fingerprint `9DC3 3CA8 3058 9DE3 B322 5C26 EEF5 756B 2EA4 2349`…). Older releases were signed with Stepan's PGP key."*
  - `docs/security-model.md:91-98` — same v1.10.3 key statement, plus legacy pointer to Stepan's key for older releases.
- Independently verified the live signature (not just the docs):
  - `gh release download v1.10.3 --pattern sha256.signed.txt`
  - `gpg --keyserver keyserver.ubuntu.com --recv-key "9DC3 3CA8 3058 9DE3 B322 5C26 EEF5 756B 2EA4 2349"` → imported "Specter Signer 2026".
  - `gpg --verify sha256.signed.txt` → `Good signature from "Specter Signer 2026"`, primary key fingerprint `9DC3 3CA8 3058 9DE3 B322 5C26 EEF5 756B 2EA4 2349`, signature made 2026-04-24. (Trust warning "not certified with a trusted signature" is expected for a freshly imported key and is not an integrity problem.)
- `git log --all --grep="gpg"` / `--grep="signing key"` → no commit directly references this issue; the rotation landed via the bootloader submodule bump and PRs #306 ("Remove ben and add k9ert keys for releasing") + #350 ("Update bootloader submodule to latest"), both listed in the v1.10.3 release notes.

## Current code

No application/firmware code is involved — this is a release-process / key-management issue. The relevant artefacts are docs and release assets (all paths verified to exist):

- `SECURITY.md:21-24` — current release-signer table (k9ert, miketlk; `785A…` "Specter Signer" and Stepan's `6F16…` both absent).
- `SECURITY.md:32-35` — current `sha256.signed.txt` signing key ("Specter Signer 2026", `9DC3…2349`).
- `docs/quickstart.md:10` — user-facing verification instructions reflecting the v1.10.3 key change.
- `docs/security-model.md:91-98` — security-model doc reflecting the same.
- Release v1.10.3 assets (`sha256.signed.txt`) — verified live (see above).

## Reproduction / validation

- **VERIFIED (live GPG check):** downloaded v1.10.3 `sha256.signed.txt`, imported `9DC3…2349` from the Ubuntu keyserver, and confirmed `gpg --verify` reports a good signature from "Specter Signer 2026". This directly establishes that the current release is *not* signed by the expired `785A…` key.
- **VERIFIED (docs):** `SECURITY.md:32-35` and `docs/quickstart.md:10` name `9DC3…2349` as the current signer; `785A…` is not referenced anywhere in the current repo (`grep` for `785A2269`/`A811FEF7` across all files → no matches).
- **NOT INDEPENDENTLY VERIFIED:** the expiry status of the `785A…` key itself (the issue says it is expired; I did not import and inspect it, because it is moot — the key is no longer the signer). The new "Specter Signer 2026" key's own expiry date was not checked against the syst [... cut ...]
Recommendation: LIKELY FIXED (and duplicate of #278) — verified by a live `gpg --verify` of the v1.10.3 `sha256.signed.txt` against the imported "Specter Signer 2026" key (`9DC3 3CA8 3058 9DE3 B322 5C26 EEF5 756B 2EA4 2349`), which yields a good signature; the expired `785A…` "Specter Signer" key from this issue is no longer the signer (documented in `SECURITY.md:32-35` and `docs/quickstart.md:10`). Suggest closing #291 (and the sibling #278) as fixed, and proactively planning the next key rotation before the "Specter Signer 2026" key lapses.

Schnuartz, 2026-08-31:
Closing this as fixed.

The expired signing key reported in this issue (`785A 2269 EE3A 9736 AC1A 4F4C 864B 7CF9 A811 FEF7`) is no longer used for current releases.

Since **v1.10.3**, release hash manifests (`sha256.signed.txt`) are signed with the new **"Specter Signer 2026"** key:

`9DC3 3CA8 3058 9DE3 B322 5C26 EEF5 756B 2EA4 2349`

This is documented in `SECURITY.md` and `docs/quickstart.md`. The signing-key rotation was implemented through #306 and the subsequent bootloader update in #350.

The v1.10.3 `sha256.signed.txt` has also been independently verified with `gpg --verify`, producing a valid signature from the new "Specter Signer 2026" key.

Historical releases may of course still show the old key as expired when verified today, but current releases no longer use that key.

**Verified fixed, closing.**

closed 2026-08-31 by Schnuartz