cryptoadvance/specter-diy #292: "Enter BIP-39 passphrase" would be better than "Enter BIP-39 password"  -> would make it clearer for the user

Schnuartz, 2025-02-03:
Most people use the word Passphrase instead of "BIP-39  password" so it would be clearer for the user what the button is doing. 

Most users don't know what BIP39 is, and "password" sounds like the Pin code. "Enter BIP-39 passphrase" would be optimal, because than we have the clarification of the term BIP39 as well as the term "passhprase"

[image]

louisthecat86, 2025-03-04:
This pull request addresses the issue. See [PR #294](https://github.com/cryptoadvance/specter-diy/pull/294).

louisthecat86, 2025-03-04:
[image]